jbuehl/solaredge #1: Need more info please?

Jantje1972, 2015-05-21:
Can you give me more info on your project with the solar edge inverter like on witch hardware its running and so on ?

Greetings Jan,

jbuehl, 2015-05-21:
It's running on a Raspberry Pi model B.  Because I need to sniff all the traffic coming from the inverters, I also have a second USB ethernet adapter that allows the RPi to be a bridge from my network to the inverters.

Jantje1972, 2015-05-21:
You have a very cool project there, at the moment i'm using DOMOTICZ for my home automation also on a raspberry but a model 2,

In DOMOTICZ the solar edge logging is not working for me so i'm looking for an alternative.

jbuehl, 2015-05-21:
Ugh!  I just realized that I need to clean this up.  There is a lot of junk in this repository that is obsolete and irrelevant.  I will do that today and let you know.

I had not heard of DOMOTICZ.  I will check it out.

jbuehl, 2015-05-21:
OK I have cleaned up the repository and updated the comments in the source.  Please let me know if you have any more questions.

Jantje1972, 2015-05-21:
Thanks for the cleanup, is much more readable, i will ask the maker off DOMOTICZ if he can use this to make a better implementation in his software!

closed 2015-05-21 by Jantje1972

wimsan, 2015-08-14:
I'm trying to get this working on my synology NAS. I get an error :+1: 
debug: True
debugFiles: True
debugData: True
headers: True
delim: ,
dataBase:
dbHostName:
userName:
password:
append: a
inFileName: stdin
invFileName:
optFileName:
jsonFileName:
reading stdin
solaredge inputSeq 0 seDataLen 14951
solaredge inputSeq 1 seDataLen 0
solaredge inputSeq 2 seDataLen 0
solaredge inputSeq 3 seDataLen 342
solaredge     seHdr a9 fe 03a9 7f190ff7 fffffffe 0500
solaredge         seType 0010 seId 7F190FF7 seDeviceLen 124
Traceback (most recent call last):
  File "/root/solaredge/seconvert2.py", line 454, in <module>
    seConvert()
  File "/root/solaredge/seconvert2.py", line 197, in seConvert
    seHeader(inRec+inFile.read(2))
  File "/root/solaredge/seconvert2.py", line 238, in seHeader
    seInvData = readData(inFile, invInFmt, seDeviceLen)
  File "/root/solaredge/seconvert2.py", line 262, in readData
    return list(struct.unpack(inFmt, inFile.read(seDeviceLen)))
struct.error: unpack requires a string argument of length 104

I have no idea why this goes wrong. My inverter is a SE3000 that sends it's data to domoticz on ip address 192,168.1.51 port 22222

please advise
wimsan