jbuehl/solaredge #6: A few errors

martynwendon, 2015-10-14:
I've been getting the occasional traceback, I'm guessing it's caused by some sort of spurious data that the python code isn't expecting.
```
Traceback (most recent call last):
  File "/usr/src/solaredge/seconvert2.py", line 534, in <module>
    seConvert()
  File "/usr/src/solaredge/seconvert2.py", line 200, in seConvert
    seHeader(inRec+readBytes(2))
  File "/usr/src/solaredge/seconvert2.py", line 234, in seHeader
    seOptData = readData(inFile, optInFmt, seDeviceLen)
  File "/usr/src/solaredge/seconvert2.py", line 310, in readData
    return list(struct.unpack(inFmt, readBytes(seDeviceLen)[:(len(inFmt)-1)*4]))
struct.error: unpack requires a string argument of length 36
Traceback (most recent call last):
  File "/usr/src/solaredge/seconvert1.py", line 253, in <module>
    readPcapRec(pcapFile)
  File "/usr/src/solaredge/seconvert1.py", line 109, in readPcapRec
    sys.stdout.flush()
IOError: [Errno 32] Broken pipe
```
```
Traceback (most recent call last):
  File "/usr/src/solaredge/seconvert2.py", line 534, in <module>
    seConvert()
  File "/usr/src/solaredge/seconvert2.py", line 200, in seConvert
    seHeader(inRec+readBytes(2))
  File "/usr/src/solaredge/seconvert2.py", line 243, in seHeader
    seOptData = convertNewOptData(seData)
  File "/usr/src/solaredge/seconvert2.py", line 298, in convertNewOptData
    (timeStamp, uptime) = struct.unpack("<LH", seData[0:6])
struct.error: unpack requires a string argument of length 6
Traceback (most recent call last):
  File "/usr/src/solaredge/seconvert1.py", line 253, in <module>
    readPcapRec(pcapFile)
  File "/usr/src/solaredge/seconvert1.py", line 109, in readPcapRec
    sys.stdout.flush()
IOError: [Errno 32] Broken pipe
```

github won't let me upload a pcap file, so I've put it at https://dl.dropboxusercontent.com/u/52590317/solaredge-20151014000004.pcap - this produces the above errors for me consistently

Please let me know if there's anything else I can provide to track down the cause?

jbuehl, 2015-10-14:
I will try to look at it in the next few days.

jbuehl, 2015-10-19:
Filtering the data by IP address had no effect but I found a bug in seconvert2.py.  I just pushed a change that seems to eliminate the crash when processing your pcap file, martynwendon.  I don't know if the output is meaningful, but at least it runs.

martynwendon, 2015-10-20:
Thanks, that seems to have sorted it, processed all my backlog pcap files without a single issue and data all looks good!

closed 2015-10-26 by jbuehl