jbuehl/solaredge #7: Identify messages sent to and from the SolarEdge server

jbuehl, 2015-10-27:
seconvert1.py should be able to reliably identify messages sent from the inverter to the SolarEdge server.  If the server being communicated with is prod.solaredge.com (217.68.152.66) this can easily be done, however the IP address has been known to change without notice to another SolarEdge server.  One alternative might be to filter messages whose source IP is on the local network (192.x.x.x).

jdwhite, 2015-12-14:
If one is able to assign a fixed address to the inverter via DHCP then it'd be easy to detect all traffic coming from the inverter.  I tried to set a static address at the inverter but never got the LAN options menu that the documentation showed, so I went the fixed address DHCP route. Failing that, looking for all outgoing traffic with a destination port of 22222 would be a reasonable option.  There's the possibility of seeing non-inverter traffic that way, but I suspect it would be infrequent and the script would ignore it anyway since it wouldn't have the magic header.

jbuehl, 2015-12-16:
I hadn't thought of using the destination port as a filter.  That's useful.  The server that my inverter is talking to is 217.68.144.150 so maybe using the destination IP range 217.68.0.0/16 could also work.

jdwhite, 2015-12-16:
Interesting considering that prod.solaredge.com (currently) resolves to 217.68.152.66, which lives in another completely different netblock owned by SolarEdge than 144.50 does.  The docs for my Inverter explicitly mention that if you use a firewall that you must allow prod.solaredge.com:22222 through.  I'm wondering if your inverter is older and/or has the 144.50 address hard coded in its firmware rather than the name.  I've seen DNS queries from my inverter in the packet dumps.

jbuehl, 2015-12-16:
Mine had been using prod.solaredge.com until earlier this year when it suddenly switched to the other IP address.  I had been filtering by address when I was capturing so I lost data for a few days until I fixed it.

closed 2016-02-26 by jbuehl