jbuehl/solaredge #10: Support other monitoring systems?

dankegel, 2016-04-18:
https://www.solarpaneltalk.com/forum/solar-panels-for-home/solar-panel-system-equipment/19587-mirroring-intercepting-sunpower-monitoring-traffic/page3
mentions that someone wrote a script to watch sunpower traffic, partly based on this project.
(He's happy to help others who want to do the same thing, but is not willing/able to upload his script.)

Maybe this project could be generalized to support sunpower, too?  

Filing this bug mostly just as a way to make it easier for sunpower users to follow breadcrumbs.

jbuehl, 2016-04-18:
I'm not sure I understand how that would work.  This project is mostly about reverse engineering the protocol that is proprietary to SolarEdge.  Other than the use of the generic tcpdump utility to capture the network traffic, I don't see how any of it would be relevant to whatever protocol Sunpower uses.

dankegel, 2016-04-18:
The upload to pvoutput is also relevant.

You could move the SolarEdge specific stuff into e.g. protocols/solaredge, and create a protocols/sunpower subdirectory, and autodetect the protocol.

But I'm not suggesting actually doing that until you have a user who needs it and wants to implement it.  It's just an idea, and mostly an excuse to point to that thread where somebody described how to decode the sunpower protocol.

jbuehl, 2016-04-18:
If someone wants to contribute a se2pvoutput.py script to this project it would be a welcome addition.

astroboy2, 2016-04-20:
actually, my mistake - i told dankegel that my script used the http post code from this script, but i was mistaken... my code is written in perl, not python.

closed 2016-05-10 by jbuehl