jbuehl/solaredge #13: Connecting via LAN

d-tamm, 2016-06-18:
This is a continuation of #11 and my questions with regard to LAN connections. I chose to open a new issue in order to keep discussions separated. I know that this is more a kind of forum usage and should not be under "issues", but I do not know a better place.
So I am trying to get data out of my SE5k inverter via LAN. After killing dnsmasq which was using my DNS port, and connecting my computer via ethernet cable and network interface enp0s25, I have tried with the following command:
`sudo python semonitor.py -d stdout -n wlp2s0`
which puts the script in a waiting mode but nothing happens. According to @jbuehl this is due to encryption, see #11. Adding -vv option, I get the following output after restarting the inverter:

> Jun 18 14:22:25 0.0.0.0:68 --> message: 40 length: 257 
Jun 18 14:22:25 192.168.18.255:68 <-- message: 41 length: 300 
Jun 18 14:22:25 waiting for dhcp message 
Jun 18 14:22:26 0.0.0.0:68 --> message: 42 length: 269 
Jun 18 14:22:26 192.168.18.255:68 <-- message: 43 length: 300 
Jun 18 14:22:26 waiting for dhcp message 
Jun 18 14:22:37 192.168.18.2:3111 --> message: 1 length: 37 
Jun 18 14:22:37 192.168.18.2:3111 <-- message: 2 length: 72 
Jun 18 14:22:37 waiting for dns message 
Jun 18 14:22:37 connection from 192.168.18.2:3110 
Jun 18 14:22:37 starting read thread 
Jun 18 14:22:46 <socket> --> message: 1 length: 56 
Jun 18 14:23:16 <socket> --> message: 2 length: 56 
Jun 18 14:23:46 <socket> --> message: 3 length: 56 
Jun 18 14:24:46 <socket> --> message: 4 length: 56 
Jun 18 14:25:46 <socket> --> message: 5 length: 56 
Jun 18 14:26:46 <socket> --> message: 6 length: 56 
Jun 18 14:29:27 192.168.18.2:3113 --> message: 3 length: 37 
Jun 18 14:29:27 192.168.18.2:3113 <-- message: 4 length: 72 
Jun 18 14:29:27 waiting for dns message 
Jun 18 14:30:00 192.168.18.2:3115 --> message: 5 length: 37 
Jun 18 14:30:00 192.168.18.2:3115 <-- message: 6 length: 72 
Jun 18 14:30:00 waiting for dns message 
Jun 18 14:32:03 192.168.18.2:3117 --> message: 7 length: 37 
Jun 18 14:32:03 192.168.18.2:3117 <-- message: 8 length: 72 
Jun 18 14:32:03 waiting for dns message 
Jun 18 14:32:05 192.168.18.2:68 --> message: 44 length: 250 
Jun 18 14:32:05 waiting for dhcp message 
Jun 18 14:32:07 0.0.0.0:68 --> message: 45 length: 257 
Jun 18 14:32:07 192.168.18.255:68 <-- message: 46 length: 300 
Jun 18 14:32:07 waiting for dhcp message 
Jun 18 14:32:09 0.0.0.0:68 --> message: 47 length: 269 
Jun 18 14:32:09 192.168.18.255:68 <-- message: 48 length: 300 
Jun 18 14:32:09 waiting for dhcp message 
Jun 18 14:32:20 192.168.18.2:3120 --> message: 9 length: 37 
Jun 18 14:32:20 192.168.18.2:3120 <-- message: 10 length: 72 
Jun 18 14:32:20 waiting for dns message 
Jun 18 14:32:22 192.168.18.2:68 --> message: 49 length: 250 

No 003d or 0503 messages as long as I can see.
@jbuehl you mentioned to extract the encryption key by RS232. My documentation does not mention RS232, only RS485. But when I open the cover, there is a 9-fold connector, with three contacts for RS485-1, three for RS485-2, and lastly three for RS232 (Rx, Tx, G). Can I suppose that this RS232 connector is active and can be used though it is not mentioned in the documentation? I ask because I would have to go and buy a serial cable.
Otherwise, is it also possible to extract the encryption keys via RS485? I see that common USB to RS485 connectors have 2 contacts, while the inverter has 3. Is it the ground contact that is unused then? In the documentation, only the connection between several inverters is described.
I begin feeling a bit stupid with all my questions. This is quite new for me and there seems to be very little documentation. I appreciate your help!

Whizkidzz, 2016-06-18:
The most simple way is to use the mini USB port in the right side of the comm board
This will show on the computer as comm port.

jbuehl, 2016-06-18:
Sorry, I should have told you to add the -vvv option which will display the contents of the messages.  It looks like the inverter is getting its IP address and successfully connecting to semonitor.py, but I don't know what the 37 and 72 byte messages are that are being exchanged.

The RS232 lines on the 9 position connector should be active and you can connect them to an adapter, or as @Whizkidzz said you can connect the mini USB port directly to a usb port.  http://www.solaredge.com/sites/default/files/application-note-upgrading-solaredge-inverter-fw.pdf RS485 does not require the ground, just the 2 active lines.  It is interesting that my inverters which are SE5000s describe the RS232 connections, while the SE5K documentation doesn't.  I don't know if you can get the key using RS485.

d-tamm, 2016-06-18:
Thank you for your patience. It seems that I am getting further. I have now been able to extract a key via USB (indeed much easier than native RS232 for me!). I got a 32 byte string but needed to execute the command as root.
Then, using the command
`python solaredge/semonitor.py -d stdout -n enp0s25 -k 7E1A0823.key -vvv`
I get this output:

> Jun 18 20:15:31 debugEnable: True 
Jun 18 20:15:31 debugFiles: True 
Jun 18 20:15:31 debugMsgs: True 
Jun 18 20:15:31 debugData: True 
Jun 18 20:15:31 debugRaw: False 
Jun 18 20:15:31 debugFileName: stdout 
Jun 18 20:15:31 haltOnException: False 
Jun 18 20:15:31 inFileName: network 
Jun 18 20:15:31 serialDevice: False 
Jun 18 20:15:31 networkDevice: True 
Jun 18 20:15:31 networkSvcs: True 
Jun 18 20:15:31 netInterface enp0s25 
Jun 18 20:15:31     ipAddr 192.168.18.1 
Jun 18 20:15:31     subnetMask 255.255.255.0 
Jun 18 20:15:31     broadcastAddr 192.168.18.255 
Jun 18 20:15:31 following: True 
Jun 18 20:15:31 passiveMode: False 
Jun 18 20:15:31 commandAction: False 
Jun 18 20:15:31 masterMode: False 
Jun 18 20:15:31 outFileName: stdout 
Jun 18 20:15:31 append: w 
Jun 18 20:15:31 keyFileName: 7E1A0823.key 
Jun 18 20:15:31 key: xxx 
Jun 18 20:15:31 opening network 
Jun 18 20:15:31 starting dhcpThread 
Jun 18 20:15:31 waiting for dhcp message 
Jun 18 20:15:31 starting dnsThread 
Jun 18 20:15:31 waiting for connection 
Jun 18 20:15:31 waiting for dns message 
Jun 18 20:16:40   
Jun 18 20:16:40 192.168.18.2:3200 --> message: 1 length: 32 
Jun 18 20:16:40 id: 3e41 
Jun 18 20:16:40 flags: 0100 
Jun 18 20:16:40 question 
Jun 18 20:16:40     name: www.google.com 
Jun 18 20:16:40     type: 0001 
Jun 18 20:16:40     class: 0001 
Jun 18 20:16:40 192.168.18.2:3200 <-- message: 2 length: 62 
Jun 18 20:16:40   
Jun 18 20:16:40 id: 3e41 
Jun 18 20:16:40 flags: 8000 
Jun 18 20:16:40 question 
Jun 18 20:16:40     name: www.google.com 
Jun 18 20:16:40     type: 0001 
Jun 18 20:16:40     class: 0001 
Jun 18 20:16:40 answer 
Jun 18 20:16:40     name: www.google.com 
Jun 18 20:16:40     type: 0001 
Jun 18 20:16:40     class: 0001 
Jun 18 20:16:40     TTL: 86400 
Jun 18 20:16:40     resource: 192.168.18.1 
Jun 18 20:16:40 waiting for dns message 
Jun 18 20:16:41   
Jun 18 20:16:41 192.168.18.2:3201 --> message: 3 length: 37 
Jun 18 20:16:41 id: 7442 
Jun 18 20:16:41 flags: 0100 
Jun 18 20:16:41 question 
Jun 18 20:16:41     name: prod2.solaredge.com 
Jun 18 20:16:41     type: 0001 
Jun 18 20:16:41     class: 0001 
Jun 18 20:16:41 192.168.18.2:3201 <-- message: 4 length: 72 
Jun 18 20:16:41   
Jun 18 20:16:41 id: 7442 
Jun 18 20:16:41 flags: 8000 
Jun 18 20:16:41 question 
[... 79 lines cut ...]
Jun 18 20:17:08 connection from 192.168.18.2:3204 
Jun 18 20:17:08 starting read thread 
Jun 18 20:17:22   
Jun 18 20:17:22 <socket> --> message: 1 length: 56 
Jun 18 20:17:22 dataLen:    0022 
Jun 18 20:17:22 dataLenInv: ffdd 
Jun 18 20:17:22 sequence:   031f 
Jun 18 20:17:22 source:     7e1a0823 
Jun 18 20:17:22 dest:       fffffffe 
Jun 18 20:17:22 function:   0503 
Jun 18 20:17:22 creating decryption object with key xxx 
Jun 18 20:17:22 Ignoring this message 
Jun 18 20:18:22   
Jun 18 20:18:22 <socket> --> message: 2 length: 56 
Jun 18 20:18:22 dataLen:    0022 
Jun 18 20:18:22 dataLenInv: ffdd 
Jun 18 20:18:22 sequence:   031f 
Jun 18 20:18:22 source:     7e1a0823 
Jun 18 20:18:22 dest:       fffffffe 
Jun 18 20:18:22 function:   0503 
Jun 18 20:18:22 creating decryption object with key xxx 
Jun 18 20:18:22 Ignoring this message 

Seems like the inverter is asking quite repeatedly for prod2.solaredge.com.
Finally, I get some 0503 messages, but not 003d as @jbuehl mentioned. Nothing more seems to happen, Any further ideas?

jbuehl, 2016-06-18:
The inverter is sending a DNS request for prod2.solaredge.com and semonitor.py is responding, but the inverter seems to ignore the response several times.  I don't know why it is doing that, but it eventually gets it and successfully connects.  The inverter is sending the 0503 message, but semonitor.py is not sending an acknowledgement message so the inverter never starts sending any 003d messages.   This is a bug which I can fix in the next day or so.  In the meantime, you could try setting the inverter to unencrypted mode using the following command over RS232

`python semonitor.py -c 0011,H329,L0/0030,H01f4,L0 -s 7f101234 -t 2 /dev/ttyUSB0`

This would mean that the decryption key would not required to decode the data and is the method that I have been using for the last few months.

d-tamm, 2016-06-19:
Getting closer! Main problem today was that I tore off the USB connector from the inverter's main board when I would put the computer on a shelf while connected via USB. Took me a while to solder back the connector!
When that was done, I could put the inverter to unencrypted mode, and now I can receive data both with the -n and the -tn options.
Now, my next project is getting it up and running on a server. Tried on my Synology server but cannot get compilation of python netifaces working. Next try is my Raspberry Pi.
I think it would be good to mention at the examples in the readme what they do, e.g. that the example above will be setting the inverter to unencrypted mode. It would also be good if all the commands could be documented. But maybe there is already documentation for this from SolarEdge, in that case reference to that in the readme would be good.
@jbuehl I know that you already invest a lot of time in this project. Please take the above as suggestions for how to increase the usability of your work for other users :)

jbuehl, 2016-06-19:
Good to hear that you are getting data, @d-tamm.  The only purpose for the netifaces module is to get the IP address, netmask, and broadcast address associated with the interface.  If you can't get that working, you could just hard code them in lines 267-270 of seConf.py.

Yes, the documentation could be improved.  I didn't expect this project to become as popular as it has.  There is no SolarEdge documentation to refer to.  All of this was done by reverse engineering their code.

d-tamm, 2016-06-19:
Great suggestion, @jbuehl . Now semonitor is working on my Synology box. I had to install the packages Python and Python Module, and to comment out the usage of netifaces in seConf.py.
Now, that the data gets to the right place, I will get to the funny part and implement some usage of it.
As for the documentation, I suspected that this has begun as a private project of yours, with no expectations to get popular. I do not think that I can help very much with the coding, but I would like to contribute with documentation if possible!

closed 2016-08-23 by d-tamm