jbuehl/solaredge #15: New to Using Solar Edge Inverter

Joshua-KSU, 2016-07-01:
Hi:

I am working with a group of students at Kansas State University on a research project. We are trying to grab data from a Solar Edge Inverter at 15 minute intervals to store and display on our own website. From what I have read so far it appears that we need to use semonitor.py to grab the data from the inverter and then use seextract.py to take that data and convert it into a readable file. Could you please explain in detail how this is done and how to set something like this up? Thank you very much for your time.

jbuehl, 2016-07-01:
Based on the ongoing interest in this, it seems like I need to spend some time to document this better.  In the meantime, if you want to monitor via ethernet you can look at issues #13 and #14 to see how to do this.  You can also monitor via the RS232 or RS485 interfaces.  I will post more in this issue shortly.

jbuehl, 2016-07-01:
SolarEdge inverter monitoring data can be accessed by any of 3 hardware interfaces - ethernet, RS232, or RS485.  Data can be parsed by semonitor.py either in an active mode, where semonitor.py interacts with the inverter directly, or in a passive mode where the data that is being sent to the SolarEdge monitoring portal is captured.

```
interface    modes
---------    -----
ethernet     active, passive
RS232        active
RS485        active, passive
```

Here is an overview of the different methods:

1. ethernet active

The inverter is made to connect to semonitor.py instead of the SolarEdge monitoring portal by resolving its hostname (usually prod.solaredge.com) to the IP address of the server running semonitor.py.  This may be done by connecting the ethernet interface of the inverter to a dedicated ethernet interface on the server running semonitor.py and specifying the -n option which will run a simplified DHCP/DNS server that responds to the inverter on the dedicated interface.  Alternatively, another DNS server in the network could be made to resolve the address, but how to do that is beyond the scope of this discussion.

In this mode, the inverter will no longer be communicating with the SolarEdge server so monitoring services and firmware updates will no longer be available.

The encryption key for the inverter must first be obtained and given to semonitor.py using the -k option, or the inverter must be set to unencrypted mode.  Either way, this requires temporarily connecting via the RS232 interface.

2. ethernet passive

Data sent between an inverter and the SolarEdge monitoring portal is captured and fed into semonitor.py.  In order to do this the network traffic between them must be monitored by a server on the network.  This will not work if the inverter and the local server are connected to an ethernet switch.  Either they could be connected through an ethernet hub, or the inverter could be connected to a dedicated network interface on a server that bridges to the local network.

Traffic may be captured to a pcap file using wireshark or tcpdump or some other program.  The resulting file must be processed through seextract.py before it is fed into semonitor.py.  In this mode, semonitor.py is reading data from a file or stdin and isn't actually reading from the network directly.

SolarEdge will place the inverter into encrypted mode, so semonitor.py will need to be given the encryption key using the -k option.  The key must be accessed via the RS232 interface.

jbuehl, 2016-07-01:
**RS485 active**

The server running semonitor.py is connected to one or more inverters through a serial port with an RS485 adapter.  All inverters must be configured as RS485 slaves and semonitor.py functions as the RS485 master.  The inverter serial numbers are specified to semonitor.py using the -s option and the -m option specifies master mode.  There cannot be another master device on the RS485 bus.

**RS485 passive**

The server running semonitor.py is connected to one or more inverters through a serial port with an RS485 adapter.  Some other device such as an inverter or a SolarEdge Control and Communication Gateway functions as the RS485 master and semonitor.py just monitors the data on the bus.  Depending on capability of the hardware running semonitor.py, there may be issues with it being able to keep up with the data rate so there may be lost or corrupted data.

jbuehl, 2016-07-01:
**RS232 active**

RS232 is a point to point interface so only active mode is meaningful.  While it is technically possible to capture RS232 traffic into a file and process the file using semonitor.py, that is out of scope of this discussion.

Specified commands may be sent to the inverter from semonitor.py using the -c option over any of the 3 hardware interfaces, but there are certain commands that the inverter will only accept when coming from the RS232 interface.  If an inverter is in encrypted mode, commands to get the encryption key or to place the inverter in unencrypted mode must be sent over RS232.

jbuehl, 2016-07-01:
**Ethernet hardware configurations**

Active

```
                  semonitor
inverter -------- server

```
Passive using ethernet hub

```
                     capture
inverter ---     --- server
            |   |
           ethernet 
             hub
              |
              |
              |
          SolarEdge
            server

```
Passive using dual capture server interfaces

```
                   capture
inverter --------- server ----
                              |
                              |
                              |
                          SolarEdge
                            server

```
**RS485 hardware configurations**
Active
```
inverter ---
            |
            |
            |     semonitor
inverter -------- server
            |
            |
            |
inverter ---
```
Passive
```
                  RS485
inverter -------- master
            |
            |
            |     semonitor
inverter -------- server
            |
            |
            |
inverter ---
```
**RS232 hardware configuration**
```
                  semonitor
inverter -------- server

```

naltatis, 2017-10-22:
@jbuehl Thanks for this explanation! Could you add this to the README?
This would definitly make it easier for new users.

jbuehl, 2017-11-24:
Added this to README.md.

closed 2017-11-24 by jbuehl