jbuehl/solaredge #16: RS485 question

madether, 2016-07-11:
I decided that ethernet was too much trouble.
I have the Solaredge rs485 adapter board installed.  
If I execute python /home/pi/solaredge/semonitor.py -o /mnt/nfs/test.json -s [inverter_sn] -t 4 /dev/ttyUSB0 it just sits there, no output.
If i execute python /home/pi/solaredge/semonitor.py -c 322 -s [inverter_sn] -vvvv -t 4 /dev/ttyUSB0 I get the usual 
{"data": {"Emon": 32784.90234375, "Etot": 40462.0, "Eyear": 39798.01953125, "Eday": 80.69844818115234, "Time1": "Mon Jul 11 10:00:00 2016"}, "command": 802, "response": 909, "sequence": 10}
The adapter board is set to sunspec, slave and device (non Se Logger.)
My best guess is I am doing something very wrong here.

madether, 2016-07-11:
python /home/pi/solaredge/semonitor.py -c 0012,H0329  -s 07f13e4a6 -t 4 /dev/ttyUSB0
{"data": {"type": 0, "value": 1}, "command": 18, "response": 144, "sequence": 60}
is this expected?

jbuehl, 2016-07-11:
For RS485, you need to add the -m option to the command line

`python /home/pi/solaredge/semonitor.py -o /mnt/nfs/test.json -m -s [inverter_sn] -t 4 /dev/ttyUSB0`

The protocol must be set to SE (SolarEdge, not Sunspec).

madether, 2016-07-11:
Jul 12 07:29:24 debugEnable: True
Jul 12 07:29:24 debugFiles: True
Jul 12 07:29:24 debugMsgs: True
Jul 12 07:29:24 debugData: True
Jul 12 07:29:24 debugRaw: False
Jul 12 07:29:24 debugFileName: stdout
Jul 12 07:29:24 haltOnException: False
Jul 12 07:29:24 inFileName: /dev/ttyUSB0
Jul 12 07:29:24 inputType: 4
Jul 12 07:29:24 serialDevice: True
Jul 12 07:29:24     baudRate: 115200
Jul 12 07:29:24 networkDevice: False
Jul 12 07:29:24 networkSvcs: False
Jul 12 07:29:24 following: True
Jul 12 07:29:24 passiveMode: False
Jul 12 07:29:24 commandAction: False
Jul 12 07:29:24 masterMode: True
Jul 12 07:29:24 slaveAddrs: 7f16e4a6
Jul 12 07:29:24 outFileName: stdout
Jul 12 07:29:24 append: w
Jul 12 07:29:24 opening /dev/ttyUSB0
Jul 12 07:29:24 starting read thread
Jul 12 07:29:24 starting master thread
Jul 12 07:29:24 dataLen:    0000
Jul 12 07:29:24 dataLenInv: ffff
Jul 12 07:29:24 sequence:   001b
Jul 12 07:29:24 source:     fffffffe
Jul 12 07:29:24 dest:       7f16e4a6
Jul 12 07:29:24 function:   0302
Jul 12 07:29:24 /dev/ttyUSB0 <-- message: 1 length: 22
Jul 12 07:29:24
Jul 12 07:29:24
Jul 12 07:29:24 /dev/ttyUSB0 --> message: 1 length: 52
Jul 12 07:29:24 dataLen:    001e
Jul 12 07:29:24 dataLenInv: ffe1
Jul 12 07:29:24 sequence:   2460
Jul 12 07:29:24 source:     7f16e4a6
Jul 12 07:29:24 dest:       fffffffc
Jul 12 07:29:24 function:   0333
Jul 12 07:29:24 Unknown function 0x0333
Jul 12 07:29:24
Jul 12 07:29:24 /dev/ttyUSB0 --> message: 2 length: 52
Jul 12 07:29:24 dataLen:    001e
Jul 12 07:29:24 dataLenInv: ffe1
Jul 12 07:29:24 sequence:   2461
Jul 12 07:29:24 source:     7f16e4a6
Jul 12 07:29:24 dest:       fffffffc
Jul 12 07:29:24 function:   0333
Jul 12 07:29:24 Unknown function 0x0333
Jul 12 07:29:24
Jul 12 07:29:24 /dev/ttyUSB0 --> message: 3 length: 52
Jul 12 07:29:24 dataLen:    001e
Jul 12 07:29:24 dataLenInv: ffe1
Jul 12 07:29:24 sequence:   2462
Jul 12 07:29:24 source:     7f16e4a6
Jul 12 07:29:24 dest:       fffffffc
Jul 12 07:29:24 function:   0333
Jul 12 07:29:24 Unknown function 0x0333
Jul 12 07:29:24
Jul 12 07:29:24 /dev/ttyUSB0 --> message: 4 length: 52
Jul 12 07:29:24 dataLen:    001e
Jul 12 07:29:24 dataLenInv: ffe1
Jul 12 07:29:24 sequence:   2463
Jul 12 07:29:24 source:     7f16e4a6
Jul 12 07:29:24 dest:       fffffffc
Jul 12 07:29:24 function:   0333
Jul 12 07:29:24 Unknown function 0x0333
Jul 12 07:29:24
Jul 12 07:29:24 /dev/ttyUSB0 --> message: 5 length: 52
Jul 12 07:29:24 dataLen:    001e
Jul 12 07:29:24 dataLenInv: ffe1
Jul 12 07:29:24 sequence:   2464
Jul 12 07:29:24 source:     7f16e4a6
Jul 12 07:29:24 dest:       fffffffc
Jul 12 07:29:24 function:   0333
Jul 12 07:29:24 Checksum error. Expected 0xe4a6, got 0x7f42
Jul 12 07:29:24 data:       1e 00 e1 ff 64 24 a6 e4 13 7f fc ff ff ff 33 03
Jul 12 07:29:24 data:       00 10 ff ff ff ff 7f ff 64 00 e8 03 00 00 a6 e4
Jul 12 07:29:24 data:       13 7f b0 fc bf 45 a6 e4 13 7f be d2 90 00 a6 e4
Jul 12 07:29:24 Ignoring this message

jbuehl, 2016-07-11:
Good news.  Now semonitor.py is successfully talking to your inverter.  It is sending the 0333 message which I have never seen before.  The name of the function is PROT_CMD_POLESTAR_SEND_POWER_REDUCER_MASTER_PACKET but I don't know what it is supposed to do.  If you let it run for a while, you should see an occasional 0500 message and some data should get written to the output file that you specify.

If you are getting performance data then you could probably ignore the 0333 message.  Maybe you could post a question in issue #8 and see if anyone who is on that thread might have an idea what it is.

madether, 2016-07-12:
So i ran the monitor for the day and no decernable output.

madether, 2016-07-22:
so any ideas why the logging just stops after the first few records?
Jul 12 07:29:24 function: 0333
Jul 12 07:29:24 Checksum error. Expected 0xe4a6, got 0x7f42
Jul 12 07:29:24 data: 1e 00 e1 ff 64 24 a6 e4 13 7f fc ff ff ff 33 03
Jul 12 07:29:24 data: 00 10 ff ff ff ff 7f ff 64 00 e8 03 00 00 a6 e4
Jul 12 07:29:24 data: 13 7f b0 fc bf 45 a6 e4 13 7f be d2 90 00 a6 e4
Jul 12 07:29:24 Ignoring this message

jbuehl, 2016-07-22:
I don't know.  Since you have the other equipment connected via RS485 the inverter is probably going to have to be designated as the master which means you will need to do passive monitoring with semonitor.py. The problem with that is the raspberry pi is not fast enough to keep up with the data in real time so you will see corruption issues such as the checksum error.  Since I don't have this equipment I'm just guessing.

You could try capturing the data into a file and then processing it later with semonitor.py.

```
cat /dev/ttyUSB0 > capture.dat
python /home/pi/solaredge/semonitor.py -d stdout -vvv capture.dat
```

madether, 2016-07-22:
Can the inverter have to master rs485 setups.  The rs485 that has the meters and the battery connect to is already set to master.  the rs485 that I am using is connect to the RS485 that is plugged into the same spot the wifi/zigbee would usually be plugged into .

jbuehl, 2016-07-22:
I'm not sure I understand.  Are you talking about 2 physical RS485 interface connections on the inverter?

The way RS485 works is that a number of devices are all connected together in parallel.  Only one of them can be designated as the master and the rest are slaves.

madether, 2016-07-22:
I have one of these http://sol-distribution.com.au/solaredge-rs485-expansion-kit plugged into the wifi/zigbee slot.