jbuehl/solaredge #18: RS485 communication occasionally stalls (deadlock?)

ocaj-nl, 2016-08-08:
Hi jbuehl,

First of all: Great scripts! They allow me to read out my solaredge inverters without having to connect me to the internet and do local logging. 
I have 2 inverters setup to collect data over RS485.

My cabling (I use around 9 meters of cat6-STP-cable) is currently only terminated at the inverter-side. This usually works fine for a few hours in a row, but sometimes I get data corruption or wrong checksums. This is no problem, as the protocol recovers from it by itself.

However every now and then the communication just completely stops. Nothing seems to happen anymore. 

Analyzing your code I think I understand what's going on: The master-thread grants bus-access to the inverter and will then wait for the read-thread to return bus-access. This will work, except in 2 cases where either the 0x0302 (grant) or the 0x039a (resp-grant) are corrupted/lost. 
In the first case the inverter just sits there and waits for the 0x302 before it is allowed to transmit data, while the master-thread waits for the 0x039a to return.
In the second case, the inverter considers the access returned to the master after sending the 0x039a, but the master-thread keeps waiting for the 0x039a.
Both cases result in a deadlock, that will only be fixed by restarting the whole script.

Obviously I need to fix the termination on the bus, but even then an occasional error may occur (my data-cable follows the same physical path as my AC-cabling, so there may be some interferences occurring, even though I use a shielded cable).

As the master-thread is intended to keep running 24/7, may I suggest to add some kind of recovery-mechanism? 
What I see in the communication is that the inverter usually responds within a few seconds after the 0x302 was sent. 
Considering this, it seems like a fair assumption that if the 0x039a was not received after 1-5 minutes that either the 0x302 or the 0x039a was lost. In that case the master-thread could also consider the bus-access to be unlocked and continu polling the inverters. This will keep the data flow running without the need for a manual restart.

Although I have done a fair bit of coding in the past, it's been a while and  I never coded in Python, so I do not have a code-proposal for you. I hope you can add this in a future update to improve the robustness in the RS485-communication.

Thanks!

jbuehl, 2016-08-09:
Thanks for this suggestion.  This is definitely a problem with my code.  I can't work on it right now, but I will try to fix it as soon as I get the time.

d-tamm, 2016-08-10:
I am seeing the same issue, communication stalls deliberately and often after quite a short time (1 to few minutes, sometimes within seconds). The inverter display then shows "Master not found". My cable is unterminated at the computer (RS485-USB adapter). I have tried 82 and 220 Ohm resistors connected to the adapter's connectors but then I got no replies from the inverter any more. @ocaj-nl Do you have any suggestions for the termination at the computer side? How is the termination done at the inverter side? (I have put the termination switch to the ON position.)

jbuehl, 2016-08-10:
I just made a change to semonitor.py and seConf.py to fix the RS485 deadlock problem with a 10 second timer.  If you run with debug flags -vv or greater you should see a message that says "RS485 master ack timeout" when timer expires and polling should continue.  I am unable to test it, so please let me know if it works.

Which RS485-USB adapter are you using, @d-tamm?  When I was testing it, I used a HXSP-2108F with a fairly short cable and no termination at the computer side.  I had a lot of problems until I opened up the adapter and removed a diode that was across the RS485 data lines and then it worked fine.  See the 4th post in this discussion http://forums.whirlpool.net.au/archive/1966464

d-tamm, 2016-08-10:
Thank you, I will test the new version asap!
The adapter I use is this one:
http://www.ebay.de/itm/291465792651
I don't know if it has any specific name or number. But I will have a look if there is a similar diode in it.

ocaj-nl, 2016-08-10:
Thanks for the new version. Unfortunately it almost immediately comes up with:
```
Exception in thread master thread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "semonitor.py", line 100, in masterCommands
    masterTimer = threading.timer(masterMsgTimeout, masterTimerExpire)
AttributeError: 'module' object has no attribute 'timer'
```

jbuehl, 2016-08-10:
That was a typo.  I just corrected it.

ocaj-nl, 2016-08-10:
It is running now and I just noticed in the debug-log the "RS485 master ack timeout"-message and it continues to communicate. So it's looking good, thanks!

I'll let it run now and let you know if it continues to run. If it is still running after a day or two (typically communication stops for me within hours) we can close this issue.

d-tamm, 2016-08-11:
I also see "RS485 master ack timeout" at the same time as the converter displays "Master not found":
```
Aug 11 07:56:39 RS485 master ack timeout 
Aug 11 07:56:44 dataLen:    0000 
Aug 11 07:56:44 dataLenInv: ffff 
Aug 11 07:56:44 sequence:   0326 
Aug 11 07:56:44 source:     fffffffe 
Aug 11 07:56:44 dest:       7e1a0823 
Aug 11 07:56:44 function:   0302 
Aug 11 07:56:44 /dev/ttyUSB0 <-- message: 297 length: 22 
Aug 11 07:56:44
```
But for me, this sequence just repeats and never recovers so long that I get in some data.
However, this may be connected to problems with my RS485 adapter. I opened it up but could not find any similar diode:
[image]

jbuehl, 2016-08-11:
Have you tried reversing the two RS485 wires, @d-tamm?

d-tamm, 2016-08-11:
Unfortunately, it does work for a while and then stop. Reversing the wires did not help. But I got 2 adapters, I will try the other one now.
By the way @jbuehl , did you see my question about your setup for the RPi being a bridge in #11 ?

ocaj-nl, 2016-08-11:
d-tamm: 
Depending on the cable length a 2-wire cable may not be enough. RS485 works with differential balanced line. If systems do not have the same ground, the communication is poort. Typically in RS485 a 3-rd wire is used to connect the GND of all the systems together. 

I have seen and used similar adapters to yours, but this only works for me for very short (1-2 meters max) cables. Try an adapter that allows for 3-wire connections.

ocaj-nl, 2016-08-11:
@jbuehl, the change with the timer on the master-thread seems to keep the communication running. I occasionally see the timeout occur and then communication continues. 
However after a while the inverter no longer responds at all and it will always run in the timeout.
This seems to happen after a checksum error occurs. 

As soon as I abort semonitor and restart the entire process the inverter starts communicating again. 

Could it be that somehow the sequence-numbers get messed up and that causes the inverter to reject any further communication? 

Here's an example of the debug-output (but there seem to be msg-numbers at different levels?):
```
Aug 11 19:51:25 rapi semonitor.py: /home/pi/se/log/20160811_recfile <-- message: 3 length: 22
Aug 11 19:51:25 rapi semonitor.py: /dev/ttyUSB1 --> message: 2 length: 309
Aug 11 19:51:25 rapi semonitor.py: /home/pi/se/log/20160811_recfile <-- message: 4 length: 309
Aug 11 19:51:25 rapi semonitor.py: Checksum error. Expected 0xffff, got 0xd1e1
Aug 11 19:51:25 rapi semonitor.py: data:       1f 01 e0 fe 90 01 d1 30 1a 7f fe ff ff ff 00 05
Aug 11 19:51:25 rapi semonitor.py: data:       80 00 96 0e 25 20 0d 00 84 13 ac 57 c1 0b b7 41
Aug 11 19:51:25 rapi semonitor.py: data:       15 04 2f 00 08 80 00 00 c9 28 20 0d 00 9a 13 ac
Aug 11 19:51:25 rapi semonitor.py: data:       57 ec 0c be f5 96 04 3c 00 08 80 00 ed cb 28 20
Aug 11 19:51:25 rapi semonitor.py: data:       0d 00 d1 13 ac 57 33 0d be f9 66 05 48 00 08 80
Aug 11 19:51:25 rapi semonitor.py: data:       00 ff c7 00 76 c8 28 20 0d 00 1a 14 ac 57 ac 0c
Aug 11 19:51:25 rapi semonitor.py: data:       bb c5 36 05 4b 00 08 10 00 d1 30 9a 7f ae 00 35
Aug 11 19:51:25 rapi semonitor.py: data:       14 ac 57 db 95 00 00 2c 01 00 00 57 af c4 41 4e
Aug 11 19:51:25 rapi semonitor.py: data:       89 dd 42 20 6b 69 41 00 b0 63 43 3d 8a 60 3f 82
Aug 11 19:51:25 rapi semonitor.py: data:       fb 47 42 ff ff 7f ff ff ff 7f ff 00 90 ac 43 ff
Aug 11 19:51:25 rapi semonitor.py: data:       ff 7f ff 20 6b b1 48 f5 28 dc 3b ff ff 7f ff 01
Aug 11 19:51:25 rapi semonitor.py: data:       00 80 3f 04 00 00 00 00 80 bb 45 00 00 c8 42 00
Aug 11 19:51:25 rapi semonitor.py: data:       00 c0 3b ff ff 7f ff ff ff 7f ff 00 80 2b 43 00
Aug 11 19:51:25 rapi semonitor.py: data:       80 41 43 00 00 b9 42 00 00 c8 42 00 00 c8 42 ff
Aug 11 19:51:25 rapi semonitor.py: data:       ff 7f ff 59 8b 05 00 00 00 00 00 00 00 00 00 00
Aug 11 19:51:25 rapi semonitor.py: data:       00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Aug 11 19:51:25 rapi semonitor.py: data:       00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Aug 11 19:51:25 rapi semonitor.py: data:       00 00 00 00 00 ff ff 7f ff 00 00 00 00 80 94 12
Aug 11 19:51:25 rapi semonitor.py: data:       34 56 79 00 00 ff ff d2 4d d1 30 1a 7f fe ff ff
Aug 11 19:51:25 rapi semonitor.py: data:       ff
Aug 11 19:51:55 rapi semonitor.py: RS485 master ack timeout
Aug 11 19:51:55 rapi semonitor.py: /dev/ttyUSB1 <-- message: 3 length: 22
Aug 11 19:51:55 rapi semonitor.py: /home/pi/se/log/20160811_recfile <-- message: 5 length: 22
Aug 11 19:52:25 rapi semonitor.py: RS485 master ack timeout
Aug 11 19:52:30 rapi semonitor.py: /dev/ttyUSB1 <-- message: 4 length: 22
Aug 11 19:52:30 rapi semonitor.py: /home/pi/se/log/20160811_recfile <-- message: 6 length: 22
Aug 11 19:53:00 rapi semonitor.py: RS485 master ack timeout
```
And from then on this repeats indefinitely.

(I'll leave any termination off now in case you need to test further with an unstable connection, the inverter keeps 30 days and after a few attempts I can get all data, so I don't loose anything currently)

jbuehl, 2016-08-11:
Please run this with the -vvv option so I can see what the messages are.

d-tamm, 2016-08-12:
@ocaj-nl - My cable is only 1.5 m long. But I will try your suggestion. I think it should be possible to connect a 3rd wire even with my adapter - USB has a ground pin, and so has the inverter interface. :)
Concerning the Ack timeout, @ocaj-nl 's results look similar to mine, with indefinite repetition. See the output I posted earlier. Restarting the script re-establishes the connection.

ocaj-nl, 2016-08-12:
@jbuehl - Communication now got stuck in the timeout-loop after this corrupted message:

```
Aug 12 11:07:54 rapi semonitor.py: dataLen:    0000
Aug 12 11:07:54 rapi semonitor.py: dataLenInv: ffff
Aug 12 11:07:54 rapi semonitor.py: sequence:   5b03
Aug 12 11:07:54 rapi semonitor.py: source:     fffffffe
Aug 12 11:07:54 rapi semonitor.py: dest:       7f13fe8e
Aug 12 11:07:54 rapi semonitor.py: function:   0302
Aug 12 11:07:54 rapi semonitor.py: /dev/ttyUSB1 <-- message: 1306 length: 22
Aug 12 11:07:54 rapi semonitor.py: 
Aug 12 11:07:54 rapi semonitor.py: /home/pi/se/log/20160812_recfile <-- message: 2602 length: 22
Aug 12 11:07:54 rapi semonitor.py: 
Aug 12 11:07:54 rapi semonitor.py: 
Aug 12 11:07:54 rapi semonitor.py: /dev/ttyUSB1 --> message: 1301 length: 143
Aug 12 11:07:54 rapi semonitor.py: /home/pi/se/log/20160812_recfile <-- message: 2603 length: 143
Aug 12 11:07:54 rapi semonitor.py: 
Aug 12 11:07:54 rapi semonitor.py: dataLen:    0079
Aug 12 11:07:54 rapi semonitor.py: dataLenInv: ff00
Aug 12 11:07:54 rapi semonitor.py: sequence:   feff
Aug 12 11:07:54 rapi semonitor.py: source:     1a30d15a
Aug 12 11:07:54 rapi semonitor.py: dest:       fffffe7f
Aug 12 11:07:54 rapi semonitor.py: function:   9aff
Aug 12 11:07:54 rapi semonitor.py: Data length doesn't match inverse length
Aug 12 11:07:54 rapi semonitor.py: data:       79 00 00 ff ff fe 5a d1 30 1a 7f fe ff ff ff 9a
Aug 12 11:07:54 rapi semonitor.py: data:       03 85 34 12 34 56 79 00 00 ff ff ff 5a 8e fe 13
Aug 12 11:07:54 rapi semonitor.py: data:       7f fe ff ff ff 9a 03 c9 41 12 34 56 79 00 00 ff
Aug 12 11:07:54 rapi semonitor.py: data:       ff 00 5b d1 30 1a 7f fe ff ff ff 9a 03 4f 1c 12
Aug 12 11:07:54 rapi semonitor.py: data:       34 56 79 00 00 ff ff 01 5b 8e fe 13 7f fe ff ff
Aug 12 11:07:54 rapi semonitor.py: data:       ff 9a 03 03 69 12 34 56 79 00 00 ff ff da 03 d1
Aug 12 11:07:54 rapi semonitor.py: data:       30 1a 7f fe ff ff ff 00 05 64 d3 12 34 56 79 00
Aug 12 11:07:54 rapi semonitor.py: data:       00 ff ff 02 5b d1 30 1a 7f fe ff ff ff 9a 03 44
Aug 12 11:07:54 rapi semonitor.py: data:       a4 12 34 56 79 00 00 ff ff 03 5b
Aug 12 11:07:54 rapi semonitor.py: Ignoring this message
Aug 12 11:08:04 rapi semonitor.py: RS485 master ack timeout
Aug 12 11:08:09 rapi semonitor.py: dataLen:    0000
Aug 12 11:08:09 rapi semonitor.py: dataLenInv: ffff
Aug 12 11:08:09 rapi semonitor.py: sequence:   5b04
Aug 12 11:08:09 rapi semonitor.py: source:     fffffffe
Aug 12 11:08:09 rapi semonitor.py: dest:       7f1a30d1
Aug 12 11:08:09 rapi semonitor.py: function:   0302
Aug 12 11:08:09 rapi semonitor.py: /dev/ttyUSB1 <-- message: 1307 length: 22
Aug 12 11:08:09 rapi semonitor.py: 
Aug 12 11:08:09 rapi semonitor.py: /home/pi/se/log/20160812_recfile <-- message: 2604 length: 22
Aug 12 11:08:09 rapi semonitor.py: 
Aug 12 11:08:19 rapi semonitor.py: RS485 master ack timeout
```
In this case it is not a checksum error, but the message length doesn't match (source is also wrong), so the message is ignored. Looking at the data there appears to be some garbage before the actual 0x039a-message.

Funny thing is that after this event not a single message is received anymore from the RS485 (I did grep the logfile on "-->"  and let it try for half an hour or so of 0x302-messages, communication didn't recover). So whatever happens, both inverters get mute after this event, but both inverters start responding again when I restart semonitor.py.

jbuehl, 2016-08-12:
The program was losing its place in the data after the garbled message.  I just made a change to seMsg.py that should fix the problem.

Regarding the RS485 cable, I don't think the ground will make any difference - http://www.planetanalog.com/author.asp?section_id=483&doc_id=562062.  You should be using a single twisted pair for the two lines, but for such a short distance it probably doesn't matter.  Is it possible that the baud rate on your inverter is not the default 115200 @d-tamm?

d-tamm, 2016-08-12:
The ground made indeed no difference. I am using a twisted pair of a 2x2 phone cable, but today I get nothing but timeouts :(

jbuehl, 2016-08-12:
I thought of something you could try @d-tamm to see if there is a problem with your hardware.  If you have two of the RS485 adapters you could connect them together with a pair of wires that are crossed over and plug them both into USB ports on your server.  Then run semonitor.py connected to one serial port and at the same time run semonitor.py connected to the other serial port in another terminal window.

```
python semonitor.py -t 4 -m -s 01234567 /dev/ttyUSB0

python semonitor.py -t 4 -d stdout -vvv /dev/ttyUSB1

```
The second instance of semonitor.py will not respond to the 0302 messages sent by the first instance, but you should see them in the debug output of the second instance.

ocaj-nl, 2016-08-13:
@jbuehl Your last fix seems to work. Communication now restores itself properly once it is running.

The only thing I notice is that sometimes a similar thing happens initially. In those cases, the communication stops almost right away after the first few messages. I checked with -vvvv for such a case and there I see:
```
Aug 13 12:12:08 rapi semonitor.py: opening /dev/ttyUSB1
Aug 13 12:12:08 rapi semonitor.py: starting read thread
Aug 13 12:12:08 rapi semonitor.py: starting master thread
Aug 13 12:12:08 rapi semonitor.py: dataLen:    0000
Aug 13 12:12:08 rapi semonitor.py: dataLenInv: ffff
Aug 13 12:12:08 rapi semonitor.py: sequence:   6ef7
Aug 13 12:12:08 rapi semonitor.py: source:     fffffffe
Aug 13 12:12:08 rapi semonitor.py: dest:       7f1a30d1
Aug 13 12:12:08 rapi semonitor.py: function:   0302
Aug 13 12:12:08 rapi semonitor.py: /dev/ttyUSB1 <-- message: 1 length: 22
Aug 13 12:12:08 rapi semonitor.py: data:       12 34 56 79 00 00 ff ff f7 6e fe ff ff ff d1 30
Aug 13 12:12:08 rapi semonitor.py: data:       1a 7f 02 03 24 e8
Aug 13 12:12:08 rapi semonitor.py: 
Aug 13 12:12:08 rapi semonitor.py: /home/pi/se/log/20160813_recfile <-- message: 1 length: 22
Aug 13 12:12:08 rapi semonitor.py: data:       12 34 56 79 00 00 ff ff f7 6e fe ff ff ff d1 30
Aug 13 12:12:08 rapi semonitor.py: data:       1a 7f 02 03 24 e8
Aug 13 12:12:08 rapi semonitor.py: 
Aug 13 12:12:08 rapi semonitor.py: /home/pi/se/log/20160813_recfile <-- message: 2 length: 4
Aug 13 12:12:08 rapi semonitor.py: data:       12 34 56 79
Aug 13 12:12:08 rapi semonitor.py: 
Aug 13 12:12:18 rapi semonitor.py: RS485 master ack timeout
Aug 13 12:12:18 rapi semonitor.py: dataLen:    0000
Aug 13 12:12:18 rapi semonitor.py: dataLenInv: ffff
Aug 13 12:12:18 rapi semonitor.py: sequence:   6ef8
Aug 13 12:12:18 rapi semonitor.py: source:     fffffffe
Aug 13 12:12:18 rapi semonitor.py: dest:       7f13fe8e
Aug 13 12:12:18 rapi semonitor.py: function:   0302
Aug 13 12:12:18 rapi semonitor.py: /dev/ttyUSB1 <-- message: 2 length: 22
Aug 13 12:12:18 rapi semonitor.py: data:       12 34 56 79 00 00 ff ff f8 6e fe ff ff ff 8e fe
Aug 13 12:12:18 rapi semonitor.py: data:       13 7f 02 03 c6 07
Aug 13 12:12:18 rapi semonitor.py: 
Aug 13 12:12:18 rapi semonitor.py: /home/pi/se/log/20160813_recfile <-- message: 3 length: 22
Aug 13 12:12:18 rapi semonitor.py: data:       12 34 56 79 00 00 ff ff f8 6e fe ff ff ff 8e fe
Aug 13 12:12:18 rapi semonitor.py: data:       13 7f 02 03 c6 07
Aug 13 12:12:18 rapi semonitor.py: 
```
As you can see, right after the first message has been sent to both USB and recfile, the magic header is also sent to the recfile.
Is this initial case something you can explain? Once the system get's passed it (I have not yet identified why it sometimes just works and sometimes, this inital message goes wrong) it will now continu to work with your latest fix. 

@d-tamm : If you don't have 120 ohm-cable and only very short, maybe you could try with the dip-switch in the inverter switched off. I assume that the termination in the inverter assumes a 120 ohm wire. 
For a short wire like yours, termination may not be needed at all.

ocaj-nl, 2016-08-14:
@jbuehl: Thanks for all your effort so far. 

If there is no easy fix, that's fine by me, no rush. Best way to deal with errors is to prevent them, so I'll continu fixing  my termination issues. 
As the inverters have a 30-day memory no data is lost, so even an occasional problem is easy to deal with.

closed 2016-08-14 by ocaj-nl

d-tamm, 2016-09-03:
@jbuehl I have how tried your suggestion of using 2 adapters with crossed wires and 2 instances of semonitor (master and slave, https://github.com/jbuehl/solaredge/issues/18#issuecomment-239540444). There is no output from the slave instance (other than the initial lines until "opening /dev/ttyUSB1"), so I assume that the hardware does not work. This is a bit surprising for me since this seems to be the most sold RS485 adapter on Ebay currently.
By the way, you specified "-s 01234567" for the master instance. Will the slave listen on that address?

jbuehl, 2016-09-03:
The address 01234567 is just a placeholder.  semonitor.py won't respond to any messages regardless of the address, but it should output any data that is received from the other instance.  I should have said you should use the -vvvv command line option.  If you don't see anything, try reversing the wires on one of the adapters.  If you still don't see anything I would be inclined to conclude that the hardware is defective.

d-tamm, 2016-09-03:
This is odd. I had tried with -vvvv previously, but not reversing the wires. Now I have reversed them at one adapter so I connect A to A and B to B, and voilà I get reactions from the slave side.
Now I am wondering what this implies for the usability of the adapters...

jbuehl, 2016-09-03:
Actually that makes sense that they should be connected A-A and B-B since RS485 is a bus and not a point to point connection like RS232.  That always confuses me.  Is the data being interpreted by semonitor.py as valid messages on the slave side?

d-tamm, 2016-09-03:
This is the slave output:
```
Sep 03 19:12:56 /dev/ttyUSB1 --> message: 3 length: 22 
Sep 03 19:12:56 data:       12 34 56 79 00 00 ff ff b1 00 fe ff ff ff 67 45 
Sep 03 19:12:56 data:       23 01 02 03 6b 07 
Sep 03 19:12:56 dataLen:    0000 
Sep 03 19:12:56 dataLenInv: ffff 
Sep 03 19:12:56 sequence:   00b1 
Sep 03 19:12:56 source:     fffffffe 
Sep 03 19:12:56 dest:       01234567 
Sep 03 19:12:56 function:   0302 
```

jbuehl, 2016-09-03:
That's is exactly what I would expect to see.  Your hardware is definitely working.

d-tamm, 2016-09-03:
OK. Thank you for your assistance @jbuehl . So maybe I make another attempt with the adapters in the future. However, my main path right now is the ethernet one.

BoHammil, 2016-09-04:
Has anyone tapped into the zigbee communication?