jbuehl/solaredge #19: No output with RS485

d-tamm, 2016-08-10:
While trying to solve the network connection issues in #11, I have now connected to my SE5k inverter by RS485. Sadly, I am not getting any output. I have terminated the cable at the inverter side by moving up the switch in the inverter. At the computer side, the cable is currently unterminated because I do not really know what resistance my cable has (2x2 twisted wires phone cable, 1.5 m long).
Here is the debug.log I get by running `sudo python semonitor.py -m -s 7E1A0823 -d ../debug.log -vvvv -t 4 /dev/ttyUSB0`:
```
Aug 10 17:44:57 debugEnable: True 
Aug 10 17:44:57 debugFiles: True 
Aug 10 17:44:57 debugMsgs: True 
Aug 10 17:44:57 debugData: True 
Aug 10 17:44:57 debugRaw: True 
Aug 10 17:44:57 debugFileName: ../debug.log 
Aug 10 17:44:57 haltOnException: False 
Aug 10 17:44:57 inFileName: /dev/ttyUSB0 
Aug 10 17:44:57 inputType: 4 
Aug 10 17:44:57 serialDevice: True 
Aug 10 17:44:57     baudRate: 115200 
Aug 10 17:44:57 networkDevice: False 
Aug 10 17:44:57 networkSvcs: False 
Aug 10 17:44:57 following: True 
Aug 10 17:44:57 passiveMode: False 
Aug 10 17:44:57 commandAction: False 
Aug 10 17:44:57 masterMode: True 
Aug 10 17:44:57 slaveAddrs: 7E1A0823 
Aug 10 17:44:57 outFileName: stdout 
Aug 10 17:44:57 append: w 
Aug 10 17:44:57 opening /dev/ttyUSB0 
Aug 10 17:44:57 starting read thread 
Aug 10 17:44:57 starting master thread 
Aug 10 17:44:57 dataLen:    0000 
Aug 10 17:44:57 dataLenInv: ffff 
Aug 10 17:44:57 sequence:   0001 
Aug 10 17:44:57 source:     fffffffe 
Aug 10 17:44:57 dest:       7e1a0823 
Aug 10 17:44:57 function:   0302 
Aug 10 17:44:57 /dev/ttyUSB0 <-- message: 1 length: 22 
Aug 10 17:44:57 data:       12 34 56 79 00 00 ff ff 01 00 fe ff ff ff 23 08 
Aug 10 17:44:57 data:       1a 7e 02 03 fd 76 
Aug 10 17:44:57   
Aug 10 17:44:57   
Aug 10 17:44:57 /dev/ttyUSB0 --> message: 1 length: 22 
Aug 10 17:44:57 data:       12 34 56 79 00 00 ff ff 01 00 23 08 1a 7e fe ff 
Aug 10 17:44:57 data:       ff ff 9a 03 ee 43 
Aug 10 17:44:57 dataLen:    0000 
Aug 10 17:44:57 dataLenInv: ffff 
Aug 10 17:44:57 sequence:   0001 
Aug 10 17:44:57 source:     7e1a0823 
Aug 10 17:44:57 dest:       fffffffe 
Aug 10 17:44:57 function:   039a 
Aug 10 17:45:02 dataLen:    0000 
Aug 10 17:45:02 dataLenInv: ffff 
Aug 10 17:45:02 sequence:   0002 
Aug 10 17:45:02 source:     fffffffe 
Aug 10 17:45:02 dest:       7e1a0823 
Aug 10 17:45:02 function:   0302 
Aug 10 17:45:02 /dev/ttyUSB0 <-- message: 2 length: 22 
Aug 10 17:45:02 data:       12 34 56 79 00 00 ff ff 02 00 fe ff ff ff 23 08 
Aug 10 17:45:02 data:       1a 7e 02 03 f2 32 
Aug 10 17:45:02   
Aug 10 17:45:02   
Aug 10 17:45:02 /dev/ttyUSB0 --> message: 2 length: 22 
Aug 10 17:45:02 data:       12 34 56 79 00 00 ff ff 02 00 23 08 1a 7e fe ff 
Aug 10 17:45:02 data:       ff ff 9a 03 e1 07 
Aug 10 17:45:02 dataLen:    0000 
Aug 10 17:45:02 dataLenInv: ffff 
Aug 10 17:45:02 sequence:   0002 
Aug 10 17:45:02 source:     7e1a0823 
Aug 10 17:45:02 dest:       fffffffe 
Aug 10 17:45:02 function:   039a 
Aug 10 17:45:07 dataLen:    0000 
Aug 10 17:45:07 dataLenInv: ffff 
Aug 10 17:45:07 sequence:   0003 
Aug 10 17:45:07 source:     fffffffe 
[... 484 lines cut ...]
Aug 10 17:47:08 data:       1a 7e 02 03 88 6e 
Aug 10 17:47:08   
Aug 10 17:47:08   
Aug 10 17:47:08 /dev/ttyUSB0 --> message: 27 length: 22 
Aug 10 17:47:08 data:       12 34 56 79 00 00 ff ff 1b 00 23 08 1a 7e fe ff 
Aug 10 17:47:08 data:       ff ff 9a 03 9b 5b 
Aug 10 17:47:08 dataLen:    0000 
Aug 10 17:47:08 dataLenInv: ffff 
Aug 10 17:47:08 sequence:   001b 
Aug 10 17:47:08 source:     7e1a0823 
Aug 10 17:47:08 dest:       fffffffe 
Aug 10 17:47:08 function:   039a 
Aug 10 17:47:13 dataLen:    0000 
Aug 10 17:47:13 dataLenInv: ffff 
Aug 10 17:47:13 sequence:   001c 
Aug 10 17:47:13 source:     fffffffe 
Aug 10 17:47:13 dest:       7e1a0823 
Aug 10 17:47:13 function:   0302 
Aug 10 17:47:13 /dev/ttyUSB0 <-- message: 28 length: 22 
Aug 10 17:47:13 data:       12 34 56 79 00 00 ff ff 1c 00 fe ff ff ff 23 08 
Aug 10 17:47:13 data:       1a 7e 02 03 92 1a 
Aug 10 17:47:13   
```
No other output with production data whatsoever. Anybody knows what functions 039a or 0302 mean? Where may this information be found?
Also, the transmission stalls after some time, however not reproducibly. Could this be due to the missing termination and the stability issue named in #18 ?

ocaj-nl, 2016-08-10:
What you see is the basic RS485-solaredge protocol where the master allows the inverter to communicate. If the inverter has nothing to say, it will return bus-access to the master.

I had this too initially and the most obvious (once you know it...) reason is that you did not configure the inverter to use RS485 as Server, but your inverter is still set to <LAN> as Server. If that is the case, the inverter will not send production infromation over RS485.

You can configure the server in the inverter-menu.

d-tamm, 2016-08-10:
Oh, this is really great! Indeed, the inverter was set to LAN as server (I did not even know that it had such a setting). Now I receive production data. Thanks a lot for the fast reply @ocaj-nl !
By the way, does this mean that I cannot use the RS485 in parallel with SolarEdge's monitoring portal?

ocaj-nl, 2016-08-10:
I personally have no interest in connecting my inverters to Solaredge's monitoring portal, so I intend to do RS485-only.  

I do not know whether the inverter supports reporting to 2 channels, I don't think it does. Check the solaredge-documentation, it has multiple connection scenarios described.

t-beech, 2016-08-17:
As ocaj-nl already pointed out, this should be described [here](http://www.solaredge.com/sites/default/files/se-communication-options-application-note.pdf). 
I would go for Multiple Inverters, RS485 Bus, Wired Ethernet, Parallel Monitoring which states
that the inverters can be set to enable MODBUS in parallel to LAN for communication with the
SolarEdge servers. I have not been able to test, since I am waiting for my pv system to be installed.

d-tamm, 2016-08-17:
Thanks for pointing out the documentation @t-beech . It is not really important for me to send to SolarEdge in parallel, I was just curious. I wonder also if the configuration you propose just adhers to the SunSpec protocol, which would imply that no data on optimizer level would be available on RS485. If I get some time, I might test this out, but it is low priority for me.

closed 2016-08-17 by d-tamm

JacquesWattel, 2016-09-23:
I am having similar issues. I am able to query individual parameters using RS485, but with the above command: sudo python semonitor.py -m -s 7E1A0823 -d ../debug.log -vvvv -t 4 /dev/ttyUSB0 I receive a RS485 master ack timeout error. Any suggestions?

JacquesWattel, 2016-09-23:
Sorted out the problem, I as receiving data now via RS485 to a Raspberry PI. The problem now is that the data is very intermitted. Is the problem perhaps with the Raspberry PI?

d-tamm, 2016-09-23:
Would you mind sharing what you changed in order to get rid of the timeouts?

JacquesWattel, 2016-09-23:
Swapped the RS485 data line around. Silly mistake.

jbuehl, 2016-09-23:
What do you mean that the data is intermittent, @JacquesWattel?  If you look at the debug data you should see a message from the Raspberry Pi every 5 seconds followed by a message from the inverter, but this does not always contain performance data.  This is only sent every minute or so, depending on how much power is being produced.  This is normal behavior.