jbuehl/solaredge #21: Listen on multiple ports in network mode

jbuehl, 2016-08-14:
An inverter will try to connect to port 22221 or 22222 (and possibly others) and may switch from one port to another.  semonitor.py needs to be able to listen on multiple port numbers and accept a connection to any of them.

d-tamm, 2016-08-16:
I can now confirm that my inverter started sending to port 22222 after cold start, and switched to port 22221 during the first night. I will report here what happens the next nights. :)

d-tamm, 2016-09-01:
Having run this system for a while now, I can report that the first switch from port 22222 to 22221 works fine, but the next night, switching back (or maybe further??) does not work. The last data in the attached pcap is for 2016-08-30 20:24.
[capture.pcap.zip](https://github.com/jbuehl/solaredge/files/449890/capture.pcap.zip)
I do not have the skills to interpret the pcap file, but if it helps @jbuehl or others I could again put my Pi in bridge mode and log the traffic with the Solaredge portal a bit longer than last time.

jbuehl, 2016-09-01:
I see that the inverter had been communicating on port 22221 and then closed it, but unlike the last PCAP you sent, it did not attempt to connect on port 22222, but rather tried to connect to port 80.  I think you will need to make another capture in bridge mode to be able to see the messages that are expected on port 80.  Based on what you are seeing, this should occur on the 2nd day of running.  When you do this, you only need to capture the TCP traffic

`tcpdump -U -w capture.pcap host 192.168.2.x and tcp
`

d-tamm, 2016-09-02:
I will set up bridge mode in the next days. In the meantime, I can state that restarting the scripts (I completely rebooted the computer) will resume data acquisition after some hours, i.e. restarting the inverter is not necessary.

d-tamm, 2016-09-05:
Hi @jbuehl. Here is my pcap file for the last days. The communication has switched several times between ports 80, 22222 and 22221 as much as I can see. tcpdump is still running in case you need more data in order to understand what is going on between my inverter and the solaredge portal.
[capture.pcap.zip](https://github.com/jbuehl/solaredge/files/454497/capture.pcap.zip)

jbuehl, 2016-09-05:
It looks like the data being sent to and from port 80 is just regular SolarEdge data, so you could try running 3 instances of semonitor.py that listen on 80, 22221, and 22222.

d-tamm, 2016-09-05:
Hmm, this gets more difficult - on port 80, there is already a web server.

jbuehl, 2016-09-06:
I think the simplest way around that is to run the web server and semonitor on different servers.

d-tamm, 2016-09-06:
Yes, sure that would be the easiest way. But I was hoping to be able to use hardware that is running 24/7 anyway.
By the way, I do not yet really understand why restarting the python scripts recovers the communication if the traffic on port 80 also just is normal SolarEdge data. Restarting the script will not trigger semonitor to send any data on port 80 to recover communication.

d-tamm, 2016-12-02:
I have been running semonitor on ports 22222, 22221 and 80 for quite a while but never got it working for a longer time unfortunately. After restarting the inverter, data will be sent to port 22222 for one day, then to port 22221. However, although there was a server running on port 80, it never got any data. I.e. the output from the port 80 server never gets further than the "waiting for connection" message.
I think it has happened that the data acquisition recovered itself, restarting on port 22222 after some days, but normally I have to reboot the server machine in order to recover.
So something seems to be not "just regular solaredge data" on port 80. Does it make sense to record one more session of data with tcpdump (for let's say 1 week) and post it here?

jbuehl, 2016-12-05:
I don't have the time to look at it.  Maybe you could post it in issue #8.  There are a lot more people who may see it there.

jbuehl, 2017-11-27:
This has now been implemented.  semonitor.py will accept connections from ports 22222, 22221, and 80.  This can be overridden with the -p option which accepts a comma delimited list of ports.

closed 2017-11-27 by jbuehl