jbuehl/solaredge #25: RS485 maximum sequence number ?

ocaj-nl, 2016-09-12:
Hi,

After running for a while my communication over RS485 as master suddenly stopped.
The error message:

```
Exception in thread master thread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/se/solaredge-master/semonitor.py", line 96, in masterCommands
    sendMsg(dataFile, formatMsg(nextSeq(), masterAddr, int(slaveAddr, 16), PROT_CMD_POLESTAR_MASTER_GRANT), recFile)
  File "/home/pi/se/solaredge-master/seMsg.py", line 174, in formatMsg
    checksum = calcCrc(struct.pack(">HLLH", msgSeq, fromAddr, toAddr, function) + data)
error: 'H' format requires 0 <= number <= 65535

```

Looking at the seseq.txt-file I can see that it has reached the maximum value for 16 bits and semonitor does not seem to deal with that situation?

Is it supposed to wrap around?

Can I safely remove the file? As it stores the sequence number on disk I always assumed that the sequence number should survive communication sessions?

Jerrythafast, 2016-09-12:
I believe it does not even need to be sequential. After all, there is no negotiation about the initial sequence number either. Have you ever tried sending random sequence numbers, @jbuehl? I think they are only used to track which messages have been answered.

ocaj-nl, 2016-09-12:
I tried deleting it and everything seems fine, which indeed asks for the question: Do we need to keep a persistent sequence number on disk or could you just as well start with number 1 on every start?

jbuehl, 2016-09-12:
I just fixed that bug in seConf.py.

I implemented the persistent sequence number after having some issues when I was testing and trying to use the same sequence number more than once.  I think @Jerrythafast is probably correct that any random number could be used as long as it is not the same as the one used in the previous message.

ocaj-nl, 2016-09-15:
Hi @jbuehl. Thanks for the fix.

As an experiment I modified nextSeq() in seConf  a bit to never read/write to the seseq.txt-file at all. 
It just starts with 1 everytime I start semonitor and then increments by 1 everytime nextSeq() is called. 

It's been running fine now for a day or two and I have not seen any problems, also after multiple restarts. 

Seems to me that we don't need persistent storage of sequence numbers !

closed 2017-05-10 by jbuehl