jbuehl/solaredge #26: RS485 Questions

ksaihtam, 2016-09-18:
Hello,

i'm new to the solaredge inverters and have some questions about the rs485 protocol.

In my understanding it is possible to get the inverter data over the rs485. Is it also possible to get the optimizer data over the rs485 interface?

Is there a way to set the power reduction over rs485?

Thanks!

jbuehl, 2016-09-18:
Yes, the optimizer performance data can be accessed through RS485.  I don't exactly know what the power reduction is, but there are messages in the RS485 data (0332 and 0333) that people have seen whose name implies they have something to do with that.  See issues #11 and #16.  In order to implement this function in the software, someone with the equipment that does this would need to monitor the RS485 messages that control power reduction and decode the data in them.  I am not aware of anyone who has done this.

ksaihtam, 2016-09-18:
Sounds good. From the manual there are 2 ways to set the power reduction. With external relays/jumpers or from the menu on the device. It would be very nice if there is a way to do that over the rs485 interface for a zero export implementation.

jbuehl, 2016-09-18:
Is it possible to set it from the SolarEdge monitoring portal?  If you can do that, it is possible to capture the messages from the network.

abdullahtahiriyo, 2016-09-25:
I am using rs485. I am trying to figure out what can actually do with it.

 I have some questions:

a) Can a command be send via rs485?, like (i):
``sudo python semonitor.py -c 0012,H0329  **-m** -s 7f14fe9f -d stdout -vvvv -t 4 /dev/ttyS0``
or
``sudo python semonitor.py -c 0012,H0329  -s 7f14fe9f -d stdout -vvvv -t 4 /dev/ttyS0``

or with rs485 we can only just wait for what it comes (ii)?

``sudo python semonitor.py -m -s 7f14fe9f -d stdout -vvvv -t 4 /dev/ttyS0``

why do I ask? To the first command (i), SE inverter does not seem to reply:

``Sep 25 10:02:09 debugEnable: True 
Sep 25 10:02:09 debugFiles: True 
Sep 25 10:02:09 debugMsgs: True 
Sep 25 10:02:09 debugData: True 
Sep 25 10:02:09 debugRaw: True 
Sep 25 10:02:09 debugFileName: stdout 
Sep 25 10:02:09 haltOnException: False 
Sep 25 10:02:09 inFileName: /dev/ttyS0 
Sep 25 10:02:09 inputType: 4 
Sep 25 10:02:09 serialDevice: True 
Sep 25 10:02:09     baudRate: 115200 
Sep 25 10:02:09 networkDevice: False 
Sep 25 10:02:09 sePort: 22222 
Sep 25 10:02:09 networkSvcs: False 
Sep 25 10:02:09 following: True 
Sep 25 10:02:09 passiveMode: False 
Sep 25 10:02:09 commandAction: True 
Sep 25 10:02:09     command: 0012 H0329 
Sep 25 10:02:09 masterMode: True 
Sep 25 10:02:09 slaveAddrs: 7f14fe9f 
Sep 25 10:02:09 outFileName: stdout 
Sep 25 10:02:09 append: w 
Sep 25 10:02:09 opening /dev/ttyS0 
Sep 25 10:02:09 dataLen:    0002 
Sep 25 10:02:09 dataLenInv: fffd 
Sep 25 10:02:09 sequence:   0177 
Sep 25 10:02:09 source:     fffffffe 
Sep 25 10:02:09 dest:       7f14fe9f 
Sep 25 10:02:09 function:   0012 
Sep 25 10:02:09 /dev/ttyS0 <-- message: 1 length: 24 
Sep 25 10:02:09 data:       12 34 56 79 02 00 fd ff 77 01 fe ff ff ff 9f fe 
Sep 25 10:02:09 data:       14 7f 12 00 29 03 42 10 
Sep 25 10:02:09``

to the second command (ii), there is a conversation that according to the source code is about PROT_CMD_POLESTAR_MASTER_GRANT, PROT_RESP_POLESTAR_MASTER_GRANT_ACK and sometimes a PROT_  RESP_PARAMS_SINGLE, see:

``Sep 25 10:06:52 data:       12 34 56 79 00 00 ff ff 7f 01 9f fe 14 7f fe ff 
Sep 25 10:06:52 data:       ff ff 9a 03 62 a9 
Sep 25 10:06:52 dataLen:    0000 
Sep 25 10:06:52 dataLenInv: ffff 
Sep 25 10:06:52 sequence:   017f 
Sep 25 10:06:52 source:     7f14fe9f 
Sep 25 10:06:52 dest:       fffffffe 
Sep 25 10:06:52 function:   039a 
Sep 25 10:06:57 dataLen:    0000 
Sep 25 10:06:57 dataLenInv: ffff 
Sep 25 10:06:57 sequence:   0180 
Sep 25 10:06:57 source:     fffffffe 
Sep 25 10:06:57 dest:       7f14fe9f 
Sep 25 10:06:57 function:   0302 
Sep 25 10:06:57 /dev/ttyS0 <-- message: 9 length: 22 
Sep 25 10:06:57 data:       12 34 56 79 00 00 ff ff 80 01 fe ff ff ff 9f fe 
Sep 25 10:06:57 data:       14 7f 02 03 37 f7 
Sep 25 10:06:57   
Sep 25 10:06:57   
[... 141 lines cut ...]
Sep 25 10:07:32 data:       12 34 56 79 00 00 ff ff 87 01 9f fe 14 7f fe ff 
Sep 25 10:07:32 data:       ff ff 9a 03 4b ca 
Sep 25 10:07:32 dataLen:    0000 
Sep 25 10:07:32 dataLenInv: ffff 
Sep 25 10:07:32 sequence:   0187 
Sep 25 10:07:32 source:     7f14fe9f 
Sep 25 10:07:32 dest:       fffffffe 
Sep 25 10:07:32 function:   039a 
Sep 25 10:07:37 dataLen:    0000 
Sep 25 10:07:37 dataLenInv: ffff 
Sep 25 10:07:37 sequence:   0188 
Sep 25 10:07:37 source:     fffffffe 
Sep 25 10:07:37 dest:       7f14fe9f 
Sep 25 10:07:37 function:   0302 
Sep 25 10:07:37 /dev/ttyS0 <-- message: 17 length: 22 
Sep 25 10:07:37 data:       12 34 56 79 00 00 ff ff 88 01 fe ff ff ff 9f fe 
Sep 25 10:07:37 data:       14 7f 02 03 1d 97 ``

No JSON at all is output.

b) Currently I had SERVER configured to LAN. **Do I need to change the SERVER to RS485-1 in the inverter?** I think I did this yesterday without success, but after 16 hours in a row, I better catch some sleep.

c) Any hints why I am not getting the JSON out?

Thanks in advance!!

jbuehl, 2016-09-25:
See my response to issue #8 to answer your first question.  You have to configure the inverter to use the SE protocol to a RS485 server for it to send the data to that interface.  The data sent to that interface shouldn't be encrypted so you won't need the encryption key if you are going to do it that way.

abdullahtahiriyo, 2016-09-25:
@jbuehl 

Thanks! I've got the key, so I can move to network configuration...

ericbuehl, 2018-07-12:
looks like this has been taken care of.  please re-open if I overlooked anything

closed 2018-07-12 by ericbuehl