jbuehl/solaredge #28: Passively monitoring traffic sent via ZigBee adapter gives "Data length too big" error

latenighthacks, 2016-09-28:
Hi Everyone, 

First off, thanks to everyone for your efforts in getting this up and running. 

I seem to be having trouble decoding packets sent from my inverter to the SolarEdge monitoring portal. My inverter is setup to send data via the SolarEdge ZigBee adapter, which then forwards it via Ethernet to my router and out to the Internet. 

I captured the packet stream using this command, 

`/usr/sbin/tcpdump -i eth2 -U -w - tcp and host 192.168.0.23 | tee capture.pcap | python solaredge/seextract.py | python solaredge/semonitor.py -k ~/7f13ae9d.key -d stdout -vvv` 

192.168.0.23 is the IP address of the SolarEdge ZigBee adapter. At first, this resulted in a lot of messages about the decryption key not being available, i.e., 

Sep 27 19:53:02 debugEnable: True 
Sep 27 19:53:02 debugFiles: True 
Sep 27 19:53:02 debugMsgs: True 
Sep 27 19:53:02 debugData: True 
Sep 27 19:53:02 debugRaw: False 
Sep 27 19:53:02 debugFileName: stdout 
Sep 27 19:53:02 haltOnException: False 
Sep 27 19:53:02 inFileName: stdin 
Sep 27 19:53:02 serialDevice: False 
Sep 27 19:53:02 networkDevice: False 
Sep 27 19:53:02 sePort: 22222 
Sep 27 19:53:02 networkSvcs: False 
Sep 27 19:53:02 following: True 
Sep 27 19:53:02 passiveMode: True 
Sep 27 19:53:02 commandAction: False 
Sep 27 19:53:02 masterMode: False 
Sep 27 19:53:02 outFileName: stdout 
Sep 27 19:53:02 append: w 
Sep 27 19:53:02 keyFileName: /home/matt/7f13ae9d.key 
Sep 27 19:53:02 key: 10de3256af65bb46a7c0853f7f38e06e 
Sep 27 19:53:02 opening stdin 
Sep 27 19:53:02 
Sep 27 19:53:02 <stdin> --> message: 2 length: 66 
Sep 27 19:53:02 dataLen:    002c 
Sep 27 19:53:02 dataLenInv: ffd3 
Sep 27 19:53:02 sequence:   010e 
Sep 27 19:53:02 source:     6710fcd2 
Sep 27 19:53:02 dest:       fffffffe 
Sep 27 19:53:02 function:   003d 
Sep 27 19:53:02 Decryption key not yet available 
Sep 27 19:53:02 Ignoring this message

Note that the timestamps are all the same because I ran the captured packets through semonitor.py just now. Eventually, the decryption key was obtained, 

Sep 27 19:53:02 <stdin> --> message: 489 length: 56 
Sep 27 19:53:02 dataLen:    0022 
Sep 27 19:53:02 dataLenInv: ffdd 
Sep 27 19:53:02 sequence:   00d2 
Sep 27 19:53:02 source:     6710fcd2 
Sep 27 19:53:02 dest:       fffffffe 
Sep 27 19:53:02 function:   0503 
Sep 27 19:53:02 Creating decryption object with key 10de3256af65bb46a7c0853f7f38e06e

However, no packets can be decrypted. I always get this same error, 

Sep 27 19:53:02 <stdin> --> message: 502 length: 66 
Sep 27 19:53:02 dataLen:    002c 
Sep 27 19:53:02 dataLenInv: ffd3 
Sep 27 19:53:02 sequence:   01c2 
Sep 27 19:53:02 source:     fffffffd 
Sep 27 19:53:02 dest:       ffffffff 
Sep 27 19:53:02 function:   003d 
Sep 27 19:53:02 Decrypting message 
Sep 27 19:53:02 dataLen:    46f4 
Sep 27 19:53:02 dataLenInv: e5b2 
Sep 27 19:53:02 sequence:   31b2 
Sep 27 19:53:02 source:     532ec11d 
Sep 27 19:53:02 dest:       03fb24c7 
Sep 27 19:53:02 function:   956e 
Sep 27 19:53:02 Data length is too big for the message 
Sep 27 19:53:02 data:       f4 46 b2 e5 b2 31 1d c1 2e 53 c7 24 fb 03 6e 95 
Sep 27 19:53:02 data:       1c 21 
Sep 27 19:53:02 Ignoring this message 

I obtained the inverter's encryption key using the USB method detailed in another issue (thanks for that tip!), so I think that's good. Does anyone have any idea about where I went wrong? In case it's helpful, please find attached the packet [capture](https://github.com/jbuehl/solaredge/files/497057/capture.zip).

jbuehl, 2016-09-28:
This is a problem with seextract.py.  Others have had this problem.  Usually it is fixed by filtering on  tcp and host, but it looks like you are already doing that.  All that seextract.py does is to extract a tcp stream from the pcap file and there are a number of utilities available that do that https://wiki.wireshark.org/Tools.  I don't want to spend any time fixing seextract.py since there are so many alternatives.  If you find one that works, please post it here.

latenighthacks, 2016-09-29:
Hi jbuehl, thanks for replying! 

I tried extracting the stream using tcpflow, 

`tcpflow -r capture.pcap`

which resulted in 8 files, i.e., 4 bidirectional streams between the ZigBee gateway and the SolarEdge servers: 

-rw-r--r-- 1 matt matt    23874 Sep 27 11:47 192.168.000.023.01308-217.068.149.119.22222
-rw-r--r-- 1 matt matt      330 Sep 27 14:13 192.168.000.023.01310-217.068.149.114.22222
-rw-r--r-- 1 matt matt    17813 Sep 27 14:17 192.168.000.023.01312-217.068.149.117.22222
-rw-r--r-- 1 matt matt      396 Sep 27 16:08 192.168.000.023.01314-217.068.149.126.22222
-rw-r--r-- 1 matt matt    20642 Sep 27 16:13 192.168.000.023.01316-217.068.149.121.22222
-rw-r--r-- 1 matt matt     8498 Sep 27 18:19 192.168.000.023.01891-217.068.149.115.22222
-rw-r--r-- 1 matt matt     6498 Sep 27 18:19 217.068.149.115.22222-192.168.000.023.01891
-rw-r--r-- 1 matt matt     7920 Sep 27 14:17 217.068.149.117.22222-192.168.000.023.01312
-rw-r--r-- 1 matt matt    10626 Sep 27 11:47 217.068.149.119.22222-192.168.000.023.01308
-rw-r--r-- 1 matt matt     9240 Sep 27 16:13 217.068.149.121.22222-192.168.000.023.01316

A couple of interesting points here: Unlike the inverters, the ZigBee gateway used a destination port of 22222 only (though, this wasn't even a full 24 hour capture, so perhaps I didn't give it enough time to switch). But, what's really interesting is that none of the IP addresses that it contacted were prod.solaredge.com (217.68.149.116), prod2.solaredge.com (217.68.144.150), or prod3.solaredge.com (217.68.152.88). The IP addresses appear to not have reverse DNS records, so I can't identify their hostnames. Nevertheless, they look like they're in the same IP block as prod.solaredge.com (217.68.149.xxx).

At this point, I'm wondering if the ZigBee gateway uses a different protocol than the Ethernet gateway..., with the exception of the message from which the decryption object is created, which appears to be interpreted correctly. 

What do you think? Any ideas on where I could start poking around to figure out if inverters logging over Ethernet use the same on-the-wire format as inverters logging via ZigBee?

jbuehl, 2016-09-30:
Splitting the TCP streams into separate files for each direction and each connection is OK, but it will complicate things for you.  Better to have everything for both directions in one file.  You should still be able to pipe the ones that contain data sent from the inverter to SolarEdge (the first 6 in your list) and see performance data.  I would be surprised if they use a different format for this interface since they use the same one for the others.

When you refer to "summary messages" what do you mean?  If you are talking about the JSON messages output by semonitor.py, then those will contain optimizer data in the "optimizers {}" section if it is available which will occur while the sun is out and they are producing power.  If you don't see anything, it could be that you have a type of optimizer that semonitor.py doesn't know how to decode.  Try running with the -d stdout -vvv options and see if there are any "Unknown device" messages.

latenighthacks, 2016-10-02:
I tried cat'ing the packet dumps generated by `tcpflow` into semonitor.py: 

`cat 192.168.000.023.01312-217.068.149.117.22222 192.* 217.* | python solaredge/semonitor.py -k ~/7f13ae9d.key -vvvv -d stdout` 

I put 192.168.000.023.01312-217.068.149.117.22222 first because it contains the message that semonitor.py uses to build the decryption object. However, this yielded the same result as piping the result of secapture.py into semonitor.py: Before the decryption object is built, packets can't be decrypted. After it's built, semonitor.py reports that packets either have a "Data length too big for this message" or "Data length doesn't match inverse length". For example, 

Oct 01 16:26:05 <stdin> --> message: 492 length: 206 
Oct 01 16:26:05 dataLen:    002c 
Oct 01 16:26:05 dataLenInv: ffd3 
Oct 01 16:26:05 sequence:   01bc 
Oct 01 16:26:05 source:     6710fcd2 
Oct 01 16:26:05 dest:       fffffffe 
Oct 01 16:26:05 function:   003d 
Oct 01 16:26:05 Discarding 140 extra bytes 
Oct 01 16:26:05 data:       35 e2 ea 57 ed b8 02 00 36 00 00 00 36 00 00 00 
Oct 01 16:26:05 data:       00 27 02 10 fc d2 b8 97 5a 47 06 48 08 00 45 00 
Oct 01 16:26:05 data:       00 28 28 df 40 00 6a 06 b8 77 d9 44 95 75 c0 a8 
Oct 01 16:26:05 data:       00 17 56 ce 05 20 65 d1 d7 45 01 48 95 29 50 10 
Oct 01 16:26:05 data:       ff 2e 51 b5 00 00 35 e2 ea 57 c8 65 0a 00 78 00 
Oct 01 16:26:05 data:       00 00 78 00 00 00 00 27 02 10 fc d2 b8 97 5a 47 
Oct 01 16:26:05 data:       06 48 08 00 45 00 00 6a 2a 87 40 00 6a 06 b6 8d 
Oct 01 16:26:05 data:       d9 44 95 75 c0 a8 00 17 56 ce 05 20 65 d1 d7 45 
Oct 01 16:26:05 data:       01 48 95 29 50 18 ff 2e 8b 2e 00 00 
Oct 01 16:26:05 Decrypting message 
Oct 01 16:26:05 dataLen:    fc89 
Oct 01 16:26:05 dataLenInv: a347 
Oct 01 16:26:05 sequence:   e1ef 
Oct 01 16:26:05 source:     4d8f9f4a 
Oct 01 16:26:05 dest:       866d566e 
Oct 01 16:26:05 function:   2986 
Oct 01 16:26:05 Data length is too big for the message 
Oct 01 16:26:05 data:       89 fc 47 a3 ef e1 4a 9f 8f 4d 6e 56 6d 86 86 29 
Oct 01 16:26:05 data:       5d 2e 
Oct 01 16:26:05 Ignoring this message 

Curiously, regardless of the message length reported on the first line (i.e., 206 in this case), dataLen (second line) is often -- but not always -- reported to be 002c. Does that make sense? 

I would also be surprised if they used a different format for this interface. It looks like every 30 s, or so, two messages of 66 bytes each are sent, e.g., 

Oct 01 16:52:03 <stdin> --> message: 162 length: 66 
Oct 01 16:52:03 dataLen:    002c 
Oct 01 16:52:03 dataLenInv: ffd3 
Oct 01 16:52:03 sequence:   0185 
Oct 01 16:52:03 source:     6710fcd2 
Oct 01 16:52:03 dest:       fffffffe 
Oct 01 16:52:03 function:   003d 
Oct 01 16:52:03 Decryption key not yet available 
Oct 01 16:52:03 Ignoring this message 
Oct 01 16:52:04   
Oct 01 16:52:04 <stdin> --> message: 163 length: 66 
Oct 01 16:52:04 dataLen:    002c 
Oct 01 16:52:04 dataLenInv: ffd3 
Oct 01 16:52:04 sequence:   1bcf 
Oct 01 16:52:04 source:     fffffffd 
Oct 01 16:52:04 dest:       ffffffff 
Oct 01 16:52:04 function:   003d 
Oct 01 16:52:04 Decryption key not yet available 
Oct 01 16:52:04 Ignoring this message 

and every 5 minutes, a longer message of about 500 bytes is sent, 

Oct 01 16:43:03 <stdin> --> message: 132 length: 492
Oct 01 16:43:03 dataLen:    01d6
Oct 01 16:43:03 dataLenInv: fe29
Oct 01 16:43:03 sequence:   017b
Oct 01 16:43:03 source:     6710fcd2
Oct 01 16:43:03 dest:       fffffffe
Oct 01 16:43:03 function:   003d
Oct 01 16:43:03 Decryption key not yet available
Oct 01 16:43:03 Ignoring this message

I suspect the 66 byte messages are the instantaneous power output and the ~500 byte messages contain the optimizer data and other details. At any rate, even if it's not the same protocol, it looks like all the same data are getting sent. 

By "summary message", I meant one of these: 

{"inverters": {"7F13AE9D": {"Uptime": 64563, "Temp": 46.95014572143555, "Vac": 242.890625, "Eac": 238.22169494628906, "Interval": 300, "Iac": 11.941561698913574, "Eday": 8710.6162109375, "Etot": 3449227.0, "Vdc": 368.75, "Pac": 2810.0, "Time": "13:31:45", "Date": "2016-08-19", "Freq": 59.98700714111328, "ID": "7F13AE9D", "Pmax": 6000.0}}, "events": {}, "optimizers": {"10749B92": {"Uptime": 24486, "Temp": 44.0, "Vmod": 27.875, "Imod": 7.05625, "Eday": 587.75, "Vopt": 25.25, "Time": "13:30:19", "Date": "2016-08-19", "Inverter": 0, "ID": "10749B92"}, "10742869": {"Uptime": 24564, "Temp": 48.0, "Vmod": 27.25, "Imod": 7.206250000000001, "Eday": 591.75, "Vopt": 24.375, "Time": "13:31:23", "Date": "2016-08-19", "Inverter": 0, "ID": "10742869"}, "10734F8F": {"Uptime": 24137, "Temp": 48.0, "Vmod": 26.875, "Imod": 7.21875, "Eday": 586.5, "Vopt": 24.5, "Time": "13:32:00", "Date": "2016-08-19", "Inverter": 0, "ID": "10734F8F"}, "10732762": {"Uptime": 24782, "Temp": 48.0, "Vmod": 27.125, "Imod": 7.1312500000000005, "Eday": 597.25, "Vopt": 24.5, "Time": "13:34:45", "Date": "2016-08-19", "Inverter": 0, "ID": "10732762"}, "10743904": {"Uptime": 24676, "Temp": 44.0, "Vmod": 27.375, "Imod": 7.1937500000000005, "Eday": 600.75, "Vopt": 24.375, "Time": "13:33:08", "Date": "2016-08-19", "Inverter": 0, "ID": "10743904"}, "10747A5B": {"Uptime": 24551, "Temp": 48.0, "Vmod": 26.875, "Imod": 7.125, "Eday": 583.0, "Vopt": 24.625, "Time": "13:31:05", "Date": "2016-08-19", "Inverter": 0, "ID": "10747A5B"}, "10742CD4": {"Uptime": 23493, "Temp": 42.0, "Vmod": 27.875, "Imod": 7.15625, "Eday": 611.0, [... cut ...]
So, it looks like my system has a type of power optimizer that semonitor.py knows about and occasionally, a message is sent from the inverter that can be decoded. Nevertheless, the majority of messages sent over the ZigBee gateway appear to be sent in some different format that cannot yet be decoded. Does that roughly square with your take on things? 

If so, could you maybe suggest how I could try to determine what this different format could be? 

Also, apologies for the wall of text. I'm trying to include enough details to keep things clear.

jbuehl, 2016-10-02:
It's difficult for me to figure out what is happening.  It looks like the messages aren't being decrypted properly, at least some of the time.  Can you post your capture.pcap file here so I can look at it?

The 66 byte messages every 30 seconds and the 500 byte messages every 5 minutes are normal.  The inverter sends a 0500 message with zero length data every 30 seconds and this is encrypted into a 003d message that is 66 bytes long.  Every 5 minutes it sends its data as well as the data for any optimizers that have reported in a 0500 message that is encrypted into a 003d message.  That is the format of the SE interface.

latenighthacks, 2016-10-02:
Please see my first post -- at the bottom it has the packet capture and the key I got from my inverter. Thanks for taking a look! Please let me know if there's any way I can help.

jbuehl, 2016-10-04:
Sorry, I misunderstood your first post where you said *none* of the messages were being decrypted.  I was thinking that you were getting some data and getting errors on some of them, which is what happens with the seextract.py problem.  You posted some performance data above.  How did you get that?  I am getting the same result which probably means that the key isn't good.  It may be that there was a problem in how it was extracted.  Can you run the semonitor.py command that extracts the key and post the output, but don't pipe it into sekey.py.

`python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f101234 -t 2  /dev/ttyUSB0 -d stdout -vvv`

latenighthacks, 2016-10-05:
You're right, and I should've been clearer: _Most_ messages don't come through, but every once in a while, some performance data do come through (apologies for that!) I don't know how I got it, though -- I just setup tcpdump as above and waited. 

Here is the output of the command that you asked for: 

matt@lighthearted:~/projects$ python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f13ae9d -t 2 /dev/ttyUSB0 -d stdout -vvvv
{"data": {"type": 0, "value": 1446174224}, "command": 18, "response": 144, "sequence": 1}
{"data": {"type": 0, "value": 1186686383}, "command": 18, "response": 144, "sequence": 2}
{"data": {"type": 0, "value": 1065730215}, "command": 18, "response": 144, "sequence": 3}
{"data": {"type": 0, "value": 1860188287}, "command": 18, "response": 144, "sequence": 4}

I substituted my inverter's ID as the -s argument. I tried it also with -s 7f101234, but nothing was returned: 

matt@lighthearted:~/projects$ python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f101234 -t 2 /dev/ttyUSB0 -d stdout -vvv
^CTraceback (most recent call last):
  File "solaredge/semonitor.py", line 149, in <module>
    doCommands(dataFile, commands, recFile)
  File "solaredge/semonitor.py", line 124, in doCommands
    msg = readMsg(dataFile, recFile)
  File "/home/matt/projects/solaredge/seMsg.py", line 68, in readMsg
    msg = readBytes(inFile, magicLen+msgHdrLen)
  File "/home/matt/projects/solaredge/seMsg.py", line 97, in readBytes
    inBuf = inFile.read(length)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 446, in read
    ready,_,_ = select.select([self.fd],[],[], self._timeout)
KeyboardInterrupt

At first, the command wasn't returning anything, so I fumbled around on the inverter's panel and set <Server> to RS232. It seems that once I did that, the inverter started responding. Nevertheless, the key that I got tonight is the same as the first key that I extracted, namely, 

10de3256af65bb46a7c0853f7f38e06e

Any idea what could be going on?

Jerrythafast, 2016-10-05:
This is probably coincidence, but the first 32 bits of your key look remarkably similar to a UNIX time stamp (2015-10-30T03:03:44Z).

jbuehl, 2016-10-05:
It looks like you are getting the values stored in the registers 0239-023c, @latenighthacks, which should contain the key.  The -s argument should specify your inverter ID, 7f101234 is just an example.  I suppose it is possible that your inverter has a version of the firmware that stores the encryption key in different registers, but I'm not sure how to find this out.

latenighthacks, 2016-10-06:
Good catch, @Jerrythafast, it does indeed look like a timestamp. I would also write it off as a coincidence right now, however. 

@jbuehl, any idea how to determine what firmware version my inverter has? I couldn't find anything scrolling through the menu. 

For now, I'll setup a laptop to act as a logging server. Let's see if my inverter really is behaving differently or if it's something to do with the ZigBee gateway.

Jerrythafast, 2016-10-06:
@latenighthacks, you should look for the 'CPU version'. It is visible on one of the pages you get when repeatedly pressing the button at the bottom of the inverter.

latenighthacks, 2016-10-07:
Thanks @Jerrythafast. The 'CPU version' of my inverter is 0003.1449. 

I hooked up a laptop to the inverter via ethernet and semonitor.py had no trouble decoding the packets! Every 300 seconds, the inverter sent a big chunk of data that included details on the optimizers. 

However, I'm not sure if the data were ever encrypted, or not. I setup the packet capture with the command 

`sudo python solaredge/semonitor.py -k 7f13ae9d.key -t n -n eth0 -d stdout -r capture -o stdout -v`, 

but I don't think I saw a decryption object getting created. Running the capture file through semonitor.py without the inverter key, 

`python solaredge/semonitor.py -d stdout capture`

results in the same output as running it through semonitor.py with the inverter key, 

`python solaredge/semonitor.py -k 7f13ae9d.key -d stdout capture`, 

so, I'm guessing the decryption key was not used. 

Just to confirming existing, known behavior: My inverter stopped sending data via the ZigBee gateway when I switched "Server" to "LAN" inside the inverter's control panel. 

I guess I will next setup a way for the inverter to talk to SolarEdge over wired ethernet and then try to sniff that to see if it produces a different result than trying to sniff the data that went out over the ZigBee gateway. 

Incidentally, in today's log, there were 3 kinds of unidentified devices: 0x0200, 0x0017 and 0x0018.

Jerrythafast, 2016-10-07:
@latenighthacks, that's nothing out of the ordinary then. Looks like the data is indeed unencrypted, but if you got it connected to the internet now this will change somewhere between now and tomorrow. You'll need your key to decrypt data sent after the switch (which is initiated by the SolarEdge server).

latenighthacks, 2016-10-13:
@Jerrythafast, @jbuehl, just a quick update: I hooked up my inverter via the LAN port and have been successfully reading performance data from it for several days, now. I am passively sniffing the packet stream via tcpdump and passing it to seextract.py which passes it to semonitor.py. Sometimes there are some unrecognized devices, sometimes not. 

It might interest you that my inverter _has not turned on encryption_. I don't know why, though.

moesizlag, 2017-01-06:
@jbuehl First thanks for all your effort you put into those tools :-)

I'm currently capturing the tcp communication between SolarEdge and the inverter by mirroring the network port on my switch towards a linux box where the data should be decoded.
Unfortunately I run into the same "Decryption key not yet available" issue, although I read out the key and verified it.
As one idea was this could be connected to different firmware versions keeping the key in different registers, those are the versions on the inverter: 
CPU Version: 2.1053
DSP1 Version: 1.210 (0x4A5BB09D)
DSP2 Version: 1.34

ericbuehl, 2018-07-12:
@latenighthacks, @moesizlag sounds like things are working -- closing

closed 2018-07-12 by ericbuehl