jbuehl/solaredge #29: Converting to database

jehojakim, 2016-09-28:
Hi jbuehl,

Though I haven't started to fetch the inverter key on our SE5000, let alone fetch data, I know I would like the data to store in a database, rather than in spreadsheet files. So I would like to write a python script that inserts the data in  mysql-database.
I would like to ask you if I could 'hijjack' one of your programs to turn it into a database-insertion program.
If that is indeed OK with you, best candidate to use seems to me se2csv.py, but maybe you have a different opinion on this.

To temper expectations, it will take me some time, because I have a rather busy life, and no programming experience in python.

If, again, OK with you, would it be possible to send me a json-file containing some data I can use to experiment? I have a 1-phase inverter, but of course I would like it to be possible to use data from a 3-phase inverter.

jbuehl, 2016-09-28:
Having a database insertion program would be a great addition this project.  There used to be an option to write to a mysql database, but I removed it due to lack of interest.  You can still find the code in this github project under the "seconvert" tag.  The program is seconvert2.py and it is in the "deprecated" folder.  I agree that the best way to do it would be to modify one of the se2*.py programs by adding the mysql code from seconvert2.py to create se2mysql.py.  Let me know if you have trouble finding it.

Here is a sample json file from my SE5000

[20160928000002.json.txt](https://github.com/jbuehl/solaredge/files/498311/20160928000002.json.txt)

jehojakim, 2016-09-28:
OK, thank you!
I cannot find the 'deprecated' folder in https://github.com/jbuehl/solaredge - am I looking at the wrong place?

jbuehl, 2016-09-28:
You have to select the "seconvert" tag using the "Branch:" menu on the left side and then that folder which is in an earlier version of the project will show up.  But here it is as well as the sql that defines the database.

[seconvert2.py.txt](https://github.com/jbuehl/solaredge/files/498769/seconvert2.py.txt)
[solar.sql.txt](https://github.com/jbuehl/solaredge/files/498770/solar.sql.txt)

jehojakim, 2016-09-28:
Ah, I see. I'm new to Github...
Thanks! No time tomorrow, will look into it this weekend.

foto-andreas, 2016-09-30:
I will also work on some database tools in the next weeks (java), but I think I will write earlier data into the db. This gives me more options for later calculations and views. Also I have seen some problems today with se2state: During shutdown of the pv system in the evening there are some panels which are reactivated for some seconds (uptime<100) the Eday value is reset to zero. In the json file from se2state it looks like these panels where inactive the whole day:
`"20184072": {"Uptime": 38840, "Temp": 18.0, "Vmod": 28.75, "Imod": 0.18125000000000002, "Eday": 710.5, "Vopt": 26.75, "Time": "16:57:53", "Date": "2016-09-30", "Inverter": 0`
`"20184072":{"Uptime": 122, "Temp": 18.0, "Vmod": 32.25, "Imod": 0.00625, "Eday": 0.0, "Vopt": 38.625, "Time": "17:07:24"...`

Currently I am saving data into files from semonitor spoofing on the ethernet and the sunspec values from the RS485 which gives me the realtime values from the inverter. So I have the option to import them into the database later.

jbuehl, 2016-10-01:
I wasn't suggesting that you should use the output of se2state.py.  Better to use the output of semonitor.py.  se2state.py does set some values to zero at the time of shutdown, but not Eday.  It that is being reset to zero then it is done by the inverter or the optimizer itself.

foto-andreas, 2016-10-01:
Thats correct. "Eday" should better be named "Euptime" ;-). If this should be taken into account, se2state needs some more logic than the simple dictionary merge. While my db import will take the values from semonitor, I can also implement some logic there to come over this. Curently I live with the Eday values.

jbuehl, 2016-10-01:
The intent of the se2*.py programs was that they are bridges between the output of semonitor.py and various data sinks and that they should be entirely independent of each other.  se2state.py is the only one that I use and it is intended to only maintain a file containing the most recent value of all the inverter and optimizer data fields, so the value of Eday is whatever the inverter or optimizer last reported.  I think the daily totals are being set to zero when the inverter goes into night mode because it considers the "day" to be over at that time.

abdullahtahiriyo, 2016-10-10:
I am using se2state.py myself. Probably is not the best way, but I did not pay much more attention when I started logging for a better way. There is a lot of room for improvement (removing writes to hard-disk, which would kill the SDCard of a raspi pretty soon), but it was fast to code (I am realizing that probably just moving the timestamp detection up would save a lot of writes).

Just if someone regards it as useful, I am running this via crontab periodically to update an InfluxDB v1.0 database and also sending the measurements to domoticz:

``
#!/bin/bash

# Extract Inverter and Optimizers from se2state JSON file
#inverter
cat /home/admin/solaredge/data/solar.json | jq -r '.inverters | map([ (."ID") , (."Uptime"|tostring), (."Temp"|tostring), (."Vac"|tostring), (."Eac"|tostring), (."Interval"|tostring), (."Iac"|tostring), (."Etot"|tostring), (."Vdc"|tostring), (."Pac"|tostring), (."Time"), (."Date"), (."Freq"|tostring), (."Eday"|tostring) ]| join(",")) | join("\n") ' > /home/admin/solaredge/data/inverters.csv

#optimizers
cat /home/admin/solaredge/data/solar.json  | jq -r '.optimizers | map([ (."ID") , (."Uptime"|tostring), (."Temp"|tostring), (."Vmod"|tostring), (."Imod"|tostring), (."Vopt"|tostring), (."Time"), (."Date"), (."Inverter"|tostring), (."Eday"|tostring) ]| join(",")) | join("\n") ' > /home/admin/solaredge/data/optimizers.csv

# Keep track of last insertion and interrupt if same timestamp
OLDTIMESTAMP=0;

if [ -e /home/admin/solaredge/data/lasttimestamp ]
then
        OLDTIMESTAMP=`cat /home/admin/solaredge/data/lasttimestamp`
fi

cat /home/admin/solaredge/data/inverters.csv | sed 's|:| |g' | sed 's|-| |g' | gawk -F',' '{print mktime($12 " " $11 "CET") }' > /home/admin/solaredge/data/lasttimestamp

#echo -n "Measurement datetime: "

cat /home/admin/solaredge/data/inverters.csv | gawk -F',' '{print $12 " " $11 }'

NEWTIMESTAMP=`cat /home/admin/solaredge/data/lasttimestamp`

if [ "$NEWTIMESTAMP" == "$OLDTIMESTAMP" ]
then
        #repeated measurement;
        #echo "Repeated...exiting..."
        exit 1
fi

# InfluxDB line protocol insertion

# Inverters
cat /home/admin/solaredge/data/inverters.csv | sed 's|:| |g' | sed 's|-| |g' | awk -F',' '{print "SolarEdgeInverters,deviceid=" $1 " Temp=" $3 ",Vac=" $4 ",Eac=" $5 ",Iac=" $7 ",Etot=" $8 ",Vdc=" $9 ",Pac=" $10 ",Freq=" $13 ",Eday=" $14 " " mktime($12 " " $11 "CET")"000000000" }' \
|  curl -i -XPOST 'http://localhost:8086/write?db=domoticz' --data-binary @-

# Optimizers
cat /home/admin/solaredge/data/optimizers.csv | sed 's|:| |g' | sed 's|-| |g' | awk -F',' '{print "SolarEdgeOptimizers,deviceid=" $1 " Temp=" $3 ",Vmod=" $4 ",Imod=" $5 ",Vopt=" $6 ",Eday=" $10 " " mktime($8 " " $7 "CET")"000000000" }' \
|  curl -i -XPOST 'http://localhost:8086/write?db=domoticz' --data-binary @-

# Domoticz inverter energy/power only

domoip=127.0.0.1
domoport=8080
solaredge_idx=11

SVALUE=`cat /home/admin/solaredge/data/inverters.csv | awk -F',' '{print $10";"$8}'`

curl -s -i -H "Accept: application/json" "http://$domoip:$domoport/json.htm?type=command&param=udevice&idx=$solaredge_idx&nvalue=0&svalue=$SVALUE"
``
my daemon for solaredge is:

``
#!/bin/bash -

ROOTDIR=/home/admin/solaredge/data/
PREFIX=

sudo tcpdump -i eth0 -s 65535 -w - 'host 192.168.120.100' | python /home/admin/solaredge/semonitor.py -vv | tee $ROOTDIR$PREFIX`date +%Y%m%d%H%M%S`.json | python /home/admin/solaredge/se2state.py -o /home/admin/solaredge/data/solar.json
``
I use grafana with the InfluxDB data, "Eday" is set to zero everyday, so the last value reported in a day is that days produced energy. "Etot" has the lifetime energy of the inverter, so from the installation date.

sjthespian, 2017-05-18:
I've been playing with ideas for getting my SolarEdge data into InfluxDB, and finally came up with the following. I started with the ideas from abdullahtahiriyo and ended up with:

- Sniff the traffic between the SolarEdge gateway and SolarCity using tcpdump on my firewall
- Forward that traffic to netcat which pipes the data into a custom python script that will run seextract and semonitor for me, and pass the output on to InfluxDB. 
- Display all this into a very work-in-progress grafana dashboard.

 This looks something like:
```
nc -l -p 42424  | ./se2influxdb.py --type n --log data/se2influxdb-`date +%Y%m%d%H%M`.log --influxdb dbuser:dbpass@localhost/solar &
ssh -f fw -- -i sk1 -U -w - 'tcp and host solaredge.example.com'
```
(the ssh command looks funky because I have a shared key on the firewall that only allows this user to run tcpdump -- that command also starts the netcat to port 42424 on my host)

The python script is a bit more involved. I'll post it here and do a PR when I get it cleaned up and more presentable. I'm open to suggestions or improvements.
```python
#!/usr/bin/python

import os
import sys
import getopt
import fileinput
import subprocess
import json
import requests
from requests.auth import HTTPBasicAuth
from datetime import datetime,tzinfo,timedelta
import time
import signal
import atexit

def usage(msg):
  if msg:
    print msg
  print "usage: %s [--file filename] [--type n|e|m] [--log file] [--influxdb [user:password@]host[:port]]/database" % sys.argv[0]
  print """
    If a filename is specified, data in that file will be parsed and sent to influxdb
    If a filename or port number is not specified, input will be taken from stdin
    The type parameter specifies the data type as follows:
        n: raw network stream, will be fed to seextract and semonitor
        e: output from seextract, will be fed to semonitor
        m: output from semonitor, will be parsed as json
    The default is a json file from semonitor.py
    """
  sys.exit(1)

# Write data to influx db, expects data to already be formatted for writing
def influxDBwrite(influxdb, data):
  dbuser = None
  dbpass = None
  dbhost = None
  dbport = 8086
  dbname = 'solar'
  if '@' in influxdb:
    (authinfo, hostinfo) = influxdb.split('@')
    (dbuser, dbpass) = authinfo.split(':')
  else:
    hostinfo = influxdb
  if '/' in hostinfo:
    (hostinfo, dbname) = hostinfo.split('/')
  if ':' in hostinfo:
    (dbhost, dbport) = hostinfo.split(':')
  else:
    dbhost = hostinfo

  if debug:
    print "http://%s:%s/write?db=%s\n\t%s" % (dbhost, dbport, dbname, data)
  else:
    if dbuser:
      requests.post("http://%s:%s/write?db=%s" % (dbhost, dbport, dbname), data=data, auth=HTTPBasicAuth(dbuser, dbpass))
    else:
[... 117 lines cut ...]
  datafh = semon.stdout

# Read data, calling appropriate function for each type
datatypes = {
  'inverters':  processInverter,
  'events':     processEvent,
  'optimizers': processOptimizer
}
if logfile:
  try:
    logfilefh = open(logfile, 'w')
  except (OSError, IOError) as e:
    print "Unable to create logfile %s: %s" % (logfile, e)

for line in iter(datafh.readline, ''):
  data = json.loads(line)
  if logfile:
    logfilefh.write(line)
    logfilefh.write("\n")
  for type in data:
    if type in datatypes:
      datatypes[type](influxdb, data[type])
    else:
      print "ERROR: Unable to process %s data! (%s)" % (type, data[type])
```

ericbuehl, 2018-07-12:
@sjthespian please open a PR so we can comment and get your code integrated!

lekkerduidelijk, 2019-04-12:
@sjthespian @ericbuehl Looking into the InfluxDB setup as well. 
Is the script / PR available yet?