jbuehl/solaredge #31: Home Assistant component

TeraHz, 2016-10-29:
Hi,

It would be great if this can be converted to a home assistant component. It seems like a great automation project to integrate with. It is also a python project so that surely helps. 

The se2mqtt script can be used currently via MQTT -> HA, but it isn't as nice.

Just a suggestion!

https://home-assistant.io/
https://home-assistant.io/developers/creating_components/

jbuehl, 2016-10-31:
I agree that it would be a good thing to have a se2hass program.  If someone writes it I will add it to the project.

MarcHagen, 2019-09-20:
Leaving these here:
https://www.home-assistant.io/components/solaredge_local/
https://www.home-assistant.io/components/solaredge/