jbuehl/solaredge #32: ID of optimizers parseId

Arvoreen75, 2017-01-01:
I got my Solaredge connected to the grid some days ago and I am now collecting and analyzing the data, and I was wondering why the ID's of some of my optimizers showed up wrong.

I was seeing 1030a179 instead of 10a0a179 and so on. 

Looking at he code I found in seData.py
`# remove the extra bit that is sometimes set in a device ID and upcase the letters
def parseId(seId):
return ("%x" % (seId & 0xff7fffff)).upper()`
changing the 7 in the mask to f and the ID's are right but this would probably be to simple. There is defineatly a reason for masking one bit out that I do ignore.

Jerrythafast, 2017-01-01:
Are you sure you are seeing 1030a179? The code you mentioned would change this into 1020a179.

If that is what it's doing: that's clearly a bug. It should only do that if the device is an inverter.

More details: The inverter consists of multiple internal devices; the ID you are usually seeing is that of the communication board. There is another device that also sometimes sends output, the ID of that device is the commboard's ID or'ed with 0x00100000. The code you mentioned is responsible for calculating the commboard's ID from that other one.

Arvoreen75, 2017-01-01:
Sorry my mistake.

I see 1020A179 and I have 10A0A179.

Seem to be one of the Newer Optimizers as the code parseNewOptData is used.

closed 2017-11-24 by jbuehl