jbuehl/solaredge #33: How to

lagreat, 2017-02-14:
This may be nothing new to all the contributors who have diligently created and updated this repository. However, I ended up creating this query as I cannot find a way to send PM to some of the dontributors.
I just got my SE7600 installed and in production. I'm reading a lot on this one and other forums so I can ask my installer to give me access I need to see what I want. BTW, I paid for my system outright.
On reading this for over a month I am getting this as a starting point.
- Only one interface out of LAN, RS, Zigbee, Cellular can be active at a time. My install has cellular so there is no way I know to snoop data.
- I want to use RPi3 and connect RS485 to collect data however the server is showing cellular so how will I get the data while still letting the portal updates to go thru? What type of connection must I make; any schematics will help immensely
- @abdullatahiroyo has all 3 connection per one of his post RS485, RS232 & LAN - how does he get data out of three simultaneously if first statement is true 
- what are my options with cellular in the mix

Once I'm set I plan on writing a how to guide for someone new to this project. my first hurdle is to get this off the ground.

kurt-rhee, 2021-02-20:
I would also be interested in snooping data from cellular.

kurt-rhee, 2021-02-20:
@lagreat did you ever make any progress?