jbuehl/solaredge #34: Use of ports 22222 and 22221

GaryAitken, 2017-02-23:
Hi all,
New to this forum but greatly appreciate all the work and sleuthing done.
I have an se10000 (CAN version for brutal cold) now connected via ethernet; was using wi-fi periodically to upload stored daily data.  It used to hold 5-6 weeks but when I replaced the comm board the storage capacity went down to ~3 weeks.  That may be the result of the switch to encryption.  Plan is to download historic data and write to mysql db, then permanently snoop; will use tools here for that.
My question is regards to ports 22222 and 22221.  Does anyone know if these are actually necessary?  They appear to be the preferred ports for uploading data, with port 80 as a fallback.    Looking at a live tcpdump of my installation it appears if they are open there is no traffic on port 80, but I got the impression from reading here that many are seeing the data go up on port 80.  What I see is only data on 22222 and nothing on 22221 or 80.  If I block 22222, I then see data only on 22221.  It does a number of retries and delays including fresh dns lookups before switching.  If I then reopen 22222 and block 22221 it does not switch back, but switches to 80.  It looks like the preferred order is 22222, 22221, and then 80; but I'm wondering if either of 22222 or 22221 is actually required.

jbuehl, 2017-02-24:
Solaredge inverters used to connect to port 22222 a few years ago for data logging, but recently they seem to be connecting to ports 22221 and 80 (see issue #21).  Since I don't have a SE10000 I have no way of knowing how that one works.  Good luck.

Jerrythafast, 2017-02-24:
I would guess the reduced storage period is due to a newer version of the firmware on your new commboard. I believe in more recent versions of the firmware more data was added to the end of the inverter telemetry records. The data is not stored in an encrypted format as far as I know.

As for the different ports, I think it makes sense what you found out about the inverter falling back to 22221 and ultimately port 80. I myself solved the issue of different ports by parsing any TCP stream coming from a SolarEdge network device, which are identified by their MAC address. (Beware, the Wi-Fi module may not have a SolarEdge MAC address.)

GaryAitken, 2017-02-25:
Thanks; additional data per record makes sense as a source of reduced time-span for backed-up (non-transmitted) data.  At the same time, in terms of encryption, I was thinking of the additional memory required to hold the encryption algorithms, not encrypted data.  Could be a combination of both.
I use fixed (local net space) IP addrs, so it's easy to detect the inverter packets; MAC addrs aren't an issue in that case.
My curiosity regarding 22222 and 22221 was thinking one could block them with no ill-effect.  If all three ports or actually used in a redundant manner, then any two could be blocked.  Depending on how Solaredge implements their server(s), it might (or might not) effect the reliability of outbound connections to upload data to solaredge; but other than that there would be no effect.

jbuehl, 2017-11-27:
Closed with issue #21.

closed 2017-11-27 by jbuehl