jbuehl/solaredge #35: se2cvs.py: KeyError 'Vac' with 3 phase inverter

ethpi, 2017-04-22:
Hi,

many thanks for all your fantastic work. 

I started to log the data of my solaredge-4k inverter (3 phase) and 18 optimizers over RS485. `semonitor.py` works fine. Now I'd like to write the data in a mySQL database. Therefore I tried to use `se2csv.py`. The file `solar.json` contains one line of data:

`{"inverters": {"7E180E50": {"Uptime": 75002, "Freq3": 49.9990234375, "Freq2": 49.9982795715332, "Freq1": 49.999778747558594, "data29": 78.0, "data21": 4286578687, "data22": 4286578687, "data23": 4286578687, "GndFrR": 6900.03125, "data31": 3169517568, "EdayDC": 4286578687, "Vac3": 232.9375, "Eac": 43.30878448486328, "Vac1": 232.53125, "ID": "7E180E50", "Date": "2017-04-22", "CosPhi3": 1.0, "CosPhi2": 1.0, "IoutDC": -0.03955078125, "Temp": 27.459383010864258, "Interval": 300, "Iac3": 0.5551663637161255, "CosPhi1": 1.0, "Iac1": 0.5501027703285217, "Time": "17:09:52", "Idc": 4286578687, "Edc": 4286578687, "Iac2": 0.561946451663971, "Eday": 4351.13818359375, "Etot": 9455554.0, "Vdc": 747.125, "Vac2": 234.03125, "mode": 5, "Irdc": 0.0026874998584389687}}, "optimizers": {"20050055": {"Uptime": 37295, "Temp": 14.0, "Vmod": 29.375, "Imod": 0.7250000000000001, "ID": "20050055", "Vopt": 40.625, "Time": "17:11:19", "Date": "2017-04-22", "Inverter": 0, "Eday": 268.75}, "20050893": {"Uptime": 37438, "Temp": 14.0, "Vmod": 26.75, "Imod": 1.09375, "ID": "20050893", "Vopt": 43.625, "Time": "17:08:30", "Date": "2017-04-22", "Inverter": 0, "Eday": 306.25}, "20050BED": {"Uptime": 37524, "Temp": 14.0, "Vmod": 27.625, "Imod": 1.05, "ID": "20050BED", "Vopt": 44.125, "Time": "17:08:14", "Date": "2017-04-22", "Inverter": 0, "Eday": 299.75}, "200506C0": {"Uptime": 37095, "Temp": 14.0, "Vmod": 31.375, "Imod": 1.56875, "ID": "200506C0", "Vopt": 43.125, "Time": "17:05:02", "Date": "2017-04-22", "Inverter": 0, "Eday": 288.25}, "20050D1B": {"Uptime": 37698, "Temp": 14.0, "Vmod": 26.75, "Imod": 1.25625, "ID": "20050D1B", "Vopt": 43.25, "Time": "17:07:05", "Date": "2017-04-22", "Inverter": 0, "Eday": 306.0}, "200507F9": {"Uptime": 37334, "Temp": 14.0, "Vmod": 30.375, "Imod": 1.4125, "ID": "200507F9", "Vopt": 41.875, "Time": "17:05:39", "Date": "2017-04-22", "Inverter": 0, "Eday": 282.5}, "2005089E": {"Uptime": 37388, "Temp": 14.0, "Vmod": 26.625, "Imod": 1.06875, "ID": "2005089E", "Vopt": 43.75, "Time": "17:07:53", "Date": "2017-04-22", "Inverter": 0, "Eday": 295.5}, "20050540": {"Uptime": 37458, "Temp": 14.0, "Vmod": 29.75, "Imod": 0.7875000000000001, "ID": "20050540", "Vopt": 40.875, "Time": "17:10:21", "Date": "2017-04-22", "Inverter": 0, "Eday": 272.0}, "20050D54": {"Uptime": 37529, "Temp": 14.0, "Vmod": 26.875, "Imod": 1.04375, "ID": "20050D54", "Vopt": 43.625, "Time": "17:08:33", "Date": "2017-04-22", "Inverter": 0, "Eday": 294.75}, "200507EC": {"Uptime": 37288, "Temp": 14.0, "Vmod": 29.625, "Imod": 1.16875, "ID": "200507EC", "Vopt": 42.375, "Time": "17:06:36", "Date": "2017-04-22", "Inverter": 0, "Eday": 281.75}, "200507DA": {"Uptime": 37697, "Temp": 14.0, "Vmod": 26.75, "Imod": 1.14375, "ID": "200507DA", "Vopt": 43.125, "Time": "17:07:26", "Date": "2017-04-22", "Inverter": 0, "Eday": 301.75}, "20050945": {"Uptime": 37606, "Temp": 14.0, "Vmod": 30.375, "Imod": 0.76875, "ID": "20050945", "Vopt": 41.5, "Time": "17:10:41", "Date": "2017-04-22", "Inverter": 0, "Eday": 283.75}, "20050467": {"Uptime": 37293, "Temp": 14.0, "Vmod": 31.0, "Imod": 0.84375, "ID": "20050467", "Vopt": 42.125, "Time": "17:08:49", "Date": "2017-04-22", "Inverter": 0, "Eday": 275.0}, "20050516": {"Uptime": 37757, "Temp": 14.0, "Vmod": 30.375, "Imod": 0.8500000000000001, "ID": "20050516", "Vopt": 41.75, "Time": "17:10:15", "Date": "2017-04-22", "Inverter": 0, "Eday": 296.5}, "20050525": {"Uptime": 37367, "Temp": 14.0, "Vmod": 31.125, "Imod": 1.3875000000000002, "ID": "20050525", "Vopt": 40.625, "Time": "17:05:17", "Date": "2017-04-22", "Inverter": 0, "Eday": 256.0}, "20050526": {"Uptime": 37255, "Temp": 14.0, "Vmod": 28.875, "Imod": 0.9375, "ID": "20050526", "Vopt": 41.0, "Time": "17:08:17", "Date": "2017-04-22", "Inverter": 0, "Eday": 281.0}, "200508CB": {"Uptime": 37161, "Temp": 14.0, "Vmod": 30.0, "Imod": 0.8187500000000001, "ID": "200508CB", "Vopt": 38.125, "Time": "17:09:01", "Date": "2017-04-22", "Inverter": 0, "Eday": 259.5}, "20050D48": {"Uptime": 37067, "Temp": 14.0, "Vmod": 30.5, "Imod": 0.875, "ID": "20050D48", "Vopt": 40.125, "Time": "17:07:40", "Date": "2017-04-22", "Inverter": 0, "Eday": 262.25}}}
`

But the commandline 

`solaredge/se2csv.py -i inverter.csv -o optimizer.csv solar.json`

produces the errormessage:

```
Traceback (most recent call last):
  File "solaredge/se2csv.py", line 103, in <module>
    writeData(json.loads(jsonStr), invFile, optFile)
  File "solaredge/se2csv.py", line 64, in writeData
    invSeq = writeDevData(invFile, invOutFmt, msgDict["inverters"][seId], invItems, invSeq)
  File "solaredge/se2csv.py", line 74, in writeDevData
    outMsg = delim.join([(outFmt[i] % devDict[devItems[i]]) for i in range(len(devItems))])
KeyError: 'Vac'
```

If I remove `-i inverter.csv` from the commandline above, `se2csv.py` works well.

It seems, that `se2csv.py` doesn't notice, that I use a 3 phase inverter. Can anybody help me to convert my json-files into csv-files?

Many thanks, Jan

--------------------------
By the way: In` seDataParams.py` in line 95 it is assumed, that data29 means the power limit. I can confirm this, because my inverter is limited to 78%. This is exactly the value in `solar.json` above.

jbuehl, 2017-04-23:
Yes, se2csv.py was written before support for 3 phase inverters was added to semonitor.py.  If you want to load data into a mySQL database, why go through the step of converting it to a csv file first?    You could create a se2mysql.py program that would load the data directly from the json file.  An earlier version of what is now called semonitor.py had the option to do that, but I removed it since no one appeared to be using it and I wanted to decouple the data collection from storage.

You can access the old version in the github project if you go to the drop down menu that says "Branch: master" and select "Tag: seconvert".  The file is called seconvert2.py and it is in the deprecated folder.

ethpi, 2017-04-23:
Thanks for your answer

You're right, it's better to make it directly. The problem is, I've never written a python script. But it's a good reason to learn python.

By changing the assignments in `seDataParams.py` for `invInFmt`, `invIdx`, `invItems`and `invOutFmt` to the respective variables for 3 phase inverters, `se2csv.py` works for my inverter correctly.

I'll search for suitable code in `seconvert2.py`. In case of success, I will leave a message here.

Jan

ethpi, 2017-04-30:
Hi,

I adapted `se2csv.py` to `se2mysql.py`. I'm not familar with python, but it works for me. If anybody is interested: [se2mysql.zip](https://github.com/jbuehl/solaredge/files/966941/se2mysql.zip)

closed 2020-12-30 by ethpi