jbuehl/solaredge #40: Checksum error on 0x003d messages

Arvoreen75, 2017-05-31:
Hello,

As long as the messages were sent unencrypted everything went well, I got an awful lot of data :-)
Now since the data is encoded I am able to decode the data but the granularity of the data went down.
digging deeper into this I see now a lot of Checksum errors.

I tried a bit around to find the cause but with no luck.

Is someone seeing the same thing?

```
May 31 09:54:01 <stdin> --> message: 1352 length: 582
May 31 09:54:01 data:       12 34 56 79 24 02 db fd 8b 02 f4 21 1a 7e fe ff
May 31 09:54:01 data:       ff ff 3d 00 5b fb 96 06 43 67 cb f7 4b 67 14 99
May 31 09:54:01 data:       3b 65 10 86 1f 26 c2 09 37 c7 99 34 eb 58 b2 d6
May 31 09:54:01 data:       69 7d 89 87 ce 82 bc 0c 5d 73 7a f9 ea 3e 07 10
May 31 09:54:01 data:       97 6e c4 db bd 62 7c b0 ce d3 4a 93 4f 6a 15 1a
May 31 09:54:01 data:       01 0c 95 d8 83 d6 79 42 8d b9 46 5f 66 5c d6 b8
May 31 09:54:01 data:       a4 84 fb 79 67 e4 b6 91 e3 38 cd f9 74 6c 51 a8
May 31 09:54:01 data:       70 83 44 42 11 6d b2 69 4b 40 9d 8b ca 5d 48 07
May 31 09:54:01 data:       71 42 1a 0d 66 fa 20 c9 a8 07 d7 1c 15 b6 e0 28
May 31 09:54:01 data:       73 a1 bf a4 b2 17 67 c0 46 11 bc 72 de 3e 8f ec
May 31 09:54:01 data:       57 fe 04 d2 c0 0c ac 23 58 95 9b d7 b9 db 4c 0c
May 31 09:54:01 data:       49 3b 57 2e ab 64 87 ee ed b4 e7 52 74 d7 d4 59
May 31 09:54:01 data:       d0 ca c9 7e 27 ca 55 0e fe 2d 8d 3c 6e a5 52 05
May 31 09:54:01 data:       3c 0d 43 cc 33 2d b3 9e f8 ee 00 9b a3 cd 75 b9
May 31 09:54:01 data:       7c d6 79 78 c6 94 ae 4a 56 54 57 a4 d3 2f 48 46
May 31 09:54:01 data:       e8 98 a0 5e bb 11 63 dc 2a 2f 3f 50 1d 72 a6 a9
May 31 09:54:01 data:       3f f9 fe e2 8b de 6d 85 87 df af d7 8f 00 37 7d
May 31 09:54:01 data:       75 c2 e9 79 a8 ba 96 aa de 3e 80 e0 f0 bc 84 87
May 31 09:54:01 data:       19 16 75 0d e0 4e 5e 9b 0f 30 1d 18 1e 22 7d e8
May 31 09:54:01 data:       e7 31 bf 82 c6 8a c3 00 fe 0b b2 9d a6 fb 47 6c
May 31 09:54:01 data:       a2 f7 a4 c6 fa 5c 3b fe 49 5d 15 68 aa 2b c0 77
May 31 09:54:01 data:       5e ff b8 ac cd 2e b4 49 4b 6f cf a9 5e 02 4c d6
May 31 09:54:01 data:       a1 b9 3d ac 6b f3 dc 25 8f 14 06 6f 18 58 07 5e
May 31 09:54:01 data:       d2 4a 62 a8 63 75 74 c1 41 9b de 21 3f 0a 8c 3e
May 31 09:54:01 data:       81 1b ad 9b 3c 7b ba 64 78 64 58 e0 48 ad 2d c7
May 31 09:54:01 data:       62 3f 16 4d c4 31 b0 34 8f 23 ea 76 0c c3 0a 40
May 31 09:54:01 data:       f8 9d 01 f0 82 ed e2 41 93 c8 5a f6 b5 03 6f be
May 31 09:54:01 data:       81 75 a9 d9 93 0f 5d 16 b3 40 48 83 fb 5c 69 00
May 31 09:54:01 data:       fa ba 53 b5 42 51 1f bd b0 af 2c ea 3a d2 81 f9
May 31 09:54:01 data:       b8 7a 13 cc 6e 9a 4e 04 28 72 e7 3b f3 48 4e 82
May 31 09:54:01 data:       2d ba d2 c4 b9 ba 20 bf c4 1b 54 31 9b f7 95 ed
May 31 09:54:01 data:       f7 00 84 65 a1 30 93 2a a9 b0 ce 44 44 ab a4 09
May 31 09:54:01 data:       ca 7b f5 c4 18 79 8c 90 35 84 32 ae 32 ad 64 6e
May 31 09:54:01 data:       8a 7d 03 6a 71 5f f0 c5 b4 ed 00 00 d1 8a ba 14
May 31 09:54:01 data:       44 10 66 40 0a b0 35 4b b6 91 aa 73 0f 61 2b 0f
May 31 09:54:01 data:       f4 31 f3 d1 8b 25 9d e0 31 28 3a b9 26 55 c8 e5
May 31 09:54:01 data:       00 00 d4 47 2f 3b
May 31 09:54:01 dataLen:    0224
May 31 09:54:01 dataLenInv: fddb
May 31 09:54:01 sequence:   028b
May 31 09:54:01 source:     7e1a21f4
May 31 09:54:01 dest:       fffffffe
May 31 09:54:01 function:   003d
May 31 09:54:01 Discarding 12 extra bytes
May 31 09:54:01 data:       3a b9 26 55 c8 e5 00 00 d4 47 2f 3b
May 31 09:54:01 Checksum error. Expected 0x2831, got 0x2edc
May 31 09:54:01 data:       24 02 db fd 8b 02 f4 21 1a 7e fe ff ff ff 3d 00
May 31 09:54:01 data:       5b fb 96 06 43 67 cb f7 4b 67 14 99 3b 65 10 86
May 31 09:54:01 data:       1f 26 c2 09 37 c7 99 34 eb 58 b2 d6 69 7d 89 87
May 31 09:54:01 data:       ce 82 bc 0c 5d 73 7a f9 ea 3e 07 10 97 6e c4 db
May 31 09:54:01 data:       bd 62 7c b0 ce d3 4a 93 4f 6a 15 1a 01 0c 95 d8
May 31 09:54:01 data:       83 d6 79 42 8d b9 46 5f 66 5c d6 b8 a4 84 fb 79
May 31 09:54:01 data:       67 e4 b6 91 e3 38 cd f9 74 6c 51 a8 70 83 44 42
May 31 09:54:01 data:       11 6d b2 69 4b 40 9d 8b ca 5d 48 07 71 42 1a 0d
May 31 09:54:01 data:       66 fa 20 c9 a8 07 d7 1c 15 b6 e0 28 73 a1 bf a4
May 31 09:54:01 data:       b2 17 67 c0 46 11 bc 72 de 3e 8f ec 57 fe 04 d2
May 31 09:54:01 data:       c0 0c ac 23 58 95 9b d7 b9 db 4c 0c 49 3b 57 2e
May 31 09:54:01 data:       ab 64 87 ee ed b4 e7 52 74 d7 d4 59 d0 ca c9 7e
May 31 09:54:01 data:       27 ca 55 0e fe 2d 8d 3c 6e a5 52 05 3c 0d 43 cc
May 31 09:54:01 data:       33 2d b3 9e f8 ee 00 9b a3 cd 75 b9 7c d6 79 78
May 31 09:54:01 data:       c6 94 ae 4a 56 54 57 a4 d3 2f 48 46 e8 98 a0 5e
May 31 09:54:01 data:       bb 11 63 dc 2a 2f 3f 50 1d 72 a6 a9 3f f9 fe e2
May 31 09:54:01 data:       8b de 6d 85 87 df af d7 8f 00 37 7d 75 c2 e9 79
May 31 09:54:01 data:       a8 ba 96 aa de 3e 80 e0 f0 bc 84 87 19 16 75 0d
May 31 09:54:01 data:       e0 4e 5e 9b 0f 30 1d 18 1e 22 7d e8 e7 31 bf 82
May 31 09:54:01 data:       c6 8a c3 00 fe 0b b2 9d a6 fb 47 6c a2 f7 a4 c6
May 31 09:54:01 data:       fa 5c 3b fe 49 5d 15 68 aa 2b c0 77 5e ff b8 ac
May 31 09:54:01 data:       cd 2e b4 49 4b 6f cf a9 5e 02 4c d6 a1 b9 3d ac
May 31 09:54:01 data:       6b f3 dc 25 8f 14 06 6f 18 58 07 5e d2 4a 62 a8
May 31 09:54:01 data:       63 75 74 c1 41 9b de 21 3f 0a 8c 3e 81 1b ad 9b
May 31 09:54:01 data:       3c 7b ba 64 78 64 58 e0 48 ad 2d c7 62 3f 16 4d
May 31 09:54:01 data:       c4 31 b0 34 8f 23 ea 76 0c c3 0a 40 f8 9d 01 f0
May 31 09:54:01 data:       82 ed e2 41 93 c8 5a f6 b5 03 6f be 81 75 a9 d9
May 31 09:54:01 data:       93 0f 5d 16 b3 40 48 83 fb 5c 69 00 fa ba 53 b5
May 31 09:54:01 data:       42 51 1f bd b0 af 2c ea 3a d2 81 f9 b8 7a 13 cc
May 31 09:54:01 data:       6e 9a 4e 04 28 72 e7 3b f3 48 4e 82 2d ba d2 c4
May 31 09:54:01 data:       b9 ba 20 bf c4 1b 54 31 9b f7 95 ed f7 00 84 65
May 31 09:54:01 data:       a1 30 93 2a a9 b0 ce 44 44 ab a4 09 ca 7b f5 c4
May 31 09:54:01 data:       18 79 8c 90 35 84 32 ae 32 ad 64 6e 8a 7d 03 6a
May 31 09:54:01 data:       71 5f f0 c5 b4 ed 00 00 d1 8a ba 14 44 10 66 40
May 31 09:54:01 data:       0a b0 35 4b b6 91 aa 73 0f 61 2b 0f f4 31 f3 d1
May 31 09:54:01 data:       8b 25 9d e0 31 28 3a b9 26 55 c8 e5 00 00 d4 47
May 31 09:54:01 data:       2f 3b
May 31 09:54:01 Ignoring this message
May 31 09:54:01
May 31 09:54:01 <stdin> --> message: 1353 length: 66
May 31 09:54:01 data:       12 34 56 79 2c 00 d3 ff de 0c fd ff ff ff ff ff
May 31 09:54:01 data:       ff ff 3d 00 cd e8 e9 79 1f 48 41 d9 c5 06 5a b1
May 31 09:54:01 data:       36 82 85 1d 3c 46 93 cf c5 35 5b 14 97 71 c9 9c
May 31 09:54:01 data:       20 fc d0 94 45 7b 42 32 19 7b 67 a5 fc f9 6d 1e
May 31 09:54:01 data:       b6 60
May 31 09:54:01 dataLen:    002c
May 31 09:54:01 dataLenInv: ffd3
May 31 09:54:01 sequence:   0cde
May 31 09:54:01 source:     fffffffd
May 31 09:54:01 dest:       ffffffff
May 31 09:54:01 function:   003d
May 31 09:54:01 Decrypting message
May 31 09:54:01 dataLen:    0000
May 31 09:54:01 dataLenInv: ffff
May 31 09:54:01 sequence:   033c
May 31 09:54:01 source:     fffffffe
May 31 09:54:01 dest:       7e1a21f4
May 31 09:54:01 function:   0080
May 31 09:54:01
May 31 09:54:01 <stdin> --> message: 1354 length: 6
May 31 09:54:01 data:       12 34 56 79 2c 00
May 31 09:54:01 Ignoring this message
May 31 09:54:01
May 31 09:54:01 <stdin> --> message: 1355 length: 118
May 31 09:54:01 data:       12 34 56 79 60 00 9f ff 8c 02 f4 21 1a 7e fe ff
May 31 09:54:01 data:       ff ff 3d 00 cb a2 97 d4 43 a4 c1 e6 10 1c 0f a5
May 31 09:54:01 data:       42 69 e1 b3 8e c0 3d 0f e1 1c de fd 93 9e 3c 73
May 31 09:54:01 data:       76 fa 28 ae 1b e3 75 06 77 69 16 e0 fa ce 73 6e
May 31 09:54:01 data:       29 b7 7f 58 2b a6 67 c9 ef 3e f7 d7 9f 78 52 06
May 31 09:54:01 data:       af 46 fa 60 19 c1 e1 3a 12 a0 e3 39 85 33 5a 94
May 31 09:54:01 data:       d5 81 b3 33 81 fa 23 2b 4a e6 5c 84 86 05 0c f5
May 31 09:54:01 data:       37 a5 fc 29 23 5b
May 31 09:54:01 dataLen:    0060
May 31 09:54:01 dataLenInv: ff9f
May 31 09:54:01 sequence:   028c
May 31 09:54:01 source:     7e1a21f4
May 31 09:54:01 dest:       fffffffe
May 31 09:54:01 function:   003d
May 31 09:54:01 Decrypting message
May 31 09:54:01 dataLen:    0034
May 31 09:54:01 dataLenInv: ffcb
May 31 09:54:01 sequence:   ffff
May 31 09:54:01 source:     7e1a21f4
May 31 09:54:01 dest:       fffffffe
May 31 09:54:01 function:   03c2
May 31 09:54:01 Unknown function 0x03c2
May 31 09:54:01
```

justinhuntgc, 2017-05-31:
I also had this issue, I could only decode 1/10 packets or so successfully.   If it's of value I can set this up again on the weekend and capture some logs.

Arvoreen75, 2017-05-31:
Might be of interest. I have an Se12.5K Inverter with Optimizers.
1/10 of data could be correct. So did you find a solution?

justinhuntgc, 2017-05-31:
I abandoned it as I thought it may be related to comm board issues I was having but I dont beleive that to be the case as the fix was a firmware upgrade.  

I will try and setup the monitoring again and catch some logs over the weekend.  I am reasonably sure that it should be logging a data packet every 5 minutes, however I only get valid packets a few times an hour.  It is possible to export from the solaredge portal in 5 minute periods so I'm pretty confident there is a bug somewhere.

My inverter is a 10k, and I was using a port mirror on my switch to capture the comm's with solaredge, so I'm not sure if its related to the capture method.