jbuehl/solaredge #41: Real-time data over serial line

ksaihtam, 2017-06-05:
Hello,

i have found the commands and the format of the data structures to get real time measurement data from a single phase inverter. The commands are appended. Maybe someone can find the meaning of the unknown fields.

Regards.

`PROT_CMD_VENUSMNGR_READ_ISE_MEAS1`

- Data format (for python struct)
  - "<ffffffffffL"
- Field description
  - 'Vdc','Vac','Unk0','Vac2','Fac','Cac','Unk1','Unk2','Unk3','Pac','Status'

`PROT_CMD_VENUSMNGR_READ_ISE_MEAS2`

- Data format (for python struct)
  - "<fffffffL"
- Field description
  - 'Vac','Pac','Unk0','Unk1','Unk2','Unk3','Riso','Status'

`PROT_CMD_VENUSMNGR_READ_SE_MEAS`

- Data format (for python struct)
  - "<ffffffffff"
- Field description
  - 'Vdc','Vac','Unk0','Unk1','Vac2','Unk2','Fac','Unk3','Unk4','Unk5'

ksaihtam, 2017-07-02:
Hello,

to get the telemetry data over the serial line consecutive PROT_CMD_VENUSMNGR_GET_TELEM commands will sometimes return the current optimizer data. Most time, i poll it with a interval of 0.2 seconds, this command will return a response with no content.
The returned message with the optimizer data looks like this (i removed some serial numbers):

`01 02 10 00 XX XX 9A 7F 1A 01 XX XX 37 20 XX XX 9A 7F F2 01 A9 75 25 0B 0D 83 2F 00 00 02 9C 58 59 3D 8A 00 00 00 00 00`

I identified a timestamp "02 9C 58 59" but currently not more. Anyone know the format of this message?

And there is also a message with some inverter data like the dc voltage:

`01 01 4C 00 XX XX 9A 7F 00 60 AD 43 .....`

Regards.

vandalon, 2017-09-15:
Can you tell me please what params for semonitor.py you use to get this data?

ksaihtam, 2017-09-19:
I use a modified/own simple version to do this.
I can't tell you what parameters you need but you need to send the PROT_CMD_VENUSMNGR_GET_TELEM command to the inverter.

vandalon, 2017-09-19:
Hi ksaihtam, are you willing to share your own modified version? :)

ksaihtam, 2017-09-19:
You need to modify the serial port and inv id in cmd_test.py
Maybe it will run. It can take up to some minutes to recv. the optimizer data.

[se_inv_test.zip](https://github.com/jbuehl/solaredge/files/1313896/se_inv_test.zip)

vandalon, 2017-09-19:
Ah, you wait for the inverter to push data. I'm trying to force it as fast as possible. I did find a way. but that only gives the usage data in Etot/Eyear/Emonth/Eday. Enough to calculate Pac so I think this is the best way to go for now if you want data every 10 seconds or so.