jbuehl/solaredge #42: Exception Timestamp with inverter data

fahimmac, 2017-06-15:
Hi All,

I was trying to use semonitor.py with my SolarEdge inverter. While using the -c 322 command with semonitor.py i can get energy values from the inverter.

But while trying to get the raw values from inverter, i came across a timestamp exception. The logging was working fine in last couple of days but now i am having this error.

Also i tried omitting the timestamp function in order to get only the inverter data. But had no luck.

`` pi@logger` ~/inverter/sedgenew $ sudo python semonitor.py -m -s 07e1a409e -d stdout -vvv -t 4 /dev/ttyUSB0
Jun 15 14:03:06 debugEnable: True 
Jun 15 14:03:06 debugFiles: True 
Jun 15 14:03:06 debugMsgs: True 
Jun 15 14:03:06 debugData: True 
Jun 15 14:03:06 debugRaw: False 
Jun 15 14:03:06 debugFileName: stdout 
Jun 15 14:03:06 haltOnException: False 
Jun 15 14:03:06 inFileName: /dev/ttyUSB0 
Jun 15 14:03:06 inputType: 4 
Jun 15 14:03:06 serialDevice: True 
Jun 15 14:03:06     baudRate: 115200 
Jun 15 14:03:06 networkDevice: False 
Jun 15 14:03:06 networkSvcs: False 
Jun 15 14:03:06 following: True 
Jun 15 14:03:06 passiveMode: False 
Jun 15 14:03:06 commandAction: False 
Jun 15 14:03:06 masterMode: True 
Jun 15 14:03:06 slaveAddrs: 07e1a409e 
Jun 15 14:03:06 outFileName: stdout 
Jun 15 14:03:06 append: w 
Jun 15 14:03:06 opening /dev/ttyUSB0 
Jun 15 14:03:06 starting read thread 
Jun 15 14:03:06 starting master thread 
Jun 15 14:03:06 dataLen:    0000 
Jun 15 14:03:06 dataLenInv: ffff 
Jun 15 14:03:06 sequence:   00e7 
Jun 15 14:03:06 source:     fffffffe 
Jun 15 14:03:06 dest:       7e1a409e 
Jun 15 14:03:06 function:   0302 
Jun 15 14:03:06 /dev/ttyUSB0 <-- message: 1 length: 22 
Jun 15 14:03:06   
Jun 15 14:03:43   
Jun 15 14:03:43 /dev/ttyUSB0 --> message: 1 length: 520 
Jun 15 14:03:43 dataLen:    01f2 
Jun 15 14:03:43 dataLenInv: fe0d 
Jun 15 14:03:43 sequence:   024d 
Jun 15 14:03:43 source:     7e1a409e 
Jun 15 14:03:43 dest:       fffffffe 
Jun 15 14:03:43 function:   0500 
Jun 15 14:03:43 Exception: **timestamp out of range for platform time_t**  ``

Any help would be really great. 

Thanks a lot in advance.

jbuehl, 2017-06-15:
You might want to check what the date and time is set to in the inverter and also in the server running the program.  It looks like one of the time conversion routines in the operating system is saying that the value of the timestamp it is trying to parse is unable to be converted to local time.

fahimmac, 2017-06-23:
i just checked in the inverter and the server (raspberry pi). But the problem still persists. Also with the -c 322 command the time converted from inverter is different. It always stays the same.

Jun 23 13:23:20 {"data": {"Emon": 1664814.5, "Etot": 6624749.0, "Eyear": 4049066.5, "Eday": 746.5416259765625, "Time": "Fri Jun 23 02:00:00 2017"}, "command": 802, "response": 909, "sequence": 245} 

I was just wondering if there is any other way to skip the time conversion routine ? So i can get the the raw data from the inverter. Thanks a lot in advance.

jbuehl, 2017-06-24:
Can you run semonitor.py and include the -x option?  That will provide more information about what is happening when it gets the exception.

fahimmac, 2017-06-24:
This is the output i am getting from the inverter.

pi@logger ~/inverter/solaredge $ sudo python semonitor.py -m -x -s 07e1a409e -d stdout -vvv -t 4 /dev/ttyUSB0

Jun 24 17:42:43 debugEnable: True 
Jun 24 17:42:43 debugFiles: True 
Jun 24 17:42:43 debugMsgs: True 
Jun 24 17:42:43 debugData: True 
Jun 24 17:42:43 debugRaw: False 
Jun 24 17:42:43 debugFileName: stdout 
Jun 24 17:42:43 haltOnException: True 
Jun 24 17:42:43 inFileName: /dev/ttyUSB0 
Jun 24 17:42:43 inputType: 4 
Jun 24 17:42:43 serialDevice: True 
Jun 24 17:42:43     baudRate: 115200 
Jun 24 17:42:43 networkDevice: False 
Jun 24 17:42:43 sePort: 22222 
Jun 24 17:42:43 networkSvcs: False 
Jun 24 17:42:43 following: True 
Jun 24 17:42:43 passiveMode: False 
Jun 24 17:42:43 commandAction: False 
Jun 24 17:42:43 masterMode: True 
Jun 24 17:42:43 slaveAddrs: 07e1a409e 
Jun 24 17:42:43 outFileName: stdout 
Jun 24 17:42:43 append: w 
Jun 24 17:42:43 opening /dev/ttyUSB0 
Jun 24 17:42:43 starting read thread 
Jun 24 17:42:43 starting master thread 
Jun 24 17:42:43 dataLen:    0000 
Jun 24 17:42:43 dataLenInv: ffff 
Jun 24 17:42:43 sequence:   0007 
Jun 24 17:42:43 source:     fffffffe 
Jun 24 17:42:43 dest:       7e1a409e 
Jun 24 17:42:43 function:   0302 
Jun 24 17:42:43 /dev/ttyUSB0 <-- message: 1 length: 22 
Jun 24 17:42:43   
Jun 24 17:42:53 RS485 master ack timeout 
Jun 24 17:42:58 dataLen:    0000 
Jun 24 17:42:58 dataLenInv: ffff 
Jun 24 17:42:58 sequence:   0008 
Jun 24 17:42:58 source:     fffffffe 
Jun 24 17:42:58 dest:       7e1a409e 
Jun 24 17:42:58 function:   0302 
Jun 24 17:42:58 /dev/ttyUSB0 <-- message: 2 length: 22 
Jun 24 17:42:58   
Jun 24 17:43:08 RS485 master ack timeout 
Jun 24 17:43:13 dataLen:    0000 
Jun 24 17:43:13 dataLenInv: ffff 
Jun 24 17:43:13 sequence:   0009 
Jun 24 17:43:13 source:     fffffffe 
Jun 24 17:43:13 dest:       7e1a409e 
Jun 24 17:43:13 function:   0302 
Jun 24 17:43:13 /dev/ttyUSB0 <-- message: 3 length: 22 
Jun 24 17:43:13   
Jun 24 17:43:23 RS485 master ack timeout 
Jun 24 17:43:28 dataLen:    0000 
Jun 24 17:43:28 dataLenInv: ffff 
Jun 24 17:43:28 sequence:   000a 
Jun 24 17:43:28 source:     fffffffe 
Jun 24 17:43:28 dest:       7e1a409e 
Jun 24 17:43:28 function:   0302 
Jun 24 17:43:28 /dev/ttyUSB0 <-- message: 4 length: 22 
Jun 24 17:43:28   
Jun 24 17:43:38 RS485 master ack timeout 
Jun 24 17:43:41   
Jun 24 17:43:41 /dev/ttyUSB0 --> message: 1 length: 520 
Jun 24 17:43:41 dataLen:    01f2 
Jun 24 17:43:41 dataLenInv: fe0d 
Jun 24 17:43:41 sequence:   022e 
Jun 24 17:43:41 source:     7e1a409e 
[... 39 lines cut ...]
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "semonitor.py", line 40, in readData
    processMsg(msg, dataFile, recFile, outFile)
  File "semonitor.py", line 56, in processMsg
    msgData = parseData(function, data)
  File "/home/pi/inverter/solaredge/seData.py", line 22, in parseData
    return parseDeviceData(data)
  File "/home/pi/inverter/solaredge/seData.py", line 147, in parseDeviceData
    eventDict[seId] = parseEventData(seId, eventItems, data[dataPtr:dataPtr+devLen])
  File "/home/pi/inverter/solaredge/seData.py", line 171, in parseEventData
    seEventData[2] = time.asctime(time.localtime(seEventData[2]))
ValueError: timestamp out of range for platform time_t

Jun 24 17:43:43 dataLen:    0000 
Jun 24 17:43:43 dataLenInv: ffff 
Jun 24 17:43:43 sequence:   000b 
Jun 24 17:43:43 source:     fffffffe 
Jun 24 17:43:43 dest:       7e1a409e 
Jun 24 17:43:43 function:   0302 
Jun 24 17:43:43 /dev/ttyUSB0 <-- message: 5 length: 22

Thank you so much for the reply.

jbuehl, 2017-06-24:
I just made a change to sedata.py that should handle the case of an invalid timestamp.  I haven't tested it, so please let me know if it works or not, and post the output that you get.

fahimmac, 2017-06-25:
This is the output i have:

pi@logger ~/inverter/solaredge $ sudo python semonitor.py -m -x -s 07e1a409e -d stdout -vvv -t 4 /dev/ttyUSB0
Jun 25 10:47:08 debugEnable: True 
Jun 25 10:47:08 debugFiles: True 
Jun 25 10:47:08 debugMsgs: True 
Jun 25 10:47:08 debugData: True 
Jun 25 10:47:08 debugRaw: False 
Jun 25 10:47:08 debugFileName: stdout 
Jun 25 10:47:08 haltOnException: True 
Jun 25 10:47:08 inFileName: /dev/ttyUSB0 
Jun 25 10:47:08 inputType: 4 
Jun 25 10:47:08 serialDevice: True 
Jun 25 10:47:08     baudRate: 115200 
Jun 25 10:47:08 networkDevice: False 
Jun 25 10:47:08 sePort: 22222 
Jun 25 10:47:08 networkSvcs: False 
Jun 25 10:47:08 following: True 
Jun 25 10:47:08 passiveMode: False 
Jun 25 10:47:08 commandAction: False 
Jun 25 10:47:08 masterMode: True 
Jun 25 10:47:08 slaveAddrs: 07e1a409e 
Jun 25 10:47:08 outFileName: stdout 
Jun 25 10:47:08 append: w 
Jun 25 10:47:08 opening /dev/ttyUSB0 
Jun 25 10:47:08 starting read thread 
Jun 25 10:47:08 starting master thread 
Jun 25 10:47:08 dataLen:    0000 
Jun 25 10:47:08 dataLenInv: ffff 
Jun 25 10:47:08 sequence:   000b 
Jun 25 10:47:08 source:     fffffffe 
Jun 25 10:47:08 dest:       7e1a409e 
Jun 25 10:47:08 function:   0302 
Jun 25 10:47:08 /dev/ttyUSB0 <-- message: 1 length: 22 
Jun 25 10:47:08   
Jun 25 10:47:11   
Jun 25 10:47:11 /dev/ttyUSB0 --> message: 1 length: 22 
Jun 25 10:47:11 dataLen:    0000 
Jun 25 10:47:11 dataLenInv: ffff 
Jun 25 10:47:11 sequence:   01f0 
Jun 25 10:47:11 source:     7e1a409e 
Jun 25 10:47:11 dest:       fffffffe 
Jun 25 10:47:11 function:   0501 
Jun 25 10:47:11 dataLen:    0008 
Jun 25 10:47:11 dataLenInv: fff7 
Jun 25 10:47:11 sequence:   01f0 
Jun 25 10:47:11 source:     fffffffe 
Jun 25 10:47:11 dest:       7e1a409e 
Jun 25 10:47:11 function:   0580 
Jun 25 10:47:11 /dev/ttyUSB0 <-- message: 2 length: 30 
Jun 25 10:47:11   
Jun 25 10:47:11   
Jun 25 10:47:11 /dev/ttyUSB0 --> message: 2 length: 520 
Jun 25 10:47:11 dataLen:    01f2 
Jun 25 10:47:11 dataLenInv: fe0d 
Jun 25 10:47:11 sequence:   01f1 
Jun 25 10:47:11 source:     7e1a409e 
Jun 25 10:47:11 dest:       fffffffe 
Jun 25 10:47:11 function:   0500 
Jun 25 10:47:11 Exception: 'long' object is not iterable 
Jun 25 10:47:11 data:       f2 01 0d fe f1 01 9e 40 1a 7e fe ff ff ff 00 05 
Jun 25 10:47:11 data:       00 03 9e 40 1a 7e 1c 00 4d 7e 3d 59 76 00 00 00 
Jun 25 10:47:11 data:       28 01 00 ee 9b 28 01 00 ff ff 00 00 ff ff 00 00 
Jun 25 10:47:11 data:       00 00 00 00 80 00 e8 d2 3a 20 0d 00 4d 7e 3d 59 
Jun 25 10:47:11 data:       5f c6 d8 5d 83 08 d2 33 10 80 00 be ed 3a 20 0d 
Jun 25 10:47:11 data:       00 69 7e 3d 59 48 c6 db 71 c3 08 41 34 10 80 00 
Jun 25 10:47:11 data:       13 e8 3a 20 0d 00 84 7e 3d 59 92 c4 dc f9 02 08 
Jun 25 10:47:11 data:       8f 28 10 80 00 95 c3 3a 20 0d 00 86 7e 3d 59 c2 
Jun 25 10:47:11 data:       c5 d3 2d 33 08 03 28 10 80 00 e2 d3 3a 20 0d 00 
Jun 25 10:47:11 data:       89 7e 3d 59 d8 c4 dd 15 03 08 24 32 10 80 00 f0 
Jun 25 10:47:11 data:       c9 3a 20 0d 00 8b 7e 3d 59 a5 c6 d7 61 43 09 b8 
Jun 25 10:47:11 data:       34 10 80 00 f4 bf 3a 20 0d 00 8f 7e 3d 59 c2 c5 
Jun 25 10:47:11 data:       d5 29 83 08 6e 23 10 80 00 26 cd 3a 20 0d 00 9f 
Jun 25 10:47:11 data:       7e 3d 59 a3 c6 d8 5d 23 09 66 28 0f 80 00 0f c4 
Jun 25 10:47:11 data:       3a 20 0d 00 a0 7e 3d 59 62 c5 e0 25 53 08 d6 2c 
Jun 25 10:47:11 data:       10 80 00 00 cd 3a 20 0d 00 ab 7e 3d 59 8b c6 d5 
Jun 25 10:47:11 data:       69 43 09 18 34 0f 80 00 a0 bf 3a 20 0d 00 b7 7e 
Jun 25 10:47:11 data:       3d 59 bc c6 d9 19 83 08 06 32 10 80 00 01 c4 3a 
Jun 25 10:47:11 data:       20 0d 00 ce 7e 3d 59 ad c6 d6 29 73 08 1d 32 0f 
Jun 25 10:47:11 data:       80 00 23 cd 3a 20 0d 00 d2 7e 3d 59 d0 c6 cf 55 
Jun 25 10:47:11 data:       73 09 45 35 10 80 00 98 bf 3a 20 0d 00 de 7e 3d 
Jun 25 10:47:11 data:       59 ca c6 df 25 33 08 3b 32 10 80 00 4d cb 3a 20 
Jun 25 10:47:11 data:       0d 00 e3 7e 3d 59 fa c5 dc 5d c3 09 8b 34 0f 80 
Jun 25 10:47:11 data:       00 e2 d3 3a 20 0d 00 eb 7e 3d 59 38 c5 e7 0d 33 
Jun 25 10:47:11 data:       08 2a 32 10 80 00 67 d8 3a 20 0d 00 f1 7e 3d 59 
Jun 25 10:47:11 data:       03 c7 db 61 63 09 f8 32 0f 80 00 09 cd 3a 20 0d 
Jun 25 10:47:11 data:       00 fb 7e 3d 59 f3 c5 df 65 e3 09 92 34 0f 80 00 
Jun 25 10:47:11 data:       cd bf 3a 20 0d 00 15 7f 3d 59 09 c6 de 41 c3 09 
Jun 25 10:47:11 data:       3f 27 0f 80 00 be ed 3a 20 0d 00 1e 7f 3d 59 fc 
Jun 25 10:47:11 data:       c6 e2 71 13 0a 4c 34 10 80 00 02 cd 3a 20 0d 00 
Jun 25 10:47:11 data:       36 7f 3d 59 16 c7 e6 59 13 0a aa 33 10 80 00 ff 
Jun 25 10:47:11 data:       cc 3a 20 0d 00 3c 7f 3d 59 58 c7 dc 61 a3 0a 49 
Jun 25 10:47:11 data:       33 0f db 74 
Exception in thread read thread:

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "semonitor.py", line 40, in readData
    processMsg(msg, dataFile, recFile, outFile)
  File "semonitor.py", line 56, in processMsg
    msgData = parseData(function, data)
  File "/home/pi/inverter/solaredge/seData.py", line 22, in parseData
    return parseDeviceData(data)
  File "/home/pi/inverter/solaredge/seData.py", line 147, in parseDeviceData
    eventDict[seId] = parseEventData(seId, eventItems, data[dataPtr:dataPtr+devLen])
  File "/home/pi/inverter/solaredge/seData.py", line 171, in parseEventData
    seEventData[2] = formatDateTime(seEventData[2])
  File "/home/pi/inverter/solaredge/seData.py", line 245, in formatDateTime
    return ''.join(x.encode('hex') for x in timeStamp)
TypeError: 'long' object is not iterable

jbuehl, 2017-06-25:
Sorry, I messed up the debug output.  It should work better now.  Try it again.

fahimmac, 2017-06-26:
Hello,
Its working now. This the part of the raw output in order for debug.

Jun 26 12:59:54 /dev/ttyUSB0 --> message: 1 length: 463 
Jun 26 12:59:54 dataLen:    01b9 
Jun 26 12:59:54 dataLenInv: fe46 
Jun 26 12:59:54 sequence:   0409 
Jun 26 12:59:54 source:     7e1a409e 
Jun 26 12:59:54 dest:       fffffffe 
Jun 26 12:59:54 function:   0500 
Jun 26 12:59:54 optimizer:      203ACD23 type: 0080 len: 000d 
Jun 26 12:59:54     Uptime : 56197 
Jun 26 12:59:54     Temp : 26.0 
Jun 26 12:59:54     Vmod : 61.125 
Jun 26 12:59:54     Imod : 0.2 
Jun 26 12:59:54     Eday : 2949.5 
Jun 26 12:59:54     Vopt : 20.0 
Jun 26 12:59:54     Time : 21:06:09 
Jun 26 12:59:54     Date : 2017-06-13 
Jun 26 12:59:54     Inverter : 0 
Jun 26 12:59:54     ID : 203ACD23 
Jun 26 12:59:54 <stdout> <-- message: 1 length: 3762 
Jun 26 12:59:54

Jun 26 12:59:54 {"inverters": {"7E1A409E": {"Uptime": 76285, "CosPhi1": 1.0, "Freq3": 49.97849655151367, "Freq2": 49.974151611328125, "Freq1": 49.977237701416016, "data29": 100.0, "data21": 4286578687, "data22": 4286578687, "data23": 4286578687, "GndFrR": 6130.619140625, "data31": 0, "EdayDC": 4286578687, "Vac3": 231.625, "Eac": 30.0, "Vac1": 226.125, "Idc": 4286578687, "Date": "2017-06-13", "CosPhi3": 1.0, "CosPhi2": 1.0, "IoutDC": -0.009521484375, "Temp": 44.214500427246094, "Interval": 300, "Iac3": 0.8671875, "Iac2": 0.890625, "Iac1": 0.859375, "Time": "21:09:57", "ID": "7E1A409E", "Eday": 71393.5078125, "Edc": 4286578687, "Etot": 5811703.0, "Vdc": 747.5, "Vac2": 227.984375, "mode": 4, "Irdc": 0.0024374998174607754}}, "meters_0x0022": {"7E1A409E": {"9_PVProduction": {"TotalE2Grid": 0, "AlwaysZero_off34_int2": 0, "seId": "7E1A409E", "TotalEfromGrid": 0, "Flag_off20_hex": "00 80", "devLen": 58, "Flag_off28_hex": "00 80", "E2X": 30, "PfromX": NaN, "P2X": 297.0, "Date": "2017-06-13", "AlwaysZero_off18_int2": 0, "Totaloff22_int4": 0, "Totaloff30_int4": 0, "Interval": 300, "EfromX": 0, "seType": "0x0022", "Time": "21:09:57", "onlyIntervalData": 1, "dateTime": 1497380997, "AlwaysZero_off26_int2": 0, "AlwaysZero_off10_int2": 0, "devType": "meters_0x0022", "Flag_off36_hex": "00 80", "Flag_off12_hex": "00 80", "recType": 9}}}, "events": {}, "optimizers": {"203AC31C": {"Uptime": 56336, "Temp": 26.0, "Vmod": 60.25, "Imod": 0.30625, "Eday": 2571.5, "Vopt": 33.125, "Time": "21:09:57", "Date": "2017-06-13", "Inverter": 0, "ID": "203AC31C"}, "203ABFA0": {"Uptime": 56464, "Temp": 26.0, "Vmod": 55.5, "Imod": 0.2, "Eday": 2162.25, "Vopt": 19.375, "Time": "21:10:20", "Date": "2017-06-13", "Inverter": 0, "ID": "203ABFA0"}, "203AC335": {"Uptime": 56170, "Temp": 24.0, "Vmod": 60.375, "Imod": 0.4, "Eday": 2967.0, "Vopt": 45.875, "Time": "21:10:38", "Date": "2017-06-13", "Inverter": 0, "ID": "203AC335"}, "203AE7F2": {"Uptime": 55701, "Temp": 24.0, "Vmod": 58.5, "Imod": 0.16875, "Eday": 2029.5, "Vopt": 17.5, "Time": "21:09:58", "Date": "2017-06-13", "Inverter": 0, "ID": "203AE7F2"}, "203ACB4D": {"Uptime": 56241, "Temp": 22.0, "Vmod": 39.625, "Imod": 0.38125000000000003, "Eday": 2982.0, "Vopt": 30.5, "Time": "21:10:33", "Date": "2017-06-13", "Inverter": 0, "ID": "203ACB4D"}, "203AC401": {"Uptime": 55699, "Temp": 26.0, "Vmod": 56.625, "Imod": 0.16875, "Eday": 2752.25, "Vopt": 17.75, "Time": "21:10:20", "Date": "2017-06-13", "Inverter": 0, "ID": "203AC401"}, "203AD2E8": {"Uptime": 56449, "Temp": 26.0, "Vmod": 57.125, "Imod": 0.3125, "Eday": 2916.0, "Vopt": 35.5, "Time": "21:10:02", "Date": "2017-06-13", "Inverter": 0, "ID": "203AD2E8"}}} 
{"inverters": {"7E1A409E": {"Uptime": 76285, "CosPhi1": 1.0, "Freq3": 49.97849655151367, "Freq2": 49.974151611328125, "Freq1": 49.977237701416016, "data29": 100.0, "data21": 4286578687, "data22": 4286578687, "data23": 4286578687, "GndFrR": 6130.619140625, "data31": 0, "EdayDC": 4286578687, "Vac3": 231.625, "Eac": 30.0, "Vac1": 226.125, "Idc": 4286578687, "Date": "2017-06-13", "CosPhi3": 1.0, "CosPhi2": 1.0, "IoutDC": -0.009521484375, "Temp": 44.214500427246094, "Interval": 300, "Iac3": 0.8671875, "Iac2": 0.890625, "Iac1": 0.859375, "Time": "21:09:57", "ID": "7E1A409E", "Eday": 71393.5078125, "Edc": 4286578687, "Etot": 5811703.0, "Vdc": 747.5, "Vac2": 227.984375, "mode": 4, "Irdc": 0.0024374998174607754}}, "meters_0x0022": {"7E1A409E": {"9_PVProduction": {"TotalE2Grid": 0, "AlwaysZero_off34_int2": 0, "seId": "7E1A409E", "TotalEfromGrid": 0, "Flag_off20_hex": "00 80", "devLen": 58, "Flag_off28_hex": "00 80", "E2X": 30, "PfromX": NaN, "P2X": 297.0, "Date": "2017-06-13", "AlwaysZero_off18_int2": 0, "Totaloff22_int4": 0, "Totaloff30_int4": 0, "Interval": 300, "EfromX": 0, "seType": "0x0022", "Time": "21:09:57", "onlyIntervalData": 1, "dateTime": 1497380997, "AlwaysZero_off26_int2": 0, "AlwaysZero_off10_int2": 0, "devType": "meters_0x0022", "Flag_off36_hex": "00 80", "Flag_off12_hex": "00 80", "recType": 9}}}, "events": {}, "optimizers": {"203AC31C": {"Uptime": 56336, "Temp": 26.0, "Vmod": 60.25, "Imod": 0.30625, "Eday": 2571.5, "Vopt": 33.125, "Time": "21:09:57", "Date": "2017-06-13", "Inverter": 0, "ID": "203AC31C"}, "203ABFA0": {"Uptime": 56464, "Temp": 26.0, "Vmod": 55.5, "Imod": 0.2, "Eday": 2162.25, "Vopt": 19.375, "Time": "21:10:20", "Date": "2017-06-13", "Inverter": 0, "ID": "203ABFA0"}, "203AC335": {"Uptime": 56170, "Temp": 24.0, "Vmod": 60.375, "Imod": 0.4, "Eday": 2967.0, "Vopt": 45.875, "Time": "21:10:38", "Date": "2017-06-13", "Inverter": 0, "ID": "203AC335"}, "203AE7F2": {"Uptime": 55701, "Temp": 24.0, "Vmod": 58.5, "Imod": 0.16875, "Eday": 2029.5, "Vopt": 17.5, "Time [... cut ...]
Jun 26 12:59:54     meters_0x0022 : {'7E1A409E': {'9_PVProduction': {'TotalE2Grid': 0, 'AlwaysZero_off34_int2': 0, 'seId': '7E1A409E', 'TotalEfromGrid': 0, 'Flag_off20_hex': '00 80', 'devLen': 58, 'Flag_off28_hex': '00 80', 'E2X': 22, 'PfromX': nan, 'P2X': 226.0, 'Date': '2017-06-13', 'AlwaysZero_off18_int2': 0, 'Totaloff22_int4': 0, 'Totaloff30_int4': 0, 'Interval': 300, 'EfromX': 0, 'seType': '0x0022', 'Time': '21:14:57', 'onlyIntervalData': 1, 'dateTime': 1497381297, 'AlwaysZero_off26_int2': 0, 'AlwaysZero_off10_int2': 0, 'devType': 'meters_0x0022', 'Flag_off36_hex': '00 80', 'Flag_off12_hex': '00 80', 'recType': 9}}} 

Thank you so much for the help.

closed 2017-06-26 by fahimmac