jbuehl/solaredge #45: How to get the keys without RS232?

hagenbauer, 2017-08-23:
Not an issue but a question. From what I have read in my documentation my inverter does not have a RS232 interface. 

Any chance that I can get the encryption keys withouth it?

Regards
Hagen

Whizkidzz, 2017-08-23:
Your inverter has a USB port inside, as you connect this to the PC, it acts as a serial port.

oliv3r, 2017-08-24:
@whizkidzz was that answer very specific to hagenbauer? As I cannot see that he has a USB interface on his inverter :)

I checked the PCB of my new HD Wave 5000 (SE5000H) and there seems to be pads for a USB connector, but it is not soldered. Also, I do not have RS232, so came to this spot for the same question.

I do have RS484 and was able to extract the key as described in the README :) I did have to delete the seseq.key file a few times in between runs for some strange reason ...

hagenbauer, 2017-08-26:
I opened the inverter and found a mini usb connector in the area that is also mentioned in this documentation. https://www.solaredge.com/sites/default/files/application-note-upgrading-solaredge-inverter-fw.pdf  So I assumed it is save to connect this to a pc.
On my server it seems that is is recognized. lsusb shows a new device

Bus 001 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC

So now I should be able to move on. Next Step key extraction

hagenbauer, 2017-08-27:
I have troubles to get results from semonitor so just to be sure I did not make any mistakes
This is what I did

`python semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7F159E84 -t 2  /dev/ttyUSB0 | python sekey.py -o 7F159E84.key`

and the key looks like (I replaced some of the numbers)

0878466kd4330b1ac57cee5f2a9cf234`
`
could this be correct or should the key look different?

JustApu, 2017-08-27:
That's what my three keys look like as well. Each 32 characters long.

Jerrythafast, 2017-08-27:
@hagenbauer, are you sure there is a `k` in your key, or is this one of the numbers you replaced? The key can't possibly contain a `k`, so if it seems to do for you, this is a bug in the script. The key should be 32 hexadecimal digits, i.e. `0-9` and `a-f`.

hagenbauer, 2017-08-27:
@Jerrythafast yes you are right. that was one of the characters I replaced. The key does only contain hexadecimal values.

felixmartens, 2017-09-14:
@oliv3r coming back to the original topic of the issue. I have the same HD Wave SE5000 Inverter installed and i can't find a RS232 in housing or the amnual either. So i came here to find out how to extract the keys via RS485. Can you give some detailed explanation what steps you took (master/slave configuration, commands) etc.?

Webunity, 2017-11-02:
So i have the SE7000 3 phase inverter which also doesn't have the USB connector soldered on. It does have the  and i tried using a 'smart meter' FTDI cable, which is reported by DMSG as follows:

> [ 2024.655097] usb 1-1.3: new full-speed USB device number 6 using dwc_otg
[ 2024.812646] usb 1-1.3: New USB device found, idVendor=0403, idProduct=6001
[ 2024.812658] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2024.812667] usb 1-1.3: Product: FT232R USB UART
[ 2024.812674] usb 1-1.3: Manufacturer: FTDI
[ 2024.812682] usb 1-1.3: SerialNumber: AI1NS8A5
[ 2024.821003] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[ 2024.821187] usb 1-1.3: Detected FT232RL
[ 2024.822807] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
[ 2106.534711] usb 1-1.3: USB disconnect, device number 6

However this doesn't work. I assume because the cable is based on the FT232R chip, and has an inverted RXD signal. Even if i remove the inversion with FT_PROG it doesn't work. So now i've ordered an FT232 5V TTL USB Serial Port Adapter which should work, it arrives tomorrow, fingers crossed.

felixmartens, 2017-11-04:
@Webunity
As I wrote above I have the new HD Wave SE5000 Inverter. 
I was able to retrieve the keys via this adapter https://www.amazon.de/dp/B007VZY4CW/ref=cm_sw_em_r_mt_awd_s_3FH.zbTN2PEKE.

But i had to issue every of the 4 commands separately and multiple times(!) to get the correct value. I then copied the keys together to one file before piping it to sekey.

Webunity, 2017-11-14:
@felixmartens that you had to switch it makes sense, right?
Serial TX -> Device RX (transmit FROM serial, device needs to RECEIVE)
Serial RX <- Device TX (transmit FROM device, serial needs to RECEIVE)

I just received a serial to USB adapter and can try tomorrow if it works for me. Fingers crossed.

felixmartens, 2017-11-14:
There you can see that I do not really understand the serial console. I am more the software developer ;D
I googled the wiring because i could not figure to match "A" and "B" that to "+"  and "-".

Webunity, 2017-11-23:
Well i still have no luck in getting my keys. I have not touched the solaredge menu so i don't know if any dip-switches have to be turned on or any configuration has to be made. I have connected GND, RX and TX to the RS232 block (nothing else) and i even inverted RX and TX without luck. Who can point me in the right direction?

felixmartens, 2017-11-23:
Are u using RS232 or RS435?  Mine has no 232 so I used 435 over the adapter. Therefore you need to change behavior to slave in menu. Can't help you with 232 though.

Webunity, 2017-11-23:
@felixmartens i used RS232 (the right 3 legs). I just did a factory reset and i am going to re-'pair' my optimizers tomorrow morning. I see an entry already of the 3 phase inverter in my MySQL table so it LOOKS to be ok now.

felixmartens, 2017-11-25:
Ok,cause you are commenting the issue about getting keys without (!) RS232. Is your Inverter one of the new HD Wave Series?

Webunity, 2017-12-09:
@felixmartens yes it is, and yes i commented in the wrong issue, sorry for that. For those interested how i managed to get it working in the end; i did a factory reset; re-paired the optimizers and waited for 2 weeks untill i got a new key which i could extract using the provided scripts.

felixmartens, 2017-12-10:
No problem. I just wanted to make sure we are talking about the the same things ;) glad you made it though!

ver-luisant, 2018-05-04:
I own an HD Wave SE6000 inverter and am unable to get the keys through RS485 (no RS232 port available).

Inverter is configured as: SE protocol / Slave / ID:1.
I use the adapter suggested by @felixmartens.
I tried both RS485-1 and RS485-2, with/without termination resistor, and swapping A/B lines, all of those  ith no success.

I see two types of errors :
- `sudo python semonitor.py  -s 7xxxxxxx -m  -t 4 -vv /dev/ttyUSB0` gives master ack timeouts: 
```
append: False
baudrate: 115200
commands: 
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stderr
master: True
outfile: stdout
ports: 22222,22221,80
record: None
slaves: 7xxxxxxx
type: 4
updatefile: None
verbose: 2
xerror: False
opening /dev/ttyUSB0
starting read thread
starting master thread
/dev/ttyUSB0 <-- message: 1 length: 22
RS485 master ack timeout
/dev/ttyUSB0 <-- message: 2 length: 22
RS485 master ack timeout
/dev/ttyUSB0 <-- message: 3 length: 22
^Cclosing /dev/ttyUSB0
closing <stdout>
```

- and when sending a command `sudo python semonitor.py -c 12,H239 -s 7xxxxxxx -m -t 4 -vv /dev/ttyUSB0`, I get the following error:
```
Traceback (most recent call last):
  File "semonitor.py", line 241, in <module>
    doCommands(args, mode, dataFile, recFile, outFile)
  File "semonitor.py", line 148, in doCommands
    se.msg.formatMsg(nextSeq(), MASTER_ADDR, args.slaves[0], se.commands.PROT_CMD_POLESTAR_MASTER_GRANT), 
  File "/home/verluisant/solaredge/se/msg.py", line 210, in formatMsg
    checksum = calcCrc(struct.pack(">HLLH", msgSeq, fromAddr, toAddr, function) + data)
struct.error: cannot convert argument to integer
```

Any help will be appreciated. Thanks.

Best Regards

alexmasolin, 2018-05-05:
As written somewhere else here, it is not (yet) possible to retrieve the key via RS485.
AFAIK every HD wave inverter have a USB port on their communication board.
The USB port is in fact the RS232 interface. Try to look for a micro USB port and connect it to your pc: it will be recognised as RS232

felixmartens, 2018-05-05:
Unfortunately I have to correct you, as I successfully retrieved the keys via rs485 as the is no usb like connector on mine.
@ver-luisant please check if your Inverter is configured as slave. I had to issue the commands from sekey multiple times and separately. Did you try that?

ver-luisant, 2018-05-05:
@alexmasolin
I confirm, USB connector is unfortunately not fitted. Unsoldered pads are visible next to the ethernet interface:
[image]

@felixmartens Yes, inverter is configured as slave, protocol set to SolarEdge. Id is kept to the default value (1). Tried both RS485 ports. When sending a command `-c 12,H239` I have a python error `struct.error: cannot convert argument to integer`. Without that command, the RS485 master ack timeout is raised.
.

ver-luisant, 2018-05-05:
I found a possible issue in semonitor.py,
line 148 should read:
`se.msg.formatMsg(nextSeq(), MASTER_ADDR, int(args.slaves[0], 16), se.commands.PROT_CMD_POLESTAR_MASTER_GRANT), `

instead of:
`se.msg.formatMsg(nextSeq(), MASTER_ADDR, args.slaves[0], se.commands.PROT_CMD_POLESTAR_MASTER_GRANT), `

This fix solved the error when sending a command (`struct.error: cannot convert argument to integer`).

The communication problem with the inverter remains though (RS485 master ack timeouts)...

jbuehl, 2018-05-05:
@ver-luisant the slave is not responding so it would probably be either a hardware problem, serial configuration problem, or the slave address doesn't match.  In order to rule out a software problem, please run it with the -vvvv option so we can see exactly what the message is that is being sent.

ver-luisant, 2018-05-08:
Attached the output with the -vvvv option:
`sudo python semonitor.py  -s 7312C737 -m -t 4 -d stdout -vvvv /dev/ttyUSB0`

> append: False
baudrate: 115200
commands: 
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stdout
master: True
outfile: stdout
ports: 22222,22221,80
record: None
slaves: 7312C737
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyUSB0
starting read thread
starting master thread
dataLen:    0000
dataLenInv: ffff
sequence:   0069
source:     fffffffe
dest:       7312c737
function:   0302
/dev/ttyUSB0 <-- message: 1 length: 22
data:       12 34 56 79 00 00 ff ff 69 00 fe ff ff ff 37 c7
data:       12 73 02 03 b6 db
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   006a
source:     fffffffe
dest:       7312c737
function:   0302
/dev/ttyUSB0 <-- message: 2 length: 22
data:       12 34 56 79 00 00 ff ff 6a 00 fe ff ff ff 37 c7
data:       12 73 02 03 b9 9f
RS485 master ack timeout
^Cclosing /dev/ttyUSB0
closing <stdout>

Hope that helps.

jbuehl, 2018-05-09:
It looks like the messages are being formed correctly, so the possibilities would be either a hardware problem or an invalid inverter identifier.  Sometimes the inverter ID has the 9th bit from the left set but I can't remember when this occurs.  You could try specifying the slave as 7392c737 instead of 7312c737 and see if that works.

ver-luisant, 2018-05-09:
Unfortunately, no success with the modified identifier.
I will order and try a different usb-RS485 adapter.

ver-luisant, 2018-06-01:
Good news. I finally succeed in getting the key.
I used a different RS485 adapter (similar to this one: https://www.ebay.com/itm/USB-to-RS485-TTL-Serial-Converter-Adapter-FTDI-interface-FT232RL-75176-Module/201414952804).
This adapter has two small tx/rx leds, very handy when debugging the connection.

I probably broke the previous adapter.
The transmission was succesful only on RS485-1 channel (possibly, I also broke RS485-2 interface on the inverter communication board...).
And as suggested by @felixmartens, I had to send the commands multiple times and separately. 

Thanks all for your help.

mutesplash, 2022-06-03:
With the HD Wave SetApp inverters, they have a USB-A port on the board, not pads.  How does this connect to the computer, since A to A is not a thing?  Do I have to connect a USB-A to RS-232 adapter to the inverter and then inverter will automatically recognize it and use it?

scr34m, 2023-09-29:
I've got an SE7K inverter with USB A port and used a cable to connect with a laptop but no device showed up under the USB tree. On the RS485 port i receive data, but unable to use for key extraction.

scr34m, 2023-09-29:
Well i found a small video https://www.youtube.com/watch?v=P_ybrwLRDCo which means the USB A port is only used for firmware update not for communication hmm

scr34m, 2023-09-30:
My SE7k already communicating with prodssl server so it's not possible to read the key over RS485. If i sent the command after the first 0x0500 function then inverter responded 00000000 value data for the 4 parameters.