jbuehl/solaredge #46: Solaredge community or, does anybody want my data?

oliv3r, 2017-08-24:
Hey all. It seems that for this cool little tool, most of the communication is happening via issues (as can be seen in #8), as if they where a forum.

Is there however a small community/forum somewhere where findings here are discussed a bit more in depth? If not, and this is the place to be, great :)

I just got a new PV setup with a new HD Wave SE5000H. It has not ever been connected to the internet yet, I have connected it via network to a local semonitor on a dedicated network interface. But since I just received my USB 485 dongle today, I started to actually pull data off the unit.

From this data set, I'm getting a lot of 'unknown...' and 'undeciphered_data' and 'meters_...' which seem to be little documented. My question is, is the json dump to any interest, specifically the deciphered data? Or is there only any true value in pcap files/full dumps?

P.S. making note of the meters_0x0022 that I mentioned above, I see this one spamming the whole night, even while the unit is in night mode. Is that expected/new/curious info (it is being decoded reasonably, but I have no meter connected at all).

jbuehl, 2017-08-25:
Posting issues seems to be the only way to collaborate on GitHub.  I am not aware of any other forums.

oliv3r, 2017-08-25:
@jbuehl thanks for the heads up then. Issues it is :)

I have since done a new dumping session using the rec file. What's the easiest way to decode/dump it (this being RS485 master data).

Also, and I'll probably file a new bug report for that, unless I am doing something wrong. Very occasionally, I get a checksum verification failed error, and everything stops from then on. The only way I can fix it, is to delete seseq.txt and restart the monitor. Then it continues fine until the next checksum fault ...

jdwhite, 2018-08-26:
I'm seeing the same issue with semonitor and RS485, consistently just after sundown I get this:
```
Aug 25 19:39:13 Checksum error. Expected 0x1043, got 0x797f
Aug 25 19:39:43 Data length doesn't match inverse length
Aug 25 19:48:44 Data length doesn't match inverse length
Aug 25 19:48:44 Data length doesn't match inverse length
Aug 25 19:48:44 Data length doesn't match inverse length
Aug 25 19:53:14 Data length doesn't match inverse length
```
and several more over the next few hours.

I restart semonitor in the morning and semonitor will read the backlog from point where I got the check sum error until current time.

Was there ever a new issue opened for the "stopping after checksum failure" issue?

oliv3r, 2018-08-27:
Hmm, I never looked into when it fails to decode, but maybe it is around sunset? I'll try to keep an eye out for this!