jbuehl/solaredge #47: Event data very incomplete

oliv3r, 2017-08-24:
While I do not have pcap files to offer, I do see 'odd' parsing off the Events on my HD Wave inverter.
For example:
`  "events": {
    "xxxxxxx": {
      "Type": 7,
      "Event3": "Tue Aug 22 18:40:25 2017",
      "Time": "18:44:17",
      "Date": "2017-08-22",
      "Event2": 1,
      "ID": "xxxxxx",
      "Event1": "Thu Jan  1 01:00:00 1970"
    }
`
First, in my short list of dumps, I see a lot of different type's here, unlike the described 0 or 1. I've seen 118, 114, 7 and 0.

Secondly, The very first Event3 I pulled from the device (which has never connected to a SE server yet) was at epoch + 1, with a time/date close to the above (e.g. an accurate timestamp). Some random timestamp later (about 2 hours) Event3 then was jan 9th, 10:20:57 1970, e.g. some larger number. A few hours after that, the timestamp was epoch + 3. I even saw it at 0 once.
The same happens to Event1, jumps around a lot during history. Event2 I've often seen at epoch + 1, 2 or 3. But sometimes also as an actual timestamp.

So maybe if the description in 
https://github.com/jbuehl/solaredge/blob/ff542b31cdc9d8ef48cc6705a7fa4171e64c52b8/seDataParams.py#L6 is accurate for Type = 0 or 1, but for the other types, it doesn't seem to hold a timestamp often.

Type 118, Event1, 2 and 3 are near epoch, e.g. very low int's
Type 114, Event3 is a low-ish number (jan 9th 1970), Event2 was 502133 and Event1 Jan 3rd 1970.
Type 7, Event3 holds a timestamp - almost 4 minutes. Event1 and 2 are low ints.
Type 0, Event3 = 0, Event1 and 2 exact timestamp match.

jbuehl, 2017-08-25:
The Event message is one that isn't well understood.  My inverters send them when they are waking up and going to sleep, but I don't know what any of the fields mean.  If you are able to figure out more, please contribute to the project.

oliv3r, 2017-08-25:
@jbuehl From our data combined it seems that they are different between the various inverters then. This was my attempt in starting to contribute ;) starting the discussion.

I only have a few days worth of data so far, I'll collect the event logs and start to see anything useful there. But since I don't have a solar-edge portal account (nor data going there) it will be tricky to figure out the intent.

I wonder If I can sign up and start sending fake data to see what changes in the SE portal.

ksaihtam, 2017-08-27:
114 is some statistic data, send every day once

118 is a wakeup event with the sleep time in seconds and some counters that will reset after a restart

```
2017-07-23 21:23:03 (118, 45546, 3, 0, 1, 0)
2017-07-23 21:28:21 (118, 317, 3, 1, 2, 0)
2017-07-24 21:04:14 (118, 84949, 3, 2, 3, 0)
2017-07-24 21:09:49 (118, 334, 3, 3, 4, 0)
2017-07-25 05:42:27 (118, 30760, 3, 4, 5, 0)
2017-07-25 07:21:26 (118, 5938, 3, 5, 6, 0)
2017-07-25 20:39:00 (118, 47848, 5, 6, 7, 0)
2017-07-25 20:49:55 (118, 655, 3, 7, 8, 0)
2017-07-26 21:14:55 (118, 87898, 3, 8, 9, 0)
2017-07-26 21:21:11 (118, 375, 3, 9, 10, 0)
2017-07-27 05:24:32 (118, 29004, 3, 10, 11, 0)
2017-07-27 21:23:39 (118, 57542, 3, 11, 12, 0)
2017-07-28 05:18:27 (118, 28491, 3, 12, 13, 0)
2017-07-28 21:15:50 (118, 29252, 3, 0, 1, 0)
2017-07-28 21:21:25 (118, 334, 3, 1, 2, 0)
2017-07-29 05:15:59 (118, 28474, 3, 2, 3, 0)
2017-07-29 05:25:16 (118, 557, 3, 3, 4, 0)
2017-07-29 21:18:15 (118, 57173, 3, 4, 5, 0)
2017-07-30 21:18:13 (118, 86397, 3, 5, 6, 0)
2017-07-30 21:23:30 (118, 315, 3, 6, 7, 0)
2017-07-31 05:21:08 (118, 28661, 3, 7, 8, 0)
2017-07-31 21:22:32 (118, 57678, 3, 8, 9, 0)
2017-08-01 05:33:55 (118, 29487, 3, 9, 10, 0)
2017-08-01 20:32:53 (118, 53933, 3, 10, 11, 0)
2017-08-01 20:40:49 (118, 474, 3, 11, 12, 0)`

oliv3r, 2017-08-30:
@ksaihtam thanks for the update. Unfortunatly, currently seMonitor doesn't output this much data for the events afaik.

We have event1, event2 and event3, where they are sometimes converted to timestamps. I've re-converted them to numbers, where they are sometimes indeed timestamps.

On my HD wave however, 114 is not statistics. Here is my output since that I have it.

Date:   2017-08-22      Time:   18:39:43        Type:   118
Date:   2017-08-22      Time:   18:44:17        Type:   7
Date:   2017-08-22      Time:   20:52:06        Type:   114
Date:   2017-08-22      Time:   21:05:49        Type:   118
Date:   2017-08-22      Time:   22:42:07        Type:   0
Date:   2017-08-22      Time:   22:47:07        Type:   114
Date:   2017-08-23      Time:   20:53:54        Type:   118
Date:   2017-08-23      Time:   22:46:30        Type:   0
Date:   2017-08-23      Time:   22:51:30        Type:   114
Date:   2017-08-24      Time:   20:53:35        Type:   118
Date:   2017-08-24      Time:   20:59:31        Type:   1
Date:   2017-08-24      Time:   22:40:02        Type:   0
Date:   2017-08-25      Time:   21:03:07        Type:   118
Date:   2017-08-25      Time:   21:32:50        Type:   1
Date:   2017-08-25      Time:   22:57:17        Type:   0
Date:   2017-08-26      Time:   20:46:44        Type:   118
Date:   2017-08-26      Time:   22:39:18        Type:   0
Date:   2017-08-26      Time:   23:14:04        Type:   1
Date:   2017-08-27      Time:   20:50:02        Type:   118
Date:   2017-08-27      Time:   22:44:17        Type:   0
Date:   2017-08-27      Time:   23:22:57        Type:   1
Date:   2017-08-28      Time:   20:58:20        Type:   118
Date:   2017-08-28      Time:   22:53:41        Type:   1
Date:   2017-08-29      Time:   20:35:47        Type:   118

The 22nd was installation day, I can't recall the power up and on/off switching sequence for that day exactly. I do remember that after dark, I had switched the unit off and removed the mains power. So in my case, 114 'could' be the switch off, and the 118 mains power loss (around 21:00). Power up at 22:42 does not sound unreasonable where the unit went into night-mode possibly at 22:51.

In my case, so far it seems 118 happens daily pretty much around 20:30 - 21:00. So i guess that's when the sun drops (i'll have to cross-reference it with the generated power for that time. The 0/1 events could be night-mode/standby, or something else all together.

How did you get your data output? I do have the 'rec' files saved so I can re-decode and attempt to extract more data.

oliv3r, 2017-09-08:
Having collected some more logs in the last 2 weeks, dumping the events json, I see the pattern repeated above, except with the addition of one extra 114, which happend exactly at the time/date that I turned on/off the power. I can't remember whether the timestamp was a power off or power on.

So it appears that 114 is either a mains loss or a mains returned. Sadly however, I just did a test, and it does not show in the logs at all. I do see some Type 1 events. I did however flip the switch to off before cutting the power. Repeating the test while keeping the switch on, did not yield the 114 either. The other thing could be, is that 114 is 'intrusion detection', as I did turn the power off to open the cover. Though I can't imagine how they are detecting this when all power is removed (except for the RC battery).

I would not be surprised that 118 in combination with 0 or 1 is night mode enabled. I do have one odd measuring point (which happened early in the morning. The sun was up and the system awoke (no long entry fort hat0 but at 07:08 it came really dark (rain) and there is a 118 entry for that. Which seems like a 'night mode enabled' thing?

Date:   2017-09-08      Time:   20:12:14        Type:   118
Date:   2017-09-08      Time:   20:55:49        Type:   1
Date:   2017-09-08      Time:   21:40:39        Type:   1
Date:   2017-09-08      Time:   21:57:38        Type:   1

But i'm now quite convinced that Type 118 is initiation of night mode and, while less certain, Type 0 and 1 appear to be 'going to off mode' whether this is caused by night mode or by a flipped switch.

Can't yet see what the difference is between the two yet as they are not always in the same order either.

oliv3r, 2017-09-18:
After looking at my inverter display (coincidentally at the right time) i saw around 19:52-ish 2.5 Volts DC. At 19.55 it was 0 Volt DC. So my expecation is that at 19.49 it was either 17 Volts DC (which is the number of my inverters) or at the 'safe' voltage point.
Looking at my log, I see the following event:
Date:   2017-09-18      Time:   19:49:24        Type:   118
So unless my clock is off by about 5 minutes (not unlikley, is the clock ever changed over RS-485?)
The 118 event is thus either 'safe voltage mode' event or the '1 Volt per optimizer safety'.

ksaihtam, 2017-09-19:
You need the content/payload of the event message to get closer to the meaning. In my opinion the content of 114 is some kind of statistic.

oliv3r, 2017-09-21:
Yeah, i'm trying to see how i can read rec files so I'll get deeper into this in the fure.

oliv3r, 2017-10-07:
btw, 114 afaik is a power off event (circuit breaker off/on) but I wrote above that I tried and did not get the response. since aug. 22 i've only gotten the following 114 events:
Date:   2017-08-22      Time:   20:52:06        Type:   114
Date:   2017-08-22      Time:   22:47:07        Type:   114
Date:   2017-08-23      Time:   22:51:30        Type:   114
Date:   2017-09-05      Time:   23:33:24        Type:   114

118 i still get every day whenever the unit goes into night-mode.

Event 7 still has only shown once, after the very first start up
Event 0 and 1 still happen a few times a day, but since 26th of sept. i now get the 0 event in the morning, rather then late at night.

Still haven't dug into the actual data yet though. I should see if there are defines for this in the .net tool ...

ksaihtam, 2017-10-09:
The content of 114 are 5 unknown numbers. Number 3 is the sum of number 1 and 2. Number 4 and 5 is 0 here. Looks like some counters with an unknown unit. I see it in the morning and evening.
I get the 118 in the morning and evening (sometimes twice). The first number is the time since the last 118 message in seconds.
The 0/1 event are almost both together with a time difference of ~1 minute in the morning and evening.

Type|Param 1| 2 | 3 | 4 | 5
---- | ---- | ---- | ---- | ---- | ----
0| Timestamp | Timestamp | ? | ? | ?
1| Timestamp | ? | Timestamp | ? | Timestamp
114| Nr 1 | Nr 2 | Nr 1 + Nr 2 | ? | ?
118| Seconds since last msg | ? | ? | ? | ?