jbuehl/solaredge #51: Can I submit data to the solaredge portal?

oliv3r, 2017-09-12:
For debugging reasons, I want to submit data to the solaredge portal. I don't want to actually connect my inverter to the portal however. I wish to replay data to the portal (I'm extracting data via RS485 right now).

I'm expecting the answer to be no, this is not possible, so I'll likely build this feature myself, just wondering if anybody is aware of the gotcha's.

jbuehl, 2017-09-13:
If you want to emulate the behavior of an inverter for sending performance data it should be possible.  You would connect to the socket on the portal, send a 0500 data message, and wait for the 0020 acknowledgement message.  The portal should accept unencrypted messages, unless they have discontinued that ability.  If you want to send encrypted messages, that will be more difficult since there isn't an encryption routine in this project.  You could just capture the 003d encrypted messages and send them, but you would have to precede it with a 0503 message containing the key.

There will also be a number of messages that the portal sends the inverter to query for other data or to cause it to do things, which maybe you can ignore, or maybe not. 

If you get this working, please contribute it to the project.  Good luck.

oliv3r, 2017-09-13:
Naturally the intention is to contribute back :)

it just takes some time to get my head wrapped around things. I will however only work on the unencrypted data however. The encrypted bits will make it (too) tricky to start with.

What i'm looking at now is to open a json file that I saved (seems more easier and logical, otherwise take a .rec dump or a tcpdump file) and convert the records that would turn into a 0500 into a 0500 message again and submit to the server. The 0020 is a good tip. I'll 'dump' all other received data for later analysis I suppose. Should be interesting what it's trying to send ...