jbuehl/solaredge #52: Push interval

vandalon, 2017-09-15:
Hi,

I've read somewhere that there is a way to change the interval the solaredge pushes data over serial? Is this true and does somebody know how to do this?

Regards,
Joris

oliv3r, 2017-09-15:
Hey @vandalon 
I'm also pulling via RS485, and since the RS485/semonitor is the master, it probes the inverter quite often. The inverter however only responds whenever it has data. Some data for example is only available after an interval of 300 seconds.

When the inverter is the master, it pushes at these predetermined intervals.

What exactly are you after/trying to accomplish?

vandalon, 2017-09-15:
Hi oliV3r, thanks for the quick reply!

I would like to read data more often than every 300 minutes. 
It would be great to have at least once a minute the power production information of the inverter.

Jerrythafast, 2017-09-15:
Have a look at this comment:
https://github.com/jbuehl/solaredge/issues/8#issuecomment-204589160

Disclaimer: totally untested!

vandalon, 2017-09-15:
I actually saw that comment before but I did not get it working effectively. Mostly because of lack of knowledge I think 😊

vandalon, 2017-09-18:
Also, Mercury are the optimizers. I would like to get real time data from the inverter. If at all possible :)

ksaihtam, 2017-09-19:
I have test this command and it seems not to work with newer inverters and optimizers. I have send this command very often to one optimizer and finally he doesn't send no more data. After a power cycle of the inverter he finally came back.