jbuehl/solaredge #56: Decryption key not yet available

felixmartens, 2017-09-23:
After digging around some time with my new HD Wave Inverter SE5000H i managed to get some date over RS485 for the encryption key commands. Mots times the answer was empty so i issued every command on its own, multiple times until i got a response 144 with a valid data set. The 4 JSON Sets i got are:
```
{"data": {"type": 0, "value": 2136429196}, "command": 18, "response": 144, "sequence": 612}
{"data": {"type": 0, "value": 1195300718}, "command": 18, "response": 144, "sequence": 624}
{"data": {"type": 0, "value": 67575110}, "command": 18, "response": 144, "sequence": 692}
{"data": {"type": 0, "value": 1654055058}, "command": 18, "response": 144, "sequence": 636}
```

(i did realise that the third one h23b is only 8 digit long but it was replicable)

Piping this data into sekey i got:

> 8c5X577f6ed73e47461d07049Xe09662 

 (the X being a valid digit)

But if i use the keyfile to decrypt my tcpdump with command

`python solaredge/seextract.py capture.pcap | python solaredge/semonitor.py -k solaredge/73125Xa3.key -o monitor.json -d stdout -vvvv
`

I get the error as in the title: 

> Sep 23 17:22:13 <stdin> --> message: 26 length: 66 
> Sep 23 17:22:13 data:       12 34 56 79 2c 00 d3 ff 5d 00 a3 57 12 73 fe ff 
> Sep 23 17:22:13 data:       ff ff 3d 00 47 c2 15 62 94 5e f4 b4 84 6c 01 51 
> Sep 23 17:22:13 data:       db a3 c7 37 8d 09 6f df 52 62 4d 22 a3 95 c4 3b 
> Sep 23 17:22:13 data:       d3 63 0c ad c9 44 46 77 20 d3 04 e1 8c 16 24 df 
> Sep 23 17:22:13 data:       20 e2 
> Sep 23 17:22:13 dataLen:    002c 
> Sep 23 17:22:13 dataLenInv: ffd3 
> Sep 23 17:22:13 sequence:   005d 
> Sep 23 17:22:13 source:     73125Xa3 
> Sep 23 17:22:13 dest:       fffffffe 
> Sep 23 17:22:13 function:   003d 
> Sep 23 17:22:13 Decryption key not yet available 
> Sep 23 17:22:13 Ignoring this message 

Can anyone point where i made the mistake. Could the too short number on h23b be the problem? Has anyone else the new HD Wave inverter and successfully decoded the tcpdump?

vandalon, 2017-09-23:
Nope, you did not make a mistake. It takes a while before the server key is transmitted , and thus captured with tcpdump :)

felixmartens, 2017-09-24:
Thank you! I wanted to verify the key and everything to work and for this i just used a small dump :D

With around 12 hours dumped it worked....

closed 2017-09-24 by felixmartens

vandalon, 2017-09-24:
once it works you can read all the pcap info from before as well. just dump it into one data file and read it :)