jbuehl/solaredge #72: Unable to get the key over RS232..

travistabbal, 2017-11-28:
I'm trying to get the key and it's stalling out. I tried just using the seextract command with -vvvv on both inverters, I'll include the output below. 

```
sudo python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f15ec9f -t 2 -vvvv  /dev/cu.usbserial-A97ZYV9
following: True
baudrate: 115200
verbose: 4
record: None
type: 2
xerror: False
commands: [['12', 'H239'], ['12', 'H23a'], ['12', 'H23b'], ['12', 'H23c']]
outfile: stdout
master: False
datasource: /dev/cu.usbserial-A97ZYV9
slaves: ['0x7f15ec9f']
interface: None
follow: False
logfile: stderr
keyfile: None
port: 22222
append: False
update_file: None
opening /dev/cu.usbserial-A97ZYV9
dataLen:    0002
dataLenInv: fffd
sequence:   000e
source:     fffffffe
dest:       7f15ec9f
function:   0012
/dev/cu.usbserial-A97ZYV9 <-- message: 1 length: 24
data:       12 34 56 79 02 00 fd ff 0e 00 fe ff ff ff 9f ec
data:       15 7f 12 00 39 02 3f 9a

^CTraceback (most recent call last):
  File "solaredge/semonitor.py", line 345, in <module>
    doCommands(dataFile, args.commands, recFile, args.type, args.slaves[0], passiveMode, keyStr, args.master, args.follow)
  File "solaredge/semonitor.py", line 161, in doCommands
    msg = se.msg.readMsg(dataFile, recFile, passiveMode, inputType, following)
  File "/Users/travis/se/solaredge/se/msg.py", line 73, in readMsg
    msg = readBytes(inFile, magicLen + msgHdrLen, following)
  File "/Users/travis/se/solaredge/se/msg.py", line 103, in readBytes
    inBuf = inFile.read(length)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

```
sudo python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7F1636E3 -t 2 -vvvv  /dev/cu.usbserial-A97ZYV9
following: True
baudrate: 115200
verbose: 4
record: None
type: 2
xerror: False
commands: [['12', 'H239'], ['12', 'H23a'], ['12', 'H23b'], ['12', 'H23c']]
outfile: stdout
master: False
datasource: /dev/cu.usbserial-A97ZYV9
slaves: ['0x7f1636e3']
interface: None
follow: False
logfile: stderr
keyfile: None
port: 22222
append: False
update_file: None
opening /dev/cu.usbserial-A97ZYV9
dataLen:    0002
dataLenInv: fffd
sequence:   000b
source:     fffffffe
dest:       7f1636e3
function:   0012
/dev/cu.usbserial-A97ZYV9 <-- message: 1 length: 24
data:       12 34 56 79 02 00 fd ff 0b 00 fe ff ff ff e3 36
data:       16 7f 12 00 39 02 42 6f

^CTraceback (most recent call last):
  File "solaredge/semonitor.py", line 345, in <module>
    doCommands(dataFile, args.commands, recFile, args.type, args.slaves[0], passiveMode, keyStr, args.master, args.follow)
  File "solaredge/semonitor.py", line 161, in doCommands
    msg = se.msg.readMsg(dataFile, recFile, passiveMode, inputType, following)
  File "/Users/travis/se/solaredge/se/msg.py", line 73, in readMsg
    msg = readBytes(inFile, magicLen + msgHdrLen, following)
  File "/Users/travis/se/solaredge/se/msg.py", line 103, in readBytes
    inBuf = inFile.read(length)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

Both inverters do the same thing, so I wonder if the data they report is the data I need and it just doesn't return? 

The chained command fails... 

```
python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f1636e3 -t 2  /dev/cu.usbserial-A97ZYV9 |python solaredge/utilities/sekey.py -o 7f1636e3.key
Traceback (most recent call last):
  File "solaredge/semonitor.py", line 345, in <module>
    doCommands(dataFile, args.commands, recFile, args.type, args.slaves[0], passiveMode, keyStr, args.master, args.follow)
  File "solaredge/semonitor.py", line 155, in doCommands
    seq = nextSeq()
  File "solaredge/semonitor.py", line 193, in nextSeq
    with open(SEQ_FILE_NAME, "w") as seqFile:
IOError: [Errno 13] Permission denied: 'seseq.txt'
Traceback (most recent call last):
  File "solaredge/utilities/sekey.py", line 24, in <module>
    data = json.loads(inFile.readline().rstrip("\n"))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```

jbuehl, 2017-11-28:
Try pulling the code again.  It looks like you are using a version from a few days ago that was broken.  There have been quite a few recent changes that should have fixed some problems.

I notice in the second example you are running it without sudo and it is getting a permission error on a file.

travistabbal, 2017-11-28:
Thanks, I'll try that. I have tried with and without sudo, but didn't think
to grab a more recent version.

travistabbal, 2017-11-29:
Sadly, still the same. "git pull" shows no updates now, though it did pull updates last night.

jbuehl, 2017-11-29:
I made the last update yesterday before you opened this issue so whatever you got last night is up to date.  I ran the same command you are trying and it worked.  What hardware connection to the inverter are you using?

jbuehl, 2017-11-29:
You should get this:
```
[root@power ~]# python solaredge/semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 7f104920 -t 2 -v /dev/ttyUSB0
append: False
baudrate: 115200
commands: 12,H239 12,H23a 12,H23b 12,H23c
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stderr
master: False
outfile: stdout
ports: 22222,22221,80
record: None
slaves: 7f104920
type: 2
updatefile: None
verbose: 1
xerror: False
opening /dev/ttyUSB0
{"data": {"type": 0, "value": 589666785}, "command": 18, "response": 144, "sequence": 149}
{"data": {"type": 0, "value": 1203494388}, "command": 18, "response": 144, "sequence": 150}
{"data": {"type": 0, "value": 1869558741}, "command": 18, "response": 144, "sequence": 151}
{"data": {"type": 0, "value": 2016574999}, "command": 18, "response": 144, "sequence": 152}
closing /dev/ttyUSB0
closing <stdout>

```

travistabbal, 2017-11-29:
I'm using RS232 with an FTDI adapter. Just RX/TX/GND with RX/TX swapped on
one end. I believe I got nothing at all when I tried RX->RX. Does the
inverter want true RS232 voltages? My devices only do TTL (0-5V). I have
some converters coming as I have a switch that works like that. I can also
try RS485 if that should work.

jbuehl, 2017-11-29:
Yes, it wants RS232 voltages.  If your inverter has the mini-USB connector you should be able to connect that directly to your Mac and it will show up as a FTDI serial device.

travistabbal, 2017-11-29:
Thanks for the clarification. Sadly, my inverters do not have the USB
connector, so I'll have to wait for the voltage converters to get here.

JustApu, 2017-11-29:
Travis,

Just double-checking your pin-out.

SolarEdge to DB9
- RX to pin 2 
- TX to pin 3
- G to pin 5

I used a FTDI USB to serial adapter on a Raspberry Pi to get my keys successfully from three SE10000 inverters back in August. (No USB port on my inverters either.)

travistabbal, 2017-11-29:
No DB9. Just a FTDI dev board from Sparkfun. It works great for most
things, but is limited to TTL voltages as it's really for Arduino and such.
I have a DB9 based serial adapter, but it's also TTL. I found out the hard
way trying to talk to the switch I mentioned. The switch and cable are
good, as they work from a "real" serial port on a server I have in the
rack. Thanks for the tip though! Hopefully I'll be able to get this going
soon, just waiting on shipping.

ericbuehl, 2018-07-12:
hopefully you were able to get this going.  i'm going to close this issue but please reopen if you're still having trouble

closed 2018-07-12 by ericbuehl