jbuehl/solaredge #74: For the newly initiated

jurijpar, 2017-12-31:
We have just had a new solar array installed and attached to a Solaredge SE 11400A inverter.  Like many before me I found out that to get the data from the system I would have to go through the solaredge server.  So that's how I ended up here.  On our particular inverter, there does not seem to be a place to set the IP address (with no connection to a router).  Going to the LAN setting it shows a Modbus connection  associating it with a TCP port (in this case 502).   So question (1) is whether others see this and have figured out a way to set the ip address, while the other question is whether there is a step by step description of how to implement the monitoring for people who are not quite as skilled as those on this "forum".   The description on this url https://github.com/jbuehl/solaredge  talk about acquiring the encryption key over an RS232 interface.  My system does not have an RS 232 interface which adds to the complication

Many thanks, @jurijpar

JustApu, 2018-01-01:
@jurijpar - Does your inverter get an IP address via DHCP? I believe that's the factory default setting. Also, did you open the cover? I believe all the inverters have RS232 but it may be a three wire connection on the terminal block as opposed to a DB9 or Micro-USB port.

dragoshenron, 2018-01-01:
@jurijpar 
Are you referring to this inverter? https://www.solaredge.com/sites/default/files/se_hd_wave_inverter_installation_guide_na.pdf

In this case there is an undocumented microUSB port. You can see it on page 48, figure 21, between the ethernet and the PRI, on the top side of the board.

That USB port is actually an RS232 with a FTDI converter.

I hope this helps.

jurijpar, 2018-01-01:
@JustApu -  since the inverter is outside and currently its about 7 F  I tend to spend little time out there ;)  

I have taken the cover off, but currently to get access to the ethernet connector with the lid in, requires me to knock out a hole in the gland - and then to crimp on the RJ 45 connector, which I am not going to do given how balmy it is out there.  The installers were going to do that (they did an excellent job, even brushing down the roof from snow to install the racks, but when I complained about the fact that sending our data to Solaredge only to get it back from them and providing a plethora of personal information was not what I wanted to do, they promised they would get back to me with Solaredge's answer and would then complete the system.   

The inverter is shown here https://www.wholesalesolar.com/cms/solaredge-se10000a-us-revenue-grade-inverter-installation-manual-3800765590.pdf. 

Page 59  and later of this manual shows how to set up the network.  It also points to  

https://www.solaredge.com/sites/default/files/solaredge-communication_options_application_note_v2_250_and_above.pdf 

which indicates that setting up either a fixed or a dynamic IP address should be a piece of cake.  Alas, when I try to edit the configuration of the LAN address it sends me immediately to the modbus submenu allowing me to only set up a TCP port on that communication protocol.   Perhaps this is because there is no dhcp server at the other end.  Presumably this is set up so that the inverter can immediately call home and register itself.  So when it gets warmer and/or the installer pulls an ethernet connection into our house, I will hook this up to an internet disconnected dhcp server.     The only mention of an RS232 connector in the manual (see pdf above) is in figure 39 where it mentions an RS485/RS232 terminal block.  I presume that this means that depending on how I configure the receiving end the connection ends up being either a DTE/DCE connection for RS 232 or a bus connection for RS485 (I hope).  If anyone can clear this  confusion up that would be great.  I did see this article http://www.electronicdesign.com/what-s-difference-between/what-s-difference-between-rs-232-and-rs-485-serial-interfaces   which explains the differences but does not really help me figure out how to enable an RS 232 connection (which it seems from what I have read I will require to access a key for decrypting the data).

@dragoshenron  Thank you for your comment. Here's a High Res image of the Solaredge SE 11400A
[image]

I have removed identification marks but left as much of the original as I can. It is pretty high resolution. If you can see where there might be  an  RS232 or a USB to RS232 connection I would be very much obliged if you could point it out

In any case, many thanks to the owner of this project and to the kind folks who are helping me unravel this mystery.  I fail to understand how companies can be so arrogant as to force private data to their servers without really telling the customer that they are doing so anywhere in their sales literature

@jurijpar

JustApu, 2018-01-01:
@jurijpar - I hear you about the wonderful temperatures outside. Thankfully, my install was during the summer.

For what it's worth, I wasn't able to find a Micro-USB port on my three SE10000 inverters. But the green terminal block below and to the right of the display looks similar to the one on my inverters. That's where I got my RS-232 connection to do the key recovery.

[image]
[image]

__SolarEdge to DB9__ (then DB9 to a FTDI serial-to-USB adapter to the Raspberry Pi)
- RX to pin 2
- TX to pin 3
- G to pin 5

P.S. The RS-485 connections already on the terminal block are for the built-in revenue grade meter in my inverters. I didn't have to do anything with them.

jurijpar, 2018-01-02:
@JustApu  Perfect, many thanks for your guidance. 
 I did see that block  and was not clear whether the RS 232 connections were on it.  Your particular design has a very similar circuit board to mine so Im sure you are correct -  the markings on mine are missing but other than that they do look similar,  When it hits above 10 F tomorrow, I will pull the connector ;)

I would like to find a way to help create a polished application for this inverter that anyone can use easily in their own home without going out to someone's server.

The Solaredge approach clearly is an enticing way to go, but I am concerned about privacy and availability.
 
I mean look at the API document Solaredge provides it says 
_"Usage limitations are enforced to prevent abuse of the monitoring server API, and these limitations may be changed in the future without notice. Additionally, a request rate limit is applied to prevent abuse of the service. If you exceed the limitations, an error message appears in the monitoring server API. If the limitation is further exceeded, the system may temporarily be nonoperational, or your access to the monitoring server API may be blocked."_

An interesting approach: Solaredge takes our data, monitors our systems, improves them based on our data and then prevents access when we want more data.    I can understand why their servers need to be protected, but they would not need to be if  we could access our data at the point of generation - from the inverters in our own homes.

There is so much more that can be done with a community that can find new approaches and ideas from existing technology.  Look at SMS. This was set up in SS7 (Signaling System Seven) to control telecom equipment and convey information between engineers,  The public made that inestimably more useful by using it for much more than technical discourse.

Anyway, getting off my soapbox, I just want to say thank you again to this community for making this capability available and to you for this timely insight. I look forward to returning that help.
@jurijpar

jurijpar, 2018-01-02:
sorry did not mean to close this...

lewismc, 2019-08-22:
This is super helpful thank you for hashing it out folks.