jbuehl/solaredge #80: Modbus over LAN/TCP port 502

mcmellow, 2018-01-24:
In july 2016 SolarEdge published a document https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf on sunspec protocol. According this document you can read modbus over LAN/TCP  using the <local ip address> port 502.
In this mode, the inverter will still be communicating with the SolarEdge server so monitoring services and firmware updates will still be available. Is it possible to add the communication mode to semonitor.py?

Kind regards
McMelloW

jbuehl, 2018-01-28:
The Sunspec protocol does not include any of the module (solar panel) level data because that is proprietary to SolarEdge.  That's the reason this project was created (see the README).  There are a number of data logger products available that will read this data https://sunspec.org/sunspec-certified-products/.  If someone wants to write a module that processes this protocol I'll add it to the project, but it won't be able to produce all the data that semonitor.py does.

mcmellow, 2018-01-30:
Thanks, for your reply.
Forgive me my ignorance on this subject. I tried to read some data with semonitor.py from my SolarEdge Inverter SE3500. But without any success so far. Do I have to configure my inverter to special communication type/setting? It has WiFi and  ethernet LAN cabling installed and I am using a Raspberry Pi 2 to read the data.

lesinigo, 2018-06-23:
I have looked into using the Modbus over Ethernet (TCP) support of my SolarEdge HD Wave inverter, but I quickly found out that when you configure the inverter for that protocol it will wait for a connection for up to _120 seconds_ and if your modbus tcp client does not connect in time the inverter will just give up and never listen on TCP again. The same also holds true if your client is connected but goes down for whatever reason (eg. system reboot), it will have just 120 seconds to reconnect.

When that timeout passes by you won't be able to put the inverter in listening mode again if you only have an Ethernet connection: you will have to physically go to the inverter and reconfigure it.
I didn't check if it is possible to reconfigure Modbus over Ethernet through a serial (RS232 / RS485) connection, but uf you have serial you shouldn't need ethernet anyway.

This 120 seconds thing makes it unreliable (in my opinion) for unattended operations, so I just gave up the Modbus over Ethernet option and will try to use the RS485 connection.

Reference: https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf

dave92082, 2019-07-09:
For what it's worth, I'm not sure if the new inverter firmware has addressed the issue that @lesinigo was seeing however I was able to connect (and re-connect) via ModBus TCP reliably since my SE5000 was installed last week. 

I can still only have one device communicating to it at a time but aside from that, I have made many connections to try and scrape all data available. I created a Prometheus exporter to populate a Grafana dashboard with stats every 5s and it has been running for a week without issues.

Beyond the SunSpec data, I did find additional holding registers with data in the 60000 range however I have not been able to make a correlation if its module data or not. Feel free to check it out if you are looking for real-time inverter data (without modules): https://github.com/dave92082/SolarEdge-Exporter

marpi82, 2019-12-06:
There is also original Sunspec python library written by Sunspec team HTTPS://github.com/sunspec/pysunspec/ and it is working with SE inwerters as with the other Sunspec enabled inverters (meters, baterries etc.)

mulles, 2019-12-12:
@marpi82  did you run into the 120seconds issue by lesinigo ? 
@lesinigo According to https://github.com/jbuehl/solaredge/issues/124#issuecomment-470076155 their might by a way to remotely "reset" the setting ? Had no chance to try it out myself!

marpi82, 2019-12-13:
> @marpi82 did you run into the 120seconds issue by lesinigo ?
No, on my inverter I didn't have this issue. I can connect enytime I need. Sometimes I was disconnected couple days and made connection without problems

dave92082, 2019-12-13:
Same here... every time I tried to connect, I was able to grab a socket. I think the firmware on the later inverters have addressed that problem

mulles, 2019-12-13:
@dave92082 @marpi82 thx for sharing. Could you state, which Inverter Models you are using and maybe even firmware version? 
Or at least if they still have an lcd display? 

Much appreciated :)

dave92082, 2019-12-13:
@mulles 

My inverter still has the LCD screen. It is a SolarEdge SE5000 w/ CPU version 3.2221.0

Further details regarding how I access it and open the TCP socket can be found here:
[](https://github.com/dave92082/SolarEdge-Exporter)

mulles, 2019-12-13:
@dave92082 thx, that great news for me, even more that you have some code ready :) 

Last question: Even though reconnect after connection loss works now. The requirement to do the intial connection ( first connection ever) within 120seconds (2min) after activation Modbustcp via inverter screen is still mandatory?

dave92082, 2019-12-13:
@mulles, no, after my solar company installed the inverter it was at least 12 hours before I attempted to connect tot it from code. My network has also gone down (upgrades,etc) since then and I have never had a problem establishing a connection. 

To date I have never had a problem connecting to the inverter with a single socket. The only problem I have had is if I forget to disconnect my prometheus exporter from the server while I try to debug on on PC. It does not seem to handle multiple connections however single connections (regardless of the disconnected time) are no problem.

The inverter has never been powered down and I have been able to connect to it with no problems throughout all of my experimentation.

marpi82, 2019-12-13:
@mulles 
I have model without LCD.
For more details You can look at https://github.com/marpi82/sunspec_test/blob/master/modsunspec/result.txt. Only problem I have is wrong LL voltage level read from meter - I have created RMA on SE Support and waiting... becouse when I connect directly to meter by USB->RS485 converter all values are ok.

mihailescu2m, 2020-02-03:
Mine works quite well too - the screenless version.
I do have an issue connecting (using the sunspec python module) - if I try to read the data immediately after making the connection, I get an error - so I have to wait for 1s after making the initial connection before actual reading data. After that, I am reading new values every 15 seconds without problems.

I am trying to aggregate the data to look like the one in the official API call for inverter technical data (https://www.solaredge.com/sites/default/files/se_monitoring_api.pdf page 35). So far almost everything matches (minus cosPhi and groundFaultResistance which are missing) except the powerLimit % - modbus data has powerFactor % but I don't think they are the same. Maybe powerLimit is DC_Power*100/AC_Power? Thoughts?

Finally - I can get optimizer data at 15-min intervals with a script that parses the monitoring website (from Layout/Playback), so this could be a good alternative for monitoring since it's not invasive and doesn't require any extra hardware, all software.

dragoshenron, 2020-02-03:
@mihailescu2m 
The power factor *is* the cos phi. Same thing.
My bet would be that Powerlimit is simply the % relative to rated AC power.

mihailescu2m, 2020-02-03:
@dragoshenron 

I see - I saw cosPhi always 1 or 0 in the data from the official API, whereas PF is ~ 0.98 in the modbus readings...
wrt PowerLimit: it may be DC power % relative to rated power? powerlimit * rated AC power is always slighly bigger than AC power (especially mornings, when AC power is still 0 and status is THROTTLED, powerlimit is > 0).

dragoshenron, 2020-02-03:
Well, cosPhi can't be 0. It would mean that all your energy is sort of wasted (in layman words).
I won't make here a full description of active, reactive and apparent power but wikipedia is your friend.

About PowerLimit: it could be DC power but it would be a stupid choice. The PowerLimit is used to curtail power injection upto a given limit or, in other words, to avoid that you push too much power into the AC grid. So it should be linked to AC not DC.

mihailescu2m, 2020-02-04:
@dragoshenron 

my values from solaredge API had cosphi 0 when power was 0, and 1 when power > 0.
the values from modbus tcp shows it's mostly ~0.98 - seems dumb from them to convert it to an int (then e.g. 0.7 would be 1 as well). Maybe they reduce it to 1 decimal? (then 0.98 is rounded up to 1.0?)

powerlimit: after observations from today (10s modbus data aggregated to 5m vs 5m solaredge api data) it seems that indeed what they call powerlimit is DC power % relative to rated power. I would need to make more observations in low light conditions however, that's when i noticed the powerlimit is noticeably higher than ac power.

awulf, 2020-02-12:
I'm not having any luck with Modbus TCP on my SE10000H-AU000BWU4 (screen less) with firmware 4.7.26. 
It's configured for Modbus TCP on port 1502 and I can see port is open, I can connect a Modbus TCP client to it (have tried a few), and request addresses, but theres never a response, thus timeout occurs.
Have tried power cycling the inverter and connecting straight away (just in case it's the connect within 120s problem), but no luck.
Have contacted SolarEdge support, but haven't got a response yet and judging by what others have said, it's unlikely I'll ever will.
There is an Import+Export meter attached to RS485-1 (Meter 1, Device ID 2)
Anyone have any ideas?

atze09, 2020-03-01:
> I'm not having any luck with Modbus TCP on my SE10000H-AU000BWU4 (screen less) with firmware 4.7.26.
> It's configured for Modbus TCP on port 1502 and I can see port is open, I can connect a Modbus [...]

Have you've been successful in the meantime?
I have the same problem.
My Inverter is SE15k with Screen and Software Version is 3.2468.

awulf, 2020-03-01:
Yeah I got it working in the end by playing around in the settings via the following sequence:
1. Flick the switch on SolarEdge to P and connect your mobile device to the WiFi hotspot (password on the side of the inverter, or use the QR code + mySolarEdge app)
2. Open the browser and visit http://172.16.0.1 5 and click on the “Communication” menu item
3. Select “RS485-2”
4. Choose Protocol “SunSpec (Non-SE Logger)”, also under that set Device ID to 1 (mine was set to 5). Now some Modbus TCP requests were responding.
5. Change the Protocol over to “SolarEdge Master” (now a SolarEdge logger script was getting data too)
6. Change back protocol to “None”
7. Power cycle the inverter for good measure, and it was all still working

@atze09 rather than using SetApp, your screened version be via the screen controls.

atze09, 2020-03-02:
> Yeah I got it working in the end by playing around in the settings via the following sequence:
>
> 1. Flick the switch on SolarEdge to P and connect your mobile device to the WiFi hotspot (password on [...]

Seem to work.
Changed the rs485-1 to solaredge Meter,sunspec protocol and ID Device 1. Then changed Back to non Solar edge device and revenue Meter. Nie modbus responds and Home Assistant gets values. I will check it in Detail.
Thank you awulf for this awesome hint!

mihailescu2m, 2020-09-01:
Did anyone managed to control the inverter over modbus?

mihailescu2m, 2020-09-21:
Just want to report that inverter control over modbus is actually possible.
I managed to set the Active Power Limit and the inverter behaves as expected.
From documentation:

> [the active power limit] is set as the percentage of the inverter’s maximum power. For example: when setting the register of SE5000 to 20, it will limit the inverter to 1000W which is 20% of 5000W

zeropoint46, 2021-01-16:
> Just want to report that inverter control over modbus is actually possible.
> I managed to set the Active Power Limit and the inverter behaves as expected.
> From documentation:

Can you explain how you did this? I'm trying to do this same thing to control another inverter/charger.

dgilmour77, 2021-05-31:
> Yeah I got it working in the end by playing around in the settings via the following sequence:
>
>     1. Flick the switch on SolarEdge to P and connect your mobile device to the WiFi hotspot (password [...]

Thanks for this.  I have a slightly different setup: RS-485-1 and RS-485-2 are both in use.  I have what I believe is the first install of the SolarEdge whole home backup unit, delivered just a few months ago.  It ties two inverters together and allows connection directly into the full load of the home.  Initially Modbus TCP was giving timeout errors as it has for others commenting here.  The sequence that seems to restart Modbus TCP is 1. Setting RS-485-2 to SunSpec, 2. Changing the port on Modbus TCP (I just alternate between 502 and 1502) and 3. Finally returning RS-485-2 to SolarEdge leader (as it must be for my two inverters to aggregate and share reporting out to the SolarEdge monitoring site.  The problem: Modbus TCP dies about every 24 hours and starts timing out again, requiring the whole sequence to be repeated.  I am in touch with a SolarEdge engineer who is trying to help, and I have told him I suspect a firmware bug that I hope can be quickly fixed.  Will report my progress back here.

tias2001, 2022-06-28:
> Thanks for this. I have a slightly different setup: RS-485-1 and RS-485-2 are both in use. I have what I believe is the first install of the SolarEdge whole home backup unit, delivered just a few [...]

Hi - Any luck with this? I am getting the 24hour issue - To fix I login via the inverter hotspot - turn off modbus tcp - wait a minute and turn it back on - it starts working again after a few minutes.. But yeah every 24 hours is pretty annoying and there doesnt seem to be a way to fix it remotely either..

mihailescu2m, 2022-06-28:
@tias2001 and probably others too:

Starting with the firmware from Feb 2022, modbus TCP is disabled on WiFi interface. Found out this the hard way.
So if you're using WiFi, you need to switch to LAN.
My solution: I've bought a cheap second hand Airport Express (30$), placed it next to the inverter and set it to bridge mode. Then put a small LAN cable between it and the inverter, and configure inverter to use LAN instead of WiFi. You could the same with a RPi but the airport express is cheaper and doesn't need any configuration.

tias2001, 2022-06-28:
> @tias2001 and probably others too:
>
> Starting with the firmware from Feb 2022, modbus TCP is disabled on WiFi interface. Found out this the hard way. So if you're using WiFi, you need to switch to [...]

Yeah nah I am connected directly to the ethernet port (I only got my inverter 2 weeks ago so have always had to use ethernet) - @mihailescu2m does your ethernet stays up for more than 24 hours?

mihailescu2m, 2022-06-28:
Yes, there's no issue after 24h.

felixmartens, 2022-06-28:
Maybe a dumb comment but 
According to SE the ModbusTCP  port is closed if there is no active connection  for two minutes. I am using ModbusTCP for about two years without problems and can confirm this behaviour. Is it possible your "remote" PC restarts/hangs i. e. and in this  time there is no connection so ModbusTCP goes to sleep?

mihailescu2m, 2022-06-28:
@felixmartens I think the connection is closed if there is no activity in 2 minutes (e.g. like a timeout error if you don't ask for data after opening a connection). But the port accepts connections anytime.

felixmartens, 2022-06-28:
No it is indeed closed. You have to reactivate it by restarting ModbusTCP(disable and enable it again) The port won't accept connections otherwise and appears closed on port scan. 
I can reproduce this on both v3 and v4  Inverter aka with and without display.  

Btw: for me ModbusTCP is found in Ethernet Confirm and you don't have to change anything on Standard Modbus