jbuehl/solaredge #83: "meters" message contains no data

felixmartens, 2018-03-03:
I have two inverters (New HD Wave ones), a battery and a ModbusMeter installed. I can extract a useful "inverter" messages every once a while, but most of the time i recognize a "meter" message every 5min, which JSON Output only contains the correct Time/Data field and the rest of the fields are mostly zero, P2X is sometimes correctly read. Can someone point me to where i can start to debug the decoding so i can decrypt meaningful "meter" messages?

jbuehl, 2018-03-03:
It looks like it is done in datadevices.py in class ParseDevice_0x0022 starting at line 441.

felixmartens, 2018-03-05:
I found this too, but i am wondering how to find the right values at this point. It seems to me that the given "register"/offset etc isn't correct, but how should i best dissect the dump to get the right values. I already have compiled current Wireshark version including Solaredge plugin. I just don't know where to start.

Geoff99, 2018-05-30:
Hi @felixmartens 

I may be a bit late offering this suggestion, but if you wade through the explanations in README.ParseDevice.md you will find some hints about how you might (with a bit of tweaking of the code) get some (voluminous) debugging style output from ParseDevice itself, by putting it into "explorer" mode.  That can be a bit helpful in figuring out 

1.  what is the correct format for a field at a particular offset, and then 
2.  what is the appropriate interpretation of the field. 

Step 1) requires perseverance (because ParseDevice in explorer mode just reports almost every decoding of the field I could think of).

Step 2) requires detective work, comparing values from semonitor output against values from the inverter itself (or the SolarEdge website) from the *same* point in time.

The "meter" device (more precisely the 0x0022 message) is a pain to decode, because it there are multiple repeats of it for the same timepoint.  A field (`recType`) early in any particular 0x0022 message instance signals what interpretation is appropriate for the subsequent fields (offsets) in that message.  Depending on the `recType`, many of the fields appear to be unused (aka are always zero).  There are also some fields that do contain non-zero data which I was unable to decode / decipher even after considerable effort.

You can access the internal documentation about what fields actually do hold useful data via a couple of commands in an interactive python session

```
from seDataDevices import ParseDevice
print ParseDevice.itemDefs()
```
(Warning - I think the code in semonitor has been substantially tidied up since I last did that, you probably need a different import statement (the `ParseDevice` code itself now resides in `se/datadevices.py`

Sorry I can't help with Wireshark - I'm a novice user of it myself and have not managed to get the solaredge plugin to work for me yet.

Hope that helps a bit.

charlygolf, 2018-06-03:
for those trying to make sense of the 0x0022 messages my ideas about the meaning of the recTypes:

2: exported to the grid
3: consumption
4: imported from the grid
6: solar energy consumed on site (selfuse)
9: solar energy

My inverter outputs all these messages every 300sec, and has a consumption modbus meter attached. I am assuming you get other RecTypes with differently configured modbus meters.

Some obvious conclusions:
9 = 6 + 2  (whatever solar energy you don't use on site will be exported to the grid)
3 = 6 + 4 (all energy your site consumes is either solar or imported from the grid)

Geoff99, 2018-06-04:
In case it helps other in the future, this is what ParseDevice_0x0022 currently (June 2018) *knows* about the 0x0022 messages (as seen from an SE6000 inverter).  Just possibly (but I have no way of testing) these field definitions may help decipher the recTypes that are not currently recognised.

```
ParseDevice_0x0022 / meters_0x0022 parses data blocks with seType = 0x0022.
================================================================================
Items are:
Byte (Length)   Word   | Item Name                 
						: Meaning

____ (______)   ______ | _________________________ 
						: ________________________________________

0    (  4   )   0.0    | dateTime                  
						: Seconds since epoch

4    (  1   )   1.0    | recType                   
						: record Type, determines the interpretation of later fields
						: 3=Consumption,
						: 5=Grid Import/Export,
						: 7=Battery,
						: 8=Unknown, almost all 0,or very very small,
						: 9=PV production

5    (  1   )   1.25   | onlyIntervalData          
						: 1=only interval data has been reported, 0=lifetime data reported as well

6    (  4   )   1.5    | TotalE2Grid               
						: Wh, Lifetime energy exported to grid (*provided* onlyIntervalData flag is not set)
						: Wh. Total lifetime energy exported to grid when recType=5 (matches SE LCD panel value)
						: always 0, when recType=3, or 7, or 8, or 9 because onlyIntervalData=1 (True)

10   (  2   )   2.5    | AlwaysZero_off10_int2     
						: Padding

12   (  2   )   3.0    | Flag_off12_hex            
						: Flag
						: 0x0000->TotalE2Grid reported
						: 0x0080->TotalE2Grid not reported

14   (  4   )   3.5    | TotalEfromGrid            
						: Wh, Lifetime energy imported from grid (*provided* onlyIntervalData flag is not set)
						: Wh. Total lifetime energy imported from grid when recType=5 (matches SE LCD panel value)
						: always 0, when recType=3, or 7, or 8, or 9 because onlyIntervalData=1 (True)

18   (  2   )   4.5    | AlwaysZero_off18_int2     
						: Padding

20   (  2   )   5.0    | Flag_off20_hex            
						: Flag
						: 0x0000->TotalEfromGrid reported
						: 0x0080->TotalEfromGrid not reported

22   (  4   )   5.5    | Totaloff22_int4           
						: Unknown, probably an energy field  (*provided* lifetime flag is set)
						: Maybe a cumulative net value, it appears to decrease overnight when importing power
						: Wh. Unknown when recType=5 (generally increasing trend, but falls overnight)
						: always 0, when recType=3, or 7, or 8, or 9 because onlyIntervalData=1 (True)

26   (  2   )   6.5    | AlwaysZero_off26_int2     
						: Padding

28   (  2   )   7.0    | Flag_off28_hex            
						: Flag
						: 0x0000->Totaloff22_int4 reported
						: 0x0080->Totaloff22_int4 not reported

30   (  4   )   7.5    | Totaloff30_int4           
						: Unknown total energy field (*provided* lifetime flag is set)
						: monotonic increasing so far including overnight.
						: Maybe something like cum total consumption scaled by about 60%?
						: Wh. Unknown when recType=5 (steadily increasing trend)
						: always 0, when recType=3, or 7, or 8, or 9 because onlyIntervalData=1 (True)

34   (  2   )   8.5    | AlwaysZero_off34_int2     
						: Padding

36   (  2   )   9.0    | Flag_off36_hex            
						: Flag
						: 0x0000->Totaloff30_int4 reported
						: 0x0080->Totaloff30_int4 not reported

38   (  4   )   9.5    | Interval                  
						: Seconds, Time Interval

42   (  4   )   10.5   | E2X                       
						: Wh, Energy to X during the interval
						: always 0 when recType=3
						: Wh. Energy exported to grid during interval when recType=5 (whenever TotalE2Grid is static, E2X=0)
						: Wh. Energy into battery when recType=7 (matches battery(0x0030).EIn)
						: Unknown, when recType=8 (almost always 0, very occasionally has small +ive value)
						: Wh. PV energy production during interval when recType=9

46   (  4   )   11.5   | EfromX                    
						: Wh, Energy in from X during interval 
						: Wh. Consumption when recType=3
						: Wh. Energy imported from grid during interval when recType=5 (whenever TotalEfromGrid is static, EfromX=0)
						: always 0 when recType=7 (I expected E from battery but it does not match battery(0x0030).EOut)
						: always 0 when recType=8 
						: always 0 when recType=9

50   (  4   )   12.5   | P2X                       
						: W, Power output to X
						: always nan when recType=3
						: W. Power to grid at end of interval when recType=5 (whenever TotalE2Grid is static, P2X=0)
						: W. Power into battery when recType=7 (is nonzero when E2X is nonzero)
						: almost always 0 when recType=8 
						: W. PV power when recType=9

54   (  4   )   13.5   | PfromX                    
						: W, Power input from X
						: W. Consumption when recType=3
						: W. Power from grid at end of interval when recType=5 (whenever TotalEfromGrid is static, PfromX=0)
						: always nan when recType=7
						: always nan when recType=8
						: always nan when recType=9

```