jbuehl/solaredge #85: offtopic? How to set the clock of my inverter

ethpi, 2018-03-15:
Hi,

since a few days the clock of my inverter (SE4K) is wrong (about 12h late). Does anybody know how to set it? 

Automatically update is activated, if I set communication to LAN, the clock is set via internet. But since I unplugged my inverter in november 2017 from internet and log the data via RS485 and semonitor.py the automatically update can't work.

I changed the internal battery (CR2032), but it seems, that this doen't solved the problem, because after two days the clock is wrong again. I can notice that the clock does not jump duing night.

Does anybody has a tip for me?

Many thans, Jan

jbuehl, 2018-03-18:
The inverter periodically asks the server that it is talking to for the current time, so the system time of the hardware you are running semonitor.py on needs to be set to the correct local time.

ethpi, 2018-03-23:
Thanks.

Did I understand correctly? If I connect my inverter via RS485 and semonitor.py to my rapsberry pi I have to check the system time of the raspberry. If this system time is correct, the inverter gets the correct time automatically.

Maybe that the timejump occured while booting the raspberry, when its systemclock wasn't updated so far. Although I checked if the uptime of the raspberry is more than 5min, I insert now the command "timedatectl set-ntp true" before starting semonitor.py.

I check this attempt for a few days.

jbuehl, 2018-03-23:
Also make sure that your time zone is set correctly in the RPi with `timedatectl set-timezone <time zone>`

ethpi, 2018-04-20:
Now everything works fine. 

I think the problem was, that I normally boot my rpi daily at about 7 a.m. In winter the inverter collected no data before this time. But in spring it collects data before booting and tried to send it instantly to the rpi, while its time wasn't set correctly.

closed 2018-07-12 by ericbuehl