jbuehl/solaredge #89: Ethernet active, DHCP failed and keyerror: broadcast

B-pek, 2018-05-10:
Hi,

I am trying to parse data by using the ethernet active mode. So far I have obtained the ecnryption key, which consists of 8 characters. 

Semonitor.py is running on a raspberry and is connected to the inverter through the ethernet interface. 
I am running the semonitor.py with this following command:

sudo python semonitor.py -o yyyymmdd.json -n eth0 -k ********.key

After running semonitor.py I got these following errors as you can see in the pictures I have taken.

[image]

The first picture is giving two errors, keyerror:2 and keyerror:broadcast

[image]

The second picture is from the inverter where it shows that the dhcp failed.

Can anyone tell me if I made a mistake or do you have a suggestion on how to solve this?

JustApu, 2018-05-10:
FWIW, my keys are 32 characters long. The 8 characters you list above look more like the inverter's serial number.

jbuehl, 2018-05-11:
I confirmed that it should work with Raspbian so no need to switch to Arch.  Here's how have it set up.  The built in ethernet interface of the RPi (eth0) is plugged into my network.  I have a USB ethernet device for the second interface (eth1) that only connects to the inverter.

`inverter -------- raspberry pi --------- network
`
```
# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.154  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2605:e000:141d:43f3:ba27:ebff:feb2:9641  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::ba27:ebff:feb2:9641  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:b2:96:41  txqueuelen 1000  (Ethernet)
        RX packets 1926776  bytes 312474125 (297.9 MiB)
        RX errors 0  dropped 34589  overruns 0  frame 0
        TX packets 615368  bytes 99651278 (95.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.1  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::9eeb:e8ff:fe0c:e865  prefixlen 64  scopeid 0x20<link>
        ether 9c:eb:e8:0c:e8:65  txqueuelen 1000  (Ethernet)
        RX packets 35140  bytes 3675780 (3.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 26483  bytes 1819497 (1.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 4520  bytes 392283 (383.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4520  bytes 392283 (383.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
eth0 gets its IP address from the network DHCP server.  eth1 is configured with a static IP address.  Here's the config file for Arch.  It's done differently for Raspbian, but I'm not sure how to do it.
```
# cat /etc/netctl/eth1 
Description='A basic static ethernet connection'
Interface=eth1
Connection=ethernet
IP=static
Address=('192.168.2.1/24')
```
I run semonitor with the following command
`# python /root/solaredge/semonitor.py -vv -t n -n eth1
`

Here is the output to the system log.
```
May 11 09:38:10 solar systemd-networkd[251]: eth1: Gained carrier
May 11 09:38:10 solar systemd-timesyncd[223]: Network configuration changed, trying to establish connection.
May 11 09:38:10 solar kernel: asix 1-1.3:1.0 eth1: link up, 100Mbps, full-duplex, lpa 0x45E1
May 11 09:38:39 solar systemd[1]: Started SolarEdge monitor.
May 11 09:38:41 solar semonitor[27155]: append: False
May 11 09:38:41 solar semonitor[27155]: baudrate: 115200
May 11 09:38:41 solar semonitor[27155]: commands:
May 11 09:38:41 solar semonitor[27155]: datasource: network
May 11 09:38:41 solar semonitor[27155]: follow: False
May 11 09:38:41 solar semonitor[27155]: interface: {17: [{'broadcast': 'ff:ff:ff:ff:ff:ff', 'addr': '9c:eb:e8:0c:e8:65'}], 2: [{'broadcast': '192.168.2.255', 'netmask': '255.255.255.0', 'addr': '192.168.2.1'}], 10: [{'netmask': 'ffff:ffff:ffff:ffff::/64', 'addr': 'fe80::9eeb:e8ff:fe0c:e865%eth1'}]}
May 11 09:38:41 solar semonitor[27155]: keyfile: None
May 11 09:38:41 solar semonitor[27155]: logfile: stderr
May 11 09:38:41 solar semonitor[27155]: master: False
May 11 09:38:41 solar semonitor[27155]: outfile: stdout
May 11 09:38:41 solar semonitor[27155]: ports: 22222,22221,80
May 11 09:38:41 solar semonitor[27155]: record: None
May 11 09:38:41 solar semonitor[27155]: slaves:
May 11 09:38:41 solar semonitor[27155]: type: n
May 11 09:38:41 solar semonitor[27155]: updatefile: None
May 11 09:38:41 solar semonitor[27155]: verbose: 2
[... 13 lines cut ...]
May 11 09:38:53 solar semonitor[27155]: 192.168.2.255:68 <-- message: 4 length: 300
May 11 09:38:53 solar semonitor[27155]: waiting for DHCP message
May 11 09:39:04 solar semonitor[27155]: 192.168.2.2:1488 --> message: 1 length: 37
May 11 09:39:04 solar semonitor[27155]: 192.168.2.2:1488 <-- message: 2 length: 72
May 11 09:39:04 solar semonitor[27155]: connection from 192.168.2.2:1487 to port 22222
May 11 09:39:04 solar semonitor[27155]: waiting for DNS message
May 11 09:39:32 solar semonitor[27155]: <socket> --> message: 2 length: 22
May 11 09:39:32 solar semonitor[27155]: <socket> <-- message: 1 length: 22
May 11 09:40:08 solar semonitor[27155]: <socket> --> message: 3 length: 22
May 11 09:40:08 solar semonitor[27155]: <socket> <-- message: 2 length: 30
May 11 09:40:08 solar semonitor[27155]: <socket> --> message: 4 length: 506
May 11 09:40:09 solar semonitor[27155]: <stdout> <-- message: 1 length: 2200
May 11 09:40:09 solar semonitor[27155]: <socket> <-- message: 3 length: 22
May 11 09:40:28 solar semonitor[27155]: <socket> --> message: 5 length: 511
May 11 09:40:29 solar semonitor[27155]: <stdout> <-- message: 2 length: 3124
May 11 09:40:29 solar semonitor[27155]: <socket> <-- message: 4 length: 22
May 11 09:41:50 solar semonitor[27155]: <socket> --> message: 6 length: 511
May 11 09:41:50 solar semonitor[27155]: <stdout> <-- message: 3 length: 3124
May 11 09:41:50 solar semonitor[27155]: <socket> <-- message: 5 length: 22
May 11 09:42:08 solar semonitor[27155]: <socket> --> message: 7 length: 508
May 11 09:42:08 solar semonitor[27155]: <stdout> <-- message: 4 length: 2693
May 11 09:42:08 solar semonitor[27155]: <socket> <-- message: 6 length: 22
[... cut ...] how the inverter requesting an IP address which is provided by semonitor.py as 192.168.2.2.  The message from the inverter and the response at 09:39:04 are the inverter requesting the DNS address for the solaredge server prod2.solaredge.com which is provided by semonitor.py as the address of the RPi at 192.168.2.1.  The inverter then connects to semonitor.py on port 22222.  The messages and responses starting at 09:39:32 re thea data messages from the inverter that are processed by semonitor.py.

closed 2018-07-12 by ericbuehl