jbuehl/solaredge #91: SolarEdge protocol

tinuzz, 2018-05-28:
Hi @jbuehl, 

I don't have any issues with the software, but I have a question, if I may. 

How / where did you obtain the knowledge to decrypt and decode the messages from the SolarEdge hardware? Is this documented somewhere publicly? (I haven't been able to find it) or do you have documentation from private sources? Did you reverse-engineer it yourself? If the latter, how did you go about this?

I studied your code, and I have a fairly good idea now about how the encryption works and how telemetry data is packed in 0x0500 messages, but I have no idea how one could ever figure out the protocol just from looking at the messages. I am trying to fully understand the protocol, because I am working on a decoding tool for research purposes.

Would you mind elaborating on this subject a little bit? Thank you in advance!

Best regards,
Martijn.

jbuehl, 2018-05-28:
I have never seen any public documentation of the SolarEdge protocol.  In 2013 when I first installed my system I was able to reverse engineer the basic 0500 messages by capturing traffic from my inverters to the SolarEdge server, making educated guesses as to how to parse the data, and validating it by comparing the values to what was being displayed on the inverters.  At that time, none of the data was encrypted.  In December 2015 SolarEdge updated their software and began encrypting the performance data sent over the network.  This was first reported in issue #8.  Over the next 3 months there was a collaborative effort by several participants to make a number of improvements and eventually to figure out the encryption.  You can follow the progress of that discussion in the issue to see how it developed.

tinuzz, 2018-05-29:
Hi @jbuehl, thank you! #8 looks like a good read, I'll dive into it when I can. Thanks! I'll close this issue.

closed 2018-05-29 by tinuzz