jbuehl/solaredge #94: Ethernet data read timeout

PerlJoe, 2018-05-31:
Dear all,

I'm highly interested in this project. I've just setup my SE7k inverter and it is working, however I'm encountering problems I wasn't able to solve until now:

I'm trying to use the ethernet active scenario and I've set up a LXC container (ubuntu, xenial, amd64), I've implemented DNS redirection, cloned the GIT repository, installed the requirements, got the encryption key out of the inverter by using the RS232 connection and now want to get the data the inverter sends so the portal.

This is where I fail. I run (replaced my inverter serial with 7e123456):

python /opt/solaredge/semonitor.py -k /opt/solaredge/7e123456.key -s 7e123456 -t n -d /opt/solaredge/debug.log -vvvv -r /opt/solaredge/data.dat

Log output (192.168.120.63 is the IP of the inverter):

May 31 21:48:25 append: False
May 31 21:48:25 baudrate: 115200
May 31 21:48:25 commands:
May 31 21:48:25 datasource: network
May 31 21:48:25 follow: False
May 31 21:48:25 interface: None
May 31 21:48:25 keyfile: <open file '/opt/solaredge/7e123456.key', mode 'r' at 0x...>
May 31 21:48:25 logfile: /opt/solaredge/debug.log
May 31 21:48:25 master: False
May 31 21:48:25 outfile: stdout
May 31 21:48:25 ports: 22222,22221,80
May 31 21:48:25 record: /opt/solaredge/data.dat
May 31 21:48:25 slaves: 7e123456
May 31 21:48:25 type: n
May 31 21:48:25 updatefile: None
May 31 21:48:25 verbose: 4
May 31 21:48:25 xerror: False
May 31 21:48:25 opening network
May 31 21:48:25 waiting for connection on port 22222
May 31 21:48:25 waiting for connection on port 22221
May 31 21:48:25 waiting for connection on port 80
May 31 21:48:56 connection from 192.168.120.63:3577 to port 22221

May 31 21:50:57 Exception:
Traceback (most recent call last):
  File "/opt/solaredge/se/msg.py", line 129, in readBytes
    inBuf = inFile.read(length)
  File "/usr/lib/python2.7/socket.py", line 384, in read
    data = self._sock.recv(left)
timeout: timed out
May 31 21:50:57 /opt/solaredge/data.dat <-- message: 1 length: 4
May 31 21:50:57 data:       12 34 56 79
May 31 21:50:57 closing <socket>
May 31 21:50:57 closing /opt/solaredge/data.dat
May 31 21:50:57 closing <stdout>

After connection from ... the log hangs and then the failure occurs and the program terminates.

I've started a tcpdump on the container and I see the following traffic:

21:48:56.191206 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.120.31 tell 192.168.120.63, length 46
21:48:56.191240 ARP, Ethernet (len 6), IPv4 (len 4), Reply 192.168.120.31 is-at 00:16:3e:ed:24:b2, length 28
21:48:56.590310 IP (tos 0x0, ttl 64, id 2658, offset 0, flags [none], proto TCP (6), length 44)
    192.168.120.63.3577 > 192.168.120.31.22221: Flags [S], cksum 0x97c6 (correct), seq 81797180, win 2152, options [mss 538], length 0
21:48:56.590366 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    192.168.120.31.22221 > 192.168.120.63.3577: Flags [S.], cksum 0x71ce (incorrect -> 0xc8a7), seq 1801778790, ack 81797181, win 29200, options [mss 1460], length 0
21:48:56.974932 IP (tos 0x0, ttl 64, id 2659, offset 0, flags [none], proto TCP (6), length 40)
    192.168.120.63.3577 > 192.168.120.31.22221: Flags [.], cksum 0x4a0d (correct), seq 1, ack 1, win 2152, length 0

21:50:57.073387 IP (tos 0x0, ttl 64, id 17297, offset 0, flags [DF], proto TCP (6), length 40)
    192.168.120.31.22221 > 192.168.120.63.3577: Flags [F.], cksum 0x71ca (incorrect -> 0xe063), seq 1, ack 1, win 29200, length 0
21:50:57.073819 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.120.31 (00:16:3e:ed:24:b2) tell 192.168.120.63, length 46
21:50:57.073837 ARP, Ethernet (len 6), IPv4 (len 4), Reply 192.168.120.31 is-at 00:16:3e:ed:24:b2, length 28
21:50:57.110145 IP (tos 0x0, ttl 64, id 2660, offset 0, flags [none], proto TCP (6), length 40)
    192.168.120.63.3577 > 192.168.120.31.22221: Flags [.], cksum 0x4a0c (correct), seq 1, ack 2, win 2152, length 0
21:50:57.143412 IP (tos 0x0, ttl 64, id 2661, offset 0, flags [none], proto TCP (6), length 40)
    192.168.120.63.3577 > 192.168.120.31.22221: Flags [F.], cksum 0x4a0b (correct), seq 1, ack 2, win 2152, length 0
21:50:57.143448 IP (tos 0x0, ttl 64, id 28782, offset 0, flags [DF], proto TCP (6), length 40)
    192.168.120.31.22221 > 192.168.120.63.3577: Flags [.], cksum 0xe062 (correct), seq 2, ack 2, win 29200, length 0

So I assume the data connection is kind of there and working, the data just cannot be read.

Maybe someone can point me in the right direction...
Thanks in advance,
PerlJoe

PerlJoe, 2018-06-01:
Me again,
I now tried terminating the network connection on my linux laptop and I have the same result. I also tried with different python versions (had to convert iteritems() to items() in env.py as iteritems() is deprecated in python3) but I get the same error everytime I try.
Using the same client with the RS232 serial connection works and I can read registers.
I also thought my key might be incorrect, so extracted it again (same) and the key looks good (32 hex digits).
Thanks ,
PerlJoe

jbuehl, 2018-06-01:
Based on what tcpdump shows, the inverter is connecting but just not sending any data.  That wouldn't be due to any problem with how you are using semonitor.py or an invalid encryption key.  The app will wait indefinitely to receive data so I think the timeout is occurring at the socket level.  Maybe you could use tcpdump or wireshark to get more detail on what is happening there.

You should be able to look on the display of the inverter to see the status of the network connection.  The inverter has to be configured to send performance metrics to the network rather than RS485, but I would think there would be some sort of handshaking over the network connection that would happen either way.

FYI - you don't need to specify the -s argument when you are using network mode, although that is harmless.

PerlJoe, 2018-06-01:
@jbuehl You're the man! Yes, there was no server configured. After I did that (set it to LAN) I got data. :1st_place_medal: 
Thanks for the quick help. This is awesome! :-)
PerlJoe

PerlJoe, 2018-06-01:
Solved, closed.

closed 2018-06-01 by PerlJoe