jbuehl/solaredge #97: Decryption key not yet available

tomdk, 2018-06-15:
I had everything working and then suddenly everything got encrypted.. :) I added a rs232 cable, read my key and generated my key file.

Now when I execute the same command, with adding the -k option ofcourse, it doesn't work anymore but I see a message "Decryption key not yet available".

What does this mean? Currently I retrieve the data via a tcpdump on my gateway triggered remotely from a raspberry pi.

ssh user@172.16.100.1 "sudo /usr/sbin/tcpdump -i eth1 -U 'tcp and host 172.16.100.16' -w -" | python semonitor.py -k 73224C4A.key -

tomdk, 2018-06-16:
Is it possible that since I read the key via rs232, my Solar Edge only communicates with: pix.incognitoiajdflwe.com.22222.

IP of my Solar Edge is 172.16.100.16. I only see this traffic from tcpdump:

14:00:02.064052 IP target.myhome.eu.1380 > pix.incognitoiajdflwe.com.22222: Flags [.], seq 2576:2642, ack 727, win 2152, length 66
14:00:02.136378 IP pix.incognitoiajdflwe.com.22222 > target.myhome.eu.1380: Flags [.], ack 2642, win 64899, length 0
14:00:22.904028 IP pix.incognitoiajdflwe.com.22222 > target.myhome.eu.1380: Flags [P.], seq 727:793, ack 2642, win 64899, length 66
14:00:23.112710 IP target.myhome.eu.1380 > pix.incognitoiajdflwe.com.22222: Flags [.], ack 793, win 2152, length 0
14:00:27.907608 ARP, Request who-has target.myhome.eu tell USG, length 28
14:00:27.907805 ARP, Reply target.myhome.eu is-at 00:27:02:11:4c:4a (oui Unknown), length 50
14:00:51.841013 IP pix.incognitoiajdflwe.com.22222 > target.myhome.eu.1380: Flags [P.], seq 859:925, ack 3166, win 65098, length 66
14:00:52.044761 IP target.myhome.eu.1380 > pix.incognitoiajdflwe.com.22222: Flags [.], ack 925, win 2152, length 0

While I have no idea what pix.incognitoiajdflwe.com is, it seems it's the only traffic thats been showing up either since I read the key or since my data became encrypted..

jbuehl, 2018-06-16:
The inverter periodically sends out a message that contains part of the key needed for decryption.  If you start monitoring the data between these messages you will see the "Decryption key not yet available" message until the key message is received.  At some point you should see a message that says "Creating decryption object with key" after which it should start decrypting the data.  I forget how often the message is sent, but it shouldn't be longer than a minute or two.

tomdk, 2018-06-16:
@jbuehl is the way that I want it to work ever going to work then? I'm feeding all tcp traffic that I find directly to semonitor.py. I feel like it's always going to have just parts of the messages then?

jbuehl, 2018-06-18:
You can do passive ethernet monitoring like you want.  You just have to wait a few minutes before the data is able to be decrypted.  Note that you should not feed the captured data directly into semonitor.py without filtering out the TCP stream containing the SolarEdge data using a utility such as tshark as noted in the README.

`tcpdump -i eth1 -U 'tcp and host 172.16.100.16' -w - | tshark -r infile.pcap -T fields -e data | ./utilities/unhexlify.py | ./semonitor.py -k 73224C4A.key -`