jbuehl/solaredge #100: Data length doesn't match inverse length for Energy Output

fahimmac, 2018-06-25:
Good Morning everyone, I need a help regarding the Solaredge monitoring.

Recently i am getting this "Data Length does not match inverse length" from my inverter. I used to get Energy info with the following command (E-day, E-total). But now this command does not seem to be working anymore. Here is the raw information.

```
pi@logger ~/inverter/sedgenew $ sudo python semonitor.py -m -c 322 -s 07e1a16cc -d stdout -vvvv -t 4 /dev/ttyUSB0
Jun 25 13:39:03 debugEnable: True 
Jun 25 13:39:03 debugFiles: True 
Jun 25 13:39:03 debugMsgs: True 
Jun 25 13:39:03 debugData: True 
Jun 25 13:39:03 debugRaw: True 
Jun 25 13:39:03 debugFileName: stdout 
Jun 25 13:39:03 haltOnException: False 
Jun 25 13:39:03 inFileName: /dev/ttyUSB0 
Jun 25 13:39:03 inputType: 4 
Jun 25 13:39:03 serialDevice: True 
Jun 25 13:39:03     baudRate: 115200 
Jun 25 13:39:03 networkDevice: False 
Jun 25 13:39:03 sePort: 22222 
Jun 25 13:39:03 networkSvcs: False 
Jun 25 13:39:03 following: True 
Jun 25 13:39:03 passiveMode: False 
Jun 25 13:39:03 commandAction: True 
Jun 25 13:39:03     command: 322 
Jun 25 13:39:03 masterMode: True 
Jun 25 13:39:03 slaveAddrs: 07e1a16cc 
Jun 25 13:39:03 outFileName: stdout 
Jun 25 13:39:03 append: w 
Jun 25 13:39:03 opening /dev/ttyUSB0 
Jun 25 13:39:03 dataLen:    0000 
Jun 25 13:39:03 dataLenInv: ffff 
Jun 25 13:39:03 sequence:   039b 
Jun 25 13:39:03 source:     fffffffe 
Jun 25 13:39:03 dest:       7e1a16cc 
Jun 25 13:39:03 function:   0302 
Jun 25 13:39:03 /dev/ttyUSB0 <-- message: 1 length: 22 
Jun 25 13:39:03 data:       12 34 56 79 00 00 ff ff 9b 03 fe ff ff ff cc 16 
Jun 25 13:39:03 data:       1a 7e 02 03 12 75 
Jun 25 13:39:03   
Jun 25 13:39:03 dataLen:    0000 
Jun 25 13:39:03 dataLenInv: ffff 
Jun 25 13:39:03 sequence:   039c 
Jun 25 13:39:03 source:     fffffffe 
Jun 25 13:39:03 dest:       7e1a16cc 
Jun 25 13:39:03 function:   0322 
Jun 25 13:39:03 /dev/ttyUSB0 <-- message: 2 length: 22 
Jun 25 13:39:03 data:       12 34 56 79 00 00 ff ff 9c 03 fe ff ff ff cc 16 
Jun 25 13:39:03 data:       1a 7e 22 03 09 d9 
Jun 25 13:39:03   
Jun 25 13:43:29   
Jun 25 13:43:29 /dev/ttyUSB0 --> message: 1 length: 246 
Jun 25 13:43:29 data:       12 34 56 79 e0 00 00 e0 e0 e0 e0 00 e0 00 e0 00 
Jun 25 13:43:29 data:       00 e0 e0 00 e0 e0 e0 e0 00 e0 e0 00 00 00 e0 00 
Jun 25 13:43:29 data:       00 e0 00 e0 e0 e0 00 00 e0 00 00 00 00 e0 00 00 
Jun 25 13:43:29 data:       e0 e0 e0 e0 00 e0 e0 00 00 e0 e0 e0 e0 00 00 00 
Jun 25 13:43:29 data:       00 00 00 00 e0 e0 e0 e0 e0 00 00 e0 00 00 00 e0 
Jun 25 13:43:29 data:       e0 e0 e0 e0 00 00 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 
Jun 25 13:43:29 data:       e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 
Jun 25 13:43:29 data:       e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 00 
Jun 25 13:43:29 data:       e0 e0 00 00 9f ff ff fe 7e 22 cf 02 20 00 81 00 
Jun 25 13:43:29 data:       00 01 01 08 10 00 00 20 40 40 04 04 af a0 78 be 
Jun 25 13:43:29 data:       7e ff ff bd e1 fe 45 91 12 20 80 d0 80 30 16 79 
Jun 25 13:43:29 data:       23 3c cf 81 b6 f0 ff cf c5 81 4b 89 90 00 18 ba 
Jun 25 13:43:29 data:       33 97 17 9c 75 8f 32 ff ff ff ff ff 7f bd fb df 
Jun 25 13:43:29 data:       b7 3e fb fc 7f bf fe ff bf ff 7c dd c3 ff fc ff 
Jun 25 13:43:29 data:       dd bf ae 00 00 00 00 04 60 ab be ff fe ff ff ff 
Jun 25 13:43:29 data:       ff ff 00 00 00 00 
Jun 25 13:43:29 dataLen:    00e0 
Jun 25 13:43:29 dataLenInv: e000 
Jun 25 13:43:29 sequence:   e0e0 
Jun 25 13:43:29 source:     00e000e0 
Jun 25 13:43:29 dest:       e00000e0 
Jun 25 13:43:29 function:   00e0 
Jun 25 13:43:29 Data length doesn't match inverse length 
Jun 25 13:43:29 data:       e0 00 00 e0 e0 e0 e0 00 e0 00 e0 00 00 e0 e0 00 
Jun 25 13:43:29 data:       e0 e0 e0 e0 00 e0 e0 00 00 00 e0 00 00 e0 00 e0 
Jun 25 13:43:29 data:       e0 e0 00 00 e0 00 00 00 00 e0 00 00 e0 e0 e0 e0 
Jun 25 13:43:29 data:       00 e0 e0 00 00 e0 e0 e0 e0 00 00 00 00 00 00 00 
Jun 25 13:43:29 data:       e0 e0 e0 e0 e0 00 00 e0 00 00 00 e0 e0 e0 e0 e0 
Jun 25 13:43:29 data:       00 00 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 
Jun 25 13:43:29 data:       e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 
Jun 25 13:43:29 data:       e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 e0 00 e0 e0 00 00 
Jun 25 13:43:29 data:       9f ff ff fe 7e 22 cf 02 20 00 81 00 00 01 01 08 
Jun 25 13:43:29 data:       10 00 00 20 40 40 04 04 af a0 78 be 7e ff ff bd 
Jun 25 13:43:29 data:       e1 fe 45 91 12 20 80 d0 80 30 16 79 23 3c cf 81 
Jun 25 13:43:29 data:       b6 f0 ff cf c5 81 4b 89 90 00 18 ba 33 97 17 9c 
Jun 25 13:43:29 data:       75 8f 32 ff ff ff ff ff 7f bd fb df b7 3e fb fc 
Jun 25 13:43:29 data:       7f bf fe ff bf ff 7c dd c3 ff fc ff dd bf ae 00 
Jun 25 13:43:29 data:       00 00 00 04 60 ab be ff fe ff ff ff ff ff 00 00 
Jun 25 13:43:29 data:       00 00 
Jun 25 13:43:29 Unknown function 0x0000 
Jun 25 13:43:29 <stdout> <-- message: 1 length: 60 
Jun 25 13:43:29 data:       7b 22 64 61 74 61 22 3a 20 22 22 2c 20 22 63 6f 
Jun 25 13:43:29 data:       6d 6d 61 6e 64 22 3a 20 38 30 32 2c 20 22 72 65 
Jun 25 13:43:29 data:       73 70 6f 6e 73 65 22 3a 20 30 2c 20 22 73 65 71 
Jun 25 13:43:29 data:       75 65 6e 63 65 22 3a 20 39 32 34 7d 
Jun 25 13:43:29   
Jun 25 13:43:29 {"data": "", "command": 802, "response": 0, "sequence": 924} 
{"data": "", "command": 802, "response": 0, "sequence": 924}
Jun 25 13:43:31 closing /dev/ttyUSB0 
Jun 25 13:43:31 closing <stdout> 
```

Does this means the wire connection between the raspberry pi and my inverter is not good ?

Any help would be much appreciated. Thanks in advance .

jbuehl, 2018-06-25:
If it was working before then it could be a hardware problem with the serial adapter, the raspberry pi, or even the inverter.  You should also verify that the baud rate matches.

fahimmac, 2018-07-03:
Hi, it was the connection between Raspberry Pi and inverter. Working fine now. Thanks !

closed 2018-07-03 by fahimmac