jbuehl/solaredge #104: Getting around SE's 15min data set

TAZ427, 2018-07-19:
Reading through the Active and Passive Ethernet methodologies, it appears to 'redirect' the active data via changing the DNS value to an IP address of the 'server' that the semonitor.py is running on.  Or it pulls the data off the SE server (which I could simply do using my API.) 

What I'm looking for is to be able to poll the SE inverter (and wattnode consumption meter) about once every 10s, and record that data.  I know it's available somehow as the Solar Edge App shows it live, and updates ever few seconds on the App, as well as the Inverters Panel Display, but can't get to it using their standard API.  

Would using the Serial Port allow the active polling of the data?   I could build an Arduino system to do the data polling over Serial Port and send accumulated data over to storage periodically, but I'm not sure if the Serial Port would be able to access this, while it's setup to pass data over Ethernet.

jbuehl, 2018-07-20:
This sounds the same as issue #41

dragoshenron, 2019-10-29:
Hello @Todd1561 ,
do you confirm that the module stats are truncated as integer number, i.e. no decimals?
E.g. module current could be 4A or 5A but not 4.33A
Thanks in advance for the feedback.

Todd1561, 2019-10-29:
I just checked now and it's just whole integers.  The same is reported in the SetApp interface.