jbuehl/solaredge #108: Encryption keys over Ethernet needed?

jshank, 2018-09-02:
Is there a way to gain access to the encryption keys over Ethernet and do you actually need the encryption keys? I seem to be getting all of the data via Ethernet mirroring but wondering if someday soon I'll lose that ability. Is there any benefit to monitoring the data via RS485 such as more frequent polling rate? Can I use RS485-2 since it's not an option as a server connection? I don't really want to give up my revenue grade meter.

# Background

I have an SE10000H-US (2018 new HDWave) inverter. It has 2 RS485 ports. RS485-1 is connected to the SE-RGMTR (revenue grade meter). There is an unsoldered USB connection on the main control board that I'm working on adding a [USB port](http://d.digikey.com/NX007SF7000mP7KLNp2040N) so I can have access to the RS232 interface but I'm not sure why yet.
 ![USB close-up](https://lh3.googleusercontent.com/lTPtl8T4Otpb9zfOH0V-QlbeP4wMD-Xl7P03LFWIEPCxuQEI0ujsfuqmfpJ_veUJZZ4ZP5DItSzEfTcy2Rmb0aiFvb7nwDRB-9N4nKafbt893pvReyEzS0x2SImZ2mdi8uTVecHsbtuSMThMRNyofPsqUfcq_00uUYCS6mGyd2I2waKhhrMmELVmkex0JMZ3dx2_fJZDqakRb6hgyKUZrM0KyZH453sj35h2gpsqk3bA2JdOrQqK7HWxqjHRj8YJFY3fzrp4eP8sDvZ9V48jWTtzahzLcLgYguUppTpkgnZu8jhCHlVtXQir1sTZo9G9UDbGrK1k9NbJHh3JT3ErrZj2u5eBtYXBkqJaepYO2z3r1aY5vzVsttbZ9vKjEWwA27og6TUgu0u4k6zSnxSYtDbJQUSCpyAfK5WWE5z8vD9eII1kSrz7SDLzCYv1CzJ--FdbhzCSfaFF-_9WT44lAcsFji303NvQWRqHiE2kW-KGjSaKnlNVlcxA-rBvsQrFW_kZS0creM1GvztlQehdWHj_VxhLqosDU2BF0I5g2nry8tgRnyICCy-UMzX9ysaYkmxUHMxM2ZHb0sIeV9tMhUVe7RE5KJZNU7pt-76K-to4ECUmg7UEgoVPZparlGpP1xuXw1db10ka-bKbZE_WlEsFO5gS_y1ci2Nzt8Grfwmkx7p4xZdVXsqCkQ=w1140-h855-no)

I have a hard-wired Ethernet connection to the inverter and I'm [mirroring](http://www.wfiltertutorial.com/EdgeSwitch_port_mirror/EdgeSwitch) the traffic with my Ubiquiti EdgeSwitch to a dedicated port on my Linux server which appears to be working even though I haven't captured the encryption keys. I receive the message `Decryption key not yet available` which is apparently not a big deal per #91. I've created a [Docker image](https://github.com/jshank/jb-solaredge-docker) to run these packages.

RS485-1 is configured as:
```
Multi Devices <M>
  Meter 1 is configured as 
    Protocol: SolarEdge <SE>
    Device ID <1>
    Meter Func. <Prod>
    Topology <Wye>
    Device Type <MTR>
```

Attempts to get the keys via RS485-1 failed with multiple error message using a [cheap RS485 adapter](https://amzn.to/2Cazccz). 

```
python semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 73178624 -m -t 4 -vvvv /dev/ttyUSB0 | python utilities/sekey.py -o 73178624.key
append: False
baudrate: 115200
commands: 12,H239 12,H23a 12,H23b 12,H23c
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stderr
master: True
outfile: stdout
ports: 22222,22221,80
record: None
slaves: 73178624
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyUSB0
dataLen:    0000
dataLenInv: ffff
sequence:   0051
source:     fffffffe
dest:       73178624
function:   0302
/dev/ttyUSB0 <-- message: 1 length: 22
data:       12 34 56 79 00 00 ff ff 51 00 fe ff ff ff 24 86
data:       17 73 02 03 4a 42
dataLen:    0002
dataLenInv: fffd
sequence:   0052
source:     fffffffe
dest:       73178624
function:   0012
/dev/ttyUSB0 <-- message: 2 length: 24
data:       12 34 56 79 02 00 fd ff 52 00 fe ff ff ff 24 86
data:       17 73 12 00 39 02 20 82
/dev/ttyUSB0 --> message: 1 length: 28
data:       12 34 56 79 06 00 f9 ff 4f 00 24 86 17 73 fe ff
data:       ff ff 90 00 3a ee a5 51 00 00 ff 2a
dataLen:    0006
dataLenInv: fff9
sequence:   004f
source:     73178624
dest:       fffffffe
function:   0090
value:     51a5ee3a
type:      0000
<stdout> <-- message: 1 length: 90
data:       7b 22 64 61 74 61 22 3a 20 7b 22 74 79 70 65 22
data:       3a 20 30 2c 20 22 76 61 6c 75 65 22 3a 20 31 33
data:       36 39 38 32 38 39 32 32 7d 2c 20 22 63 6f 6d 6d
data:       61 6e 64 22 3a 20 31 38 2c 20 22 72 65 73 70 6f
data:       6e 73 65 22 3a 20 31 34 34 2c 20 22 73 65 71 75
data:       65 6e 63 65 22 3a 20 38 32 7d
{"data": {"type": 0, "value": 1369828922}, "command": 18, "response": 144, "sequence": 82}
dataLen:    0002
dataLenInv: fffd
sequence:   0053
source:     fffffffe
dest:       73178624
function:   0012
/dev/ttyUSB0 <-- message: 3 length: 24
data:       12 34 56 79 02 00 fd ff 53 00 fe ff ff ff 24 86
data:       17 73 12 00 3a 02 22 f3
/dev/ttyUSB0 --> message: 2 length: 28
data:       12 34 56 79 06 00 f9 ff 50 00 24 86 17 73 fe ff
data:       ff ff 90 00 a8 e1 0b 38 00 00 7f 5b
dataLen:    0006
dataLenInv: fff9
sequence:   0050
source:     73178624
dest:       fffffffe
function:   0090
value:     380be1a8
type:      0000
<stdout> <-- message: 2 length: 89
data:       7b 22 64 61 74 61 22 3a 20 7b 22 74 79 70 65 22
data:       3a 20 30 2c 20 22 76 61 6c 75 65 22 3a 20 39 34
data:       30 33 30 32 37 36 30 7d 2c 20 22 63 6f 6d 6d 61
data:       6e 64 22 3a 20 31 38 2c 20 22 72 65 73 70 6f 6e
data:       73 65 22 3a 20 31 34 34 2c 20 22 73 65 71 75 65
data:       6e 63 65 22 3a 20 38 33 7d
{"data": {"type": 0, "value": 940302760}, "command": 18, "response": 144, "sequence": 83}
dataLen:    0002
dataLenInv: fffd
sequence:   0054
source:     fffffffe
dest:       73178624
function:   0012
/dev/ttyUSB0 <-- message: 4 length: 24
data:       12 34 56 79 02 00 fd ff 54 00 fe ff ff ff 24 86
data:       17 73 12 00 3b 02 28 24
/dev/ttyUSB0 --> message: 3 length: 22
data:       12 34 56 79 00 00 ff ff a3 03 24 86 17 73 fe ff
data:       ff ff 01 05 97 b0
dataLen:    0000
dataLenInv: ffff
sequence:   03a3
source:     73178624
dest:       fffffffe
function:   0501
<stdout> <-- message: 3 length: 61
data:       7b 22 64 61 74 61 22 3a 20 22 22 2c 20 22 63 6f
data:       6d 6d 61 6e 64 22 3a 20 31 38 2c 20 22 72 65 73
data:       70 6f 6e 73 65 22 3a 20 31 32 38 31 2c 20 22 73
data:       65 71 75 65 6e 63 65 22 3a 20 38 34 7d
{"data": "", "command": 18, "response": 1281, "sequence": 84}
Traceback (most recent call last):
  File "utilities/sekey.py", line 25, in <module>
    key += struct.pack("<L", data["data"]["value"])
TypeError: string indices must be integers
dataLen:    0002
dataLenInv: fffd
sequence:   0055
source:     fffffffe
dest:       73178624
function:   0012
/dev/ttyUSB0 <-- message: 5 length: 24
data:       12 34 56 79 02 00 fd ff 55 00 fe ff ff ff 24 86
data:       17 73 12 00 3c 02 28 95
/dev/ttyUSB0 --> message: 4 length: 22
data:       12 34 56 79 00 00 ff ff 51 00 24 86 17 73 fe ff
data:       ff ff 9a 03 d9 04
dataLen:    0000
dataLenInv: ffff
sequence:   0051
source:     73178624
dest:       fffffffe
function:   039a
<stdout> <-- message: 4 length: 60
data:       7b 22 64 61 74 61 22 3a 20 22 22 2c 20 22 63 6f
data:       6d 6d 61 6e 64 22 3a 20 31 38 2c 20 22 72 65 73
data:       70 6f 6e 73 65 22 3a 20 39 32 32 2c 20 22 73 65
data:       71 75 65 6e 63 65 22 3a 20 38 35 7d
{"data": "", "command": 18, "response": 922, "sequence": 85}
Traceback (most recent call last):
  File "semonitor.py", line 241, in <module>
    doCommands(args, mode, dataFile, recFile, outFile)
  File "semonitor.py", line 170, in doCommands
    }, outFile)
  File "/home/jshank/solaredge/se/data.py", line 255, in writeData
    outFile.flush()
IOError: [Errno 32] Broken pipe
```

jshank, 2018-09-07:
I finally got the 3rd and 4th memory locations read and manually ran the sekey program and pasted each line (assuming `{"type": 0, "value": 940302760}` is correct). It spit out a key but I don't think it's working. I'm getting a whole bunch of  `Decrypting message` `Data length doesn't match inverse length` and `Ignoring this message`. Any chance the memory locations have changed for CPU 3.2221?

AndyRPH, 2019-02-09:
@jshank  is your docket image all that is needed?  I have an edge router 3 and an UniFi poe8 switch.  Both with free ports so I'm sure one of them can port mirror.  My solar install goes up on Tuesday.  Same 10k HD inverter with revenue grade meter.   I want to be sure I'm ready to packet capture and not lose a window to collect the decryption keys without resorting to a serial interface which may not be available on my inverter.    Thanks!

jshank, 2019-02-09:
@AndyRPH It's still working to this day. I get valuable data from each optimizer and the system via MQTT even thought I've never successfully extracted the keys. No need to mess around with the RS485 or USB port.

```
solaredge    | optimizer:      114C8EDE type: 0080 len: 000d
solaredge    |     Uptime : 2205
solaredge    |     Temp : 12.0
solaredge    |     Vmod : 36.125
solaredge    |     Imod : 0.375
solaredge    |     Eday : 10.25
solaredge    |     Vopt : 13.5
solaredge    |     Time : 16:53:57
solaredge    |     Date : 2019-02-09
solaredge    |     Inverter : 0
solaredge    |     ID : 114C9EDC
```

AndyRPH, 2019-02-09:
Thanks ! I think I'll load that image onto my pi3b+ and get it ready for the install Tuesday.

dragoshenron, 2019-02-10:
@AndyRPH Keep us posted!