jbuehl/solaredge #109: Micro-USB Connector part for SE10000H-US?

jshank, 2018-09-08:
I haven't had any luck finding a micro-usb connector to solder onto the main board so I can gain access to the RS-232 interface. Checked Mouser and Digikey extensively. Does anyone know what part number or where one might find one?

[image]

Those are actually though holes towards the edge of the board.

jbuehl, 2018-09-08:
Wouldn't this one work?

https://www.digikey.com/product-detail/en/molex-llc/0475890001/WM17143CT-ND/1832255

jshank, 2018-09-08:
Not sure how I missed that one, thanks! I've ordered and will report back.

ContextPerspective, 2018-09-15:
> It fit perfectly. Unfortunately, I'm not getting any response from it. I could have installed it incorrectly. I used a hot air rework station for the data pins and a soldering iron for the supports. I'm wondering if there are additional SMC above the port that are also missing. Can anyone who has a USB port snag a picture or record the values of the components above the USB port?
> 
> [More images of board of the board with new USB port](https://photos.app.goo.gl/2mGU3VECcUswAr4A7)

This is a picture I took off my SE6000H a while back. Sorry these aren't clearer, but to answer your question, there does appear to be SMD components above the USB port, which thanks to this thread I just learned existed. ;)

[image]

jshank, 2018-09-16:
Thanks so much @ContextPerspective for the great images and confirming my suspicions on the missing SMD components. Looks like 4 or 5 resistors (one hidden under the display bracket in the upper-right) and a single capacitor. Next step is to determine what type of microcontroller is on these and then find the corresponding application note for connecting a USB interface. Either that or have someone take measurements on the resistors and capacitor. Unfortunately I believe you would need to remove the capacitor to measure it.

ksaihtam, 2018-09-16:
Maybe the schematic looks like this:

[image]

jbuehl, 2018-09-16:
Does the board have an FTDI chip on it?  If so, you might find a schematic for that circuit with component values in their documentation.  Even if it doesn't, you should be able to come up with reasonable values.  The two resistors are probably current limiting devices for the data lines, and there would probably be a bypass capacitor or two across the Vcc and ground. http://forum.arduino.cc/index.php?topic=23894.0

ContextPerspective, 2018-09-16:
@jshank No problem sir. I am happy to help with this cause.

@jbuehl I have not seen an FTDI chip on board, but (perhaps helpful in other ways), there does appear to be a LATTICE LCMX02-256HC FPGA on board. 

If there is an FTDI chip, it might be hidden behind the display section with the 4 stand-offs.

- https://www.latticesemi.com/Products/FPGAandCPLD/MachXO2
- https://www.digikey.com/product-detail/en/lattice-semiconductor-corporation/LCMXO2-256HC-4TG100C/220-1425-ND/2785527

[image]

[image]