jbuehl/solaredge #110: Solar Edge StorEdge Battery connection.

als1980, 2018-09-09:
We are currently working on a battery solution for the StorEdge platform. I know they confirm to sunspec in certain situations. Does anyone have info on how to communicate with the SE from the battery bank . What info is it looking for? Is the protocol open so we can test as is? I see a huge bottle neck for SE if they only embrace LG power walls. Our product will sell, roughly double the power half the cost. Using the best cells on the planet. It is also a green technology. I am a dealer for SE and can't seem to get straight answers. I would figure they would jump at the chance to bring on a new vendor so pricing can be stablized. Any info would be greatly appreciated. Right now we have a 16kwh lithium battery running 96s. So we canatj the +-400 volt solar Edge requires.
Pleassseee pm me if you have any info and if solaredge is here reach out to us.
C.T.O. 
Adam Simmons
ArmsWork Inc.
83 DELAFIELD PL. 
Staten island, NY
10310
Adam.simmoms@armswork.com
Thank you for your time.

ContextPerspective, 2018-09-09:
I am by no means an expert like the other fine folks on here, but I ran across these documents a while back looking into the communication options for my SE6000H inverter and forthcoming system expansion. 

All documents regarding SE's StorEdge platform seem to explain that the SE Inverter uses Modbus over RS-485 to speak to their their StorEdge interface box, which appears to be both a monitor and electrical interface and point of  interconnect to the battery bank. 

**The attached has some detailed wiring diagrams for various configs.**

- [storedge_interface_installation_guide.pdf](https://github.com/jbuehl/solaredge/files/2363856/storedge_interface_installation_guide.pdf)

**Other documents which might help:**
- [https://www.solaredge.com/sites/default/files/storedge_interface_wiring_quick_guide_and_on_site_checklist_for_lg_batteries.pdf](url)

- [https://www.solaredge.com/sites/default/files/storedge_interface_installation_with_powerwall_1.pdf](url)

[image]

[image]

Hope this helps

math322, 2020-09-17:
@als1980 Did you had success to get any documents?