jbuehl/solaredge #112: Reading Voltage (Vdc) or current Wattage (Pac)

chrisiaut, 2018-10-17:
Hello,

thank you so much for this awesome project. I had quite a few discussions with customer care of solaredge how stupid it is that we have to use external APIs (that can only be polled once every 10 minutes).

I have set up a Rasberry Pi to read data via RS232 to USB cable.

It works in the sense that I can send commands and get answers but what I didnt find out is:

How can I read the current Wattage or Voltage?

I read all commands in the [commands.py](https://github.com/jbuehl/solaredge/blob/master/se/commands.py) file but only found code 322 (GET_ENERGY_STATISTICS_STATUS) which prints out the enery of today, this month, this year and total

[image]

But is there a way to display the current Voltage (Vdc) or Wattage(Pac) as the display of the device shows?
[image]

Thanks

utux3, 2019-02-16:
For me, just reading (so not sending any commands) returns all data including Pac and all of the rest.

crazyelectron-io, 2020-03-19:
> For me, just reading (so not sending any commands) returns all data including Pac and all of the rest.

Can you share the exact command used and the inverter model/phases?
Does it work while also sending data to the SolarEdge Portal?
I tried listening on RS232, but unless I send an explicit command I get no data on my SE7K/3-phase model (with LCD).

Sambsop, 2021-04-06:
I found this 'forum'  because i'm also in this monitoring puzzle nowadays.  I'm really dazzled by the enormous work jbuehl has delivered. Phenomenal job Mister.
For the telemetry values  (V/ Pac). I can give you one pointer in case your inverter have LCD screen.
You can send  code 337 (reading LCD screen)  in your query and you'll got :
{"command": 823, "data": "5661635b565d205664635b565d205061635b575d3233382e3520203334302e362020202036302e31505f4f4b3a203030392f303039203c535f4f4b3e202020202020202020202020202020204f4e202000", "response": 920, "sequence": 429}
the 'real-time data is an Hex string which results in:
Vac[V] Vdc[V] Pac[W]**238.5  340.6    60.1P**_OK: 009/009 <S_OK>                ON  
From there you can extract your values.
(Note: it's possible to send to the inverter a 'click' to the  LCD button (PROT_CMD_POLESTAR_SIMULATE_BUTTON_PRESSING = 0x0338) so perhaps you can navigate into the different readings. I haven't tried yet)

I know is not ideal but I use now because the command that i used  8years ago when i started  playing with my inverters, don't work. 
For the info: At the time i first monitored my inverters with RS232  to get the Pac value (others didn't interest me) the 'Operation codes' i used were 517(x205) for telemetry and 802 (x322) for Stats, but the first one doesn't work with the semonitor.py . 

I am trying RS485 but  i can't get it to work. I have to verify the connections and other parameters.

.

Sambsop, 2021-04-07:
Just for the fun, navigating through the different screens (and getting the values) with the simulate pressing button command, works ok. But unfortunately  the code to read telemetry i used to monitor the system (long ago) hang with no response.