jbuehl/solaredge #113: Set-up connection with an SE7k

ShipsWithCannons, 2018-11-12:
A family member of mine put an SE7K-RW00ENNN2 on the wall, fully expecting to be able to just log-in on it like on a router and read the values. Of-course, SolarEdge only appears to offer cloud-based solutions, which won't work at his place due to a bad internet connection.

I'm hoping this project can help out, and so far it looks good. Properly documented, decent code and a good track record. There's just the small problem of RS232. Since I can't find it on this model. RS485, Ethernet, sure. No RS232. No connector at the bottom. No USB.

So I'm starting to think RS232 is only made accessible on North-American models. Normally I wouldn't mind a little soldering to change that, but given it's not my device and it would definitely void the warranty, not this time.

Is there a procedure to get this project to work, preferably over Ethernet, without requiring RS232 and without SE's cloud solution? I do have physical access to the device and unprotected connections are manageable.

dragoshenron, 2018-11-12:
Hello @ShipsWithCannons,

I do have a European model and no trouble to find the USB connector.
The RS232 is needed to read the encryption key and so to decrypt the network traffic.

If you don't mind to lose contact with the SE server, then logging from RS485 could be the easiest choice for you.

Cheers,
Alex

ShipsWithCannons, 2018-11-14:
Thank you for your response @dragoshenron 

Do you have the exact same model? On what board would I find the USB connector?

Since there is no communication with the SE server currently anyway, RS485 would indeed seem the obvious alternative. Not preferred, but much better than nothing.

ContextPerspective, 2018-11-15:
Looking at the datasheet, the SE7K per SolarEdge only supports the following protocols - RS485, Ethernet, Zigbee (optional), Wi-Fi (optional), Built-in GSM (optional)

This looks like the same boat as the HD Wave series . No RS282. But might be able to add RS485. Links below are for the North American models. 

**Datasheet for SE7K**
- https://www.solaredge.com/sites/default/files/se-three-phase-indoor-inverter-datasheet.pdf

ShipsWithCannons, 2018-11-15:
There is a communication bus already presented of which I'm guessing is RS485, so an expansion kit is only required if you need to communicate both with SE and with a secondary logger (since only one master can be present). Communication with SE servers is no requirement, so I think I can just put an RS485 to USB adapter between the SE7k and a laptop running the monitoring software from this repo. That's the working theory at least.

dragoshenron, 2018-11-15:
I do have a SE HD Wave SE-3000H (european model) and there is a USB port (=RS232).
The USB connector is just next to the ethernet connection.

As there is no connection to the SE server, then - as you said - just pull a cable for the RS485, use a converter (I use this cheap stuff https://www.hobbyelectronica.nl/product/max485-module-voor-rs485-seriele-communicatie-usb/ and it's doing its job) and run the script. It will work in theory as well as in practice. Remember to configure the inverter properly!

Indeed, the extra expansion RS485 board is only needed to run a separate modbus network.

Let us know if you can get the RS485 working

scr34m, 2023-09-17:
I've SE7K EU inverter, after changing the monitoring connection to RS485 then i could read some data, but after certain time the communication failed and the blue LED tun off.

The #175 lead me to this little progress.

scr34m, 2023-09-29:
@dragoshenron USB A port is technically just a socket not a valid USB device? It acts as RS232 pins?