jbuehl/solaredge #119: Discussion of Active Network Measures to force new kew exchange

JohnOmernik, 2018-12-06:
In the network protocol, there is a encryption scheme that is implemented by Solaredge. There are many issues that discuss this including #108 #116 #97 #115 and more.  

This network key (term I will use to refer to the computed key that is used to encrypt/decrypt the actual messages) is a key is computed by a periodic rotating key (This may be an iv in AES terms, or something like it, I am not a crypto expert) , and that data in the rotating key  is encrypted with a static key unique each inverter to form the network key which does the encrypt/decrypt of the traffic.  So to summarize my weird terms for things:
- static key - Key unique to each inverter that must be obtained vis RS232 or RS485
- rotating key - Key that is transferred between Solaredge and the inverter and used to compute the network key
- network key - The result of a encryption function that uses the static key and rotating key as inputs. This is used to actually encrypt/decrypt the network traffic. 

In #116, I reported that the rotating key can change/be updated anywhere from  from 5 to 12 hours. This sucks if you start a logging application where the rotating key was exchanged right before you start logging, and you missed 4-7 hours of data. 

In #116 @Jerrythafast reported that you can force a network key exchange manually by unplugging and plugging in the ethernet port. This is handy bu manual and out of the control of the semonitor.py program. 

This issue to discuss active measures to force the inverter or Solaredge to give up on the current session and initialize a new rotating key.  Put on your hacker hat folks, this is the time to be creative.  I will be looking into this today, essentially, we want to see what we can do externally, and hopefully from semonitor.py itself that will cause the same effect as a network cable being unplugged and plugged back in.  

### Some thoughts
- We could, for those with active switches log into a switch and up and down a port. However, we are looking for something that is easy to maintain, having a login/down/up/logout for every active switch out there is not something this project would want to maintain. 
- We must not have collateral damage. I.e. any sort of overdose of network packets etc that could effect other network traffic is a no no. 
- Ideally this would be a fast, consistent,  and controllable thing.  If sending a certain packet works 25% of the time, that would suck, we are looking for a silver bullet, but after discussion, if we fall back on something that most of the time that may be ok. 
- Something that should work without too many weird python modules being added to the code base. 
- ???? Other thoughts for ground rules?

JohnOmernik, 2018-12-06:
So interestingly enough, I was setting up my "lab".  I needed a live way to look for the proper key exchange packet. Based off of network caps, I used the following:

````
sudo tshark -i eno4 -f "vlan and tcp and host 192.168.225.11" -t ad -T fields -e frame.time -e data|grep -v -P "^$"|grep -P "\t[a-f0-9]{36}0305"
````

Essentially, due to my setup, I have to use a vlan filter (you  may remove that for your setup), I am looking for tcp traffic on host 192.168.225.11 (my inverter IP).  

I am watching a mirror port on my switch, and this is how I do my logging (via eno4). 

When I run this on the capture, I get:

````
Dec  4, 2018 18:07:00.202977000 CST	123456792200ddffdd00722e1373feffffff03055a50f43f4b50b01181a40a934cff7a9d12928bf94d4d7c858d770e6bd97653dab1ffda65
Dec  4, 2018 23:33:41.016872000 CST	123456792200ddffa301722e1373feffffff0305e8c5398f0fd06406d827a0afbc07a603a1b7e3e8bd1b245b4c83ac94674fcbd081eee6eb
Dec  5, 2018 04:06:49.060716000 CST	123456792200ddffb301722e1373feffffff030551557c2d372e1ab7fb21b863d32364fee7300cfbe592887f54cd2e494411a2411d3e1298
````
Which is indeed the exchange I am looking for (I believe). 

However, I set it to watch live, and went to my switch config and disabled the port (to mimic pulling the plug). I reenabled the port, and I was hoping to see the exchange... nothing... 

Ok, check my live logging, did IT see a key exchange (perhaps my grep is incorrect) Nope, no key exchange traffic. 

Pulling the plug did seem to do the reset, so that's good. One challenge we have here is that by default there are no ports listening on the Inverter. It's hard to be active with something in this setup.  More to come...

jbuehl, 2018-12-06:
It seems like you have to interrupt the TCP session in some way to force the key to be sent, but if you are using a passive setup to monitor the traffic that doesn't seem possible.  Maybe you could do something with the DHCP server to give the inverter a very short lease and if you want to force it to reconnect, the DHCP server could respond in some way that forces the inverter to reconnect to the SE server.  I'm not sure what this would be, maybe allocate a different IP address than before or maybe just not respond for a while.

jbuehl, 2018-12-06:
Another approach might be to implement a new mode in semonitor.py where it functions as a proxy between the inverter and the SE server.  Rather than passively monitoring the traffic, it would pass messages in both directions and would also have the ability to control the connection.  As with ethernet active mode, DNS would have to be spoofed in order to get the inverter to connect to the semonitor server rather than SolarEdge.  It would have the added advantage of not missing any performance data if it isn't running for a while because the inverter will store it until it successfully sends it to the SE server.

This would not be a trivial change, but it shouldn't be too difficult to implement.  It would need to include the following:
- Add a command line option to specify the mode.
- After receiving a connection from the inverter, open a socket to the same port on the SE server.
- Start a second read thread on the SE server socket.
- When a message is received from either socket, write it to the other socket.
- Don't send PROT_RESP_ACK or PROT_RESP_SERVER_GMT messages.

One potential problem is knowing which SE server hostname to connect to.  My inverter originally connected to prod.solaredge.com, but at some point began connecting to prod2.solaredge.com.  The proxy would want to connect to the same hostname that the inverter is requesting, so it means that there would have to be some coordination with the local DNS server so that when the inverter makes the DNS request for xyz.solaredge.com it it given the IP address of the proxy, and then the proxy will connect to xyz.solaredge.com.

Jerrythafast, 2018-12-06:
@JohnOmernik just pulling the plug and putting it back in is not enough, TCP is resilient enough to survive that. You need to give it a few minutes for the connection to time out.

kobuki, 2019-04-24:
@JohnOmernik an idea: you could use `iptables ... -j REJECT --reject-with tcp-reset` or `iptables ... -j RESET` (you need xt_RESET for the latter) in both directions (and then delete the rules). It would be great to do the key capture procedure without opening the inverter.