jbuehl/solaredge #123: Encryption workaround / 'stateDict' error

utux3, 2019-02-16:
Hi!

Just installed this, I'm getting data in. Looks promising...!

Two reasons for writing: 
1: There is an easy way to sidestep the encryption that is not mentioned in your README so I thought it might be useful to you.
2: se2state.py gives me an error "NameError: name 'stateDict' is not defined"

To avoid the encryption: When you reset the inverter to factory settings. it defaults to no encryption. It is only upon connection to the solaredge cloud that encryption gets negotiated. So to sidestep the issue entirely: inhibit the connection to the cloud servers, change the monitoring server to point to your own machine, setup semonitor.py to listen on network, done. Note: after a factory reset you will need to pair the optimizers again, but that's not very complicated. Source/credit for this workaround (in Dutch): https://gathering.tweakers.net/forum/list_messages/1721977

Thanks,
Maarten

jbuehl, 2019-02-17:
Yes, a factory reset inverter will send data in the clear until encryption is enabled from the SolarEdge server and it will remain unencrypted if it never connects to the network.  That's how I have been using it for several years.  Most people, however, prefer to let the inverters talk to SolarEdge and to monitor the data passively.  This was discussed in the lengthy issue #8.

se2state.py assumes that there is an existing json file.  If it doesn't exist, you can use the -i option to initialize one.  That feature didn't make it to the README.

utux3, 2019-02-17:
Okay, thanks. Funny that this is not mentioned, except buried in that issue #8.

With se2state.py I'm still having the greatest difficulties. I read the supposed solution in issue #107 That doesn't work for me, it just misbehaves all the time. If I feed it from stdin without -i it says "name 'stateDict' is not defined". If I just give it -i <foo> and -o <bar> arguments without feeding anything to stdin it says "No JSON object could be decoded". 

Only when I give it stdin plus the -i <foo> and the -o <bar> arguments I get an output file, but it seems wrong, as it has all the stats of all the devices, not some limited subset. For example it lists all 10 telemetry elements of every optimizer, instead of just the "Temperature" one. Is that the expected behaviour? 
 
Python version is 2.7.13. I've tried on two different machines/distro's, no change. With python3 it doesn't work either. 

I also have a question about invocation of semonitor.py. The readme suggests using a filename of yyyymmdd.json. So do we need to kill this script every day and restart it with a new filename, or what am I missing? Or, if I take that filename literal, I end up with a file literally named "yyyymmdd.json" which is not too useful.

Sorry for the dumb questions but I'm really stuck here...
Maarten

jbuehl, 2019-02-17:
se2state.py is intended to maintain a JSON file that contains the latest values of all the inverter and optimizer parameters as well as a few statistics such as the total energy produced by all inverters and the average temperature of all the optimizers.  It will read from a file containing semonitor.py output and will update the values in the JSON in the file specified by the -o parameter.  The purpose of the -i parameter is so you can reset the values of the summary statistics each day.  Looking at the code I just realized that the -i parameter needs to be followed by a list of inverter IDs so it can properly initialize the statistics.  For example:

`python se2state.py -i 7F104920,7F104A16 -o solar.json
`

You can name the output file of semonitor.py anything you want.  If you want to create a separate file for each day then you would need to restart it every day at midnight and including the current date in the filename would be a reasonable suggestion.

Here is the script that I use.  Using the tee program, the output is directed both to a file containing the current date in the name and also to se2state.py.
```
#!/bin/bash -

INTERFACE=eth1
DATADIR=/data/data/
STATEDIR=/root/
PREFIX=
INVERTERS=7F104920,7F104A16

# wait for the time to get set
while [ `date +%Y` == "1969" ]
do
	echo "Waiting for correct time"
	sleep 1
done

python /root/solaredge/semonitor.py -vv -t n -n ${INTERFACE} | \
tee ${DATADIR}${PREFIX}`date +%Y%m%d%H%M%S`.json | \
python /root/solaredge/conversion/se2state.py -i ${INVERTERS} -o ${STATEDIR}solar.json

```

utux3, 2019-02-17:
Thank you. That has cleared it up. It works now :)
So if I want to limit the parameters displayed, I need to look elsewhere (or filter se2state.py output by grep/awk/sed etc.)

Thanks for your help!
Maarten

closed 2019-02-17 by utux3