jbuehl/solaredge #135: Failed to parse message: Unknown format code 'x' for object of type 'str'

DatafoxChris, 2019-07-17:
Thanks for the great work on this protocol !

Command i used:
python3 semonitor.py -o 2019-07-17.json -m -s 7XXXXXXX -t 4 -b 115200 -vvvv /dev/ttyUSB2

When message meters_0x0022 is received and semonitor.py starts to save it into json outputfile, then this error is thrown:

2019-07-17.json <-- message: 3 length: 677
Failed to parse message: Unknown format code 'x' for object of type 'str'

Thanks for reading :)

jbuehl, 2019-07-17:
Can you post the debug output leading up to the error message?

DatafoxChris, 2019-07-18:
Here is the debug output.

When using semonitor without -vvvv option, this error doesn't occur and json is written into the output file.

/dev/ttyUSB2 <-- message: 61 length: 22
data:       12 34 56 79 00 00 ff ff 98 35 fe ff ff ff 96 3d
data:       12 73 02 03 9c 52
/dev/ttyUSB2 --> message: 65 length: 88
data:       12 34 56 79 42 00 bd ff 75 03 96 3d 12 73 fe ff
data:       ff ff 00 05 22 00 96 3d 12 73 3a 00 3f 5c 2f 5d
data:       09 01 00 00 00 00 00 00 00 80 00 00 00 00 00 00
data:       00 80 00 00 00 00 00 00 00 80 00 00 00 00 00 00
data:       00 80 2c 01 00 00 00 00 00 00 00 00 00 00 00 00
data:       00 00 ff ff 7f ff 45 4c
dataLen:    0042
dataLenInv: ffbd
sequence:   0375
source:     73123d96
dest:       fffffffe
function:   0500
meters_0x0022:  73123D96 type: 0022 len: 003a
    meters_0x0022 : {'73123D96': {'9_PVProduction': {'recType': 9, 'Flag_off36_hex': '00 80', 'Totaloff22_int4': 0, 'P2X': 0.0, 'Date': '2019-07-17', 'Interval': 300, 'seId': '73123D96', 'Time': '19:34:55', 'EfromX': 0, 'dateTime': 1563384895, 'Flag_off28_hex': '00 80', 'E2X': 0, 'Flag_off20_hex': '00 80', 'TotalE2Grid': 0, 'AlwaysZero_off10_int2': 0, 'seType': '0x0022', 'devType': 'meters_0x0022', 'devLen': 58, 'Flag_off12_hex': '00 80', 'AlwaysZero_off26_int2': 0, 'AlwaysZero_off34_int2': 0, 'onlyIntervalData': 1, 'AlwaysZero_off18_int2': 0, 'TotalEfromGrid': 0, 'PfromX': -3.4028234663852886e+38, 'Totaloff30_int4': 0}}}
2019-07-17-02.json <-- message: 5 length: 677
Failed to parse message: Unknown format code 'x' for object of type 'str'
data:       42 00 bd ff 75 03 96 3d 12 73 fe ff ff ff 00 05
data:       22 00 96 3d 12 73 3a 00 3f 5c 2f 5d 09 01 00 00
data:       00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 00
data:       00 00 00 00 00 80 00 00 00 00 00 00 00 80 2c 01
data:       00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff
data:       7f ff 45 4c

Geoff99, 2019-07-18:
Hi there,
The 0x0022 message is decoded in [datadevices.py](https://github.com/jbuehl/solaredge/blob/master/se/datadevices.py ), beginning around line 441.
It was originally written for Python 2.7  It's a guess, but I think this is something that hasn't been fully converted to Python3 yet.
Hope that helps track down the cause of this issue
Geoff
(PS I contributed the original ParseDevice code.  I have yet to convert *myself* to Python3, and for the next week or so I'm away from home as well, so not in a position to investigate further)

ericbuehl, 2019-07-18:
Is it possible for you to upload a data capture that reproduces this error so that it can be tested offline?

DatafoxChris, 2019-07-18:
Yes, you mean a capture made using a second rs485-to-USB module ?

jbuehl, 2019-07-18:
Use the -r option to record a copy of the data to a file.

DatafoxChris, 2019-07-18:
Made the log containing this error: 
Checksum error. Expected 0x4576, got 0xcd17
I seem to be unable to attach the log to this posting ?

DatafoxChris, 2019-07-18:
Trying to add the log...

DatafoxChris, 2019-07-18:
[2019-07-18_Logs.zip](https://github.com/jbuehl/solaredge/files/3406897/2019-07-18_Logs.zip)

This is a more complete set of json and raw logs, stopping the json output after this error is shown:

Checksum error. Expected 0x3412, got 0x6436

jbuehl, 2019-07-18:
Thanks, that's much better.

jbuehl, 2019-07-18:
@DatafoxChris - I just realized that you opened another issue #136.  Are these two different problems?    You mentioned above that the "Failed to parse message" error happens with the -vvvv option.  Does it otherwise work? I'm not clear on what is happening.

DatafoxChris, 2019-07-18:
These are two different problems. 

When using semonitor without -vvvv option, then the json is written into the output file without problems

closed 2019-07-18 by jbuehl