jbuehl/solaredge #140: RS485-1 and RS485-E configuration

antoniolobefaro, 2019-08-28:
Hi, 
first of all I describe the system configuration. 
there is a StorEdge connected electrically to a PV system and to a LG RESU CHEM battery. 
On StorEdge, the RS485-1 is connected to LG storage and to an Energy meter that monitor the energy exchanged with energy provider. the protocol is Multi because of different devices connected.

the external module RS485-E is connected to an external device where i want to run the semonitor to capture data. on RS485-E the StorEdge is set as slave and the protocol is Solaredge.

the StorEdge is also connected via Ethernet and I'm able to see all data on the SolarEdge portal.

Well, I want to monitor the system locally, but semonitor, in the configuration described above, doesn't work. 
 
Any idea? someone met the same case?

Thanks

felixmartens, 2020-05-03:
Hello, Antonio,

Do you have any success with your Installationin the meantime? I'm working with a similar setup, trying to get passive network mode working(as my RS485-1 is connected to a second Single Phase Inverter).
Thanks for your answer

antoniolobefaro, 2020-05-03:
I'm sorry, I wasn't able to manage the system only locally and at the end I managed it by remote commands with rest API

jruibarroso, 2025-03-04:
I have 2 LG Chem Resu batteries and I am looking for CAN/RS485 plug pinout to connect them to a new Huawei inverter. Anyone haves this plug pinout ? Thanks