jbuehl/solaredge #147: RS485 key retrieval value zero

fgnn, 2020-01-08:
I have a pair of SE6000H-US000BNU4 that I am trying to connect to. Some other users have reported they successfully retrieved the keys from RS485 so I thought i would give it a shot. These are the new inverters without a screen which use setapp to configure. Both have RS485-1 set as "SolarEdge Slave", and are wired together and to my raspberry pi with an RS485 to USB adapter. After figuring out that I need to change the "Server" from "LAN" to "RS485-1 Slave" to get them to respond, I was able to get a response to the key retrieval command. They responded only with a value of "0", which does not appear valid. 

Does anyone have any thoughts about this? Is it simply that RS485 can't be used to extract keys or is there something else missing? Thanks to everyone working on this project!

Over the next few days I am going to attempt a USB connection to get the RS232 port. The inverter board has a USB A connection which I find odd since I would need a nonstandard double headed Male cable. Hopefully I can use my phone cable which has type A on one end, and type C on the other for my laptop. I will update this issue about its success. 

The inverters firmware version was just updated and is 4.7.26.

The command output is below. For privacy, SN have been partially replaced inline and in the raw byte stream

```
pi@raspberrypi:~/solaredge $ python semonitor.py -c 12,H239/12,H23a/12,H23b/12,H23c -s 731BXXXX -t 4 -vvvv /dev/ttyUSB0
append: False
baudrate: 115200
commands: 12,H239 12,H23a 12,H23b 12,H23c
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stderr
master: False
outfile: stdout
ports: 22222,22221,80
record: None
slaves: 731BXXXX
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyUSB0
dataLen:    0002
dataLenInv: fffd
sequence:   005c
source:     fffffffe
dest:       731bXXXX
function:   0012
/dev/ttyUSB0 <-- message: 1 length: 24
data:       12 34 56 79 02 00 fd ff 5c 00 fe ff ff ff XX XX
data:       1b 73 12 00 39 02 7e 03
Could not open last0503.msg file, not loading
No data read from last0503.msg. Not loading
Threw out a message that was too short
Unknown function 0x0000
{"command": 18, "data": "", "response": 0, "sequence": 92}
{"command": 18, "data": "", "response": 0, "sequence": 92}
dataLen:    0002
dataLenInv: fffd
sequence:   005d
source:     fffffffe
dest:       731bXXXX
function:   0012
/dev/ttyUSB0 <-- message: 2 length: 24
data:       12 34 56 79 02 00 fd ff 5d 00 fe ff ff ff XX XX
data:       1b 73 12 00 3a 02 7c 72
/dev/ttyUSB0 --> message: 2 length: 28
data:       12 34 56 79 06 00 f9 ff 5c 00 XX XX 1b 73 fe ff
data:       ff ff 90 00 00 00 00 00 00 00 66 d9
dataLen:    0006
dataLenInv: fff9
sequence:   005c
source:     731bXXXX
dest:       fffffffe
function:   0090
value:     00000000
type:      0000
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 93}
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 93}
dataLen:    0002
dataLenInv: fffd
sequence:   005e
source:     fffffffe
dest:       731bXXXX
function:   0012
/dev/ttyUSB0 <-- message: 3 length: 24
data:       12 34 56 79 02 00 fd ff 5e 00 fe ff ff ff XX XX
data:       1b 73 12 00 3b 02 78 21
/dev/ttyUSB0 --> message: 3 length: 28
data:       12 34 56 79 06 00 f9 ff 5d 00 XX XX 1b 73 fe ff
data:       ff ff 90 00 00 00 00 00 00 00 37 49
dataLen:    0006
dataLenInv: fff9
sequence:   005d
source:     731bXXXX
dest:       fffffffe
function:   0090
value:     00000000
type:      0000
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 94}
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 94}
dataLen:    0002
dataLenInv: fffd
sequence:   005f
source:     fffffffe
dest:       731bXXXX
function:   0012
/dev/ttyUSB0 <-- message: 4 length: 24
data:       12 34 56 79 02 00 fd ff 5f 00 fe ff ff ff XX XX
data:       1b 73 12 00 3c 02 78 90
/dev/ttyUSB0 --> message: 4 length: 28
data:       12 34 56 79 06 00 f9 ff 5e 00 XX XX 1b 73 fe ff
data:       ff ff 90 00 00 00 00 00 00 00 c7 b9
dataLen:    0006
dataLenInv: fff9
sequence:   005e
source:     731bXXXX
dest:       fffffffe
function:   0090
value:     00000000
type:      0000
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 95}
{"command": 18, "data": {"type": 0, "value": 0}, "response": 144, "sequence": 95}
closing /dev/ttyUSB0
closing <stdout>
```

felixmartens, 2020-02-22:
I retrieved my key via RS485. I have the old Inverter Model Witz Display. But I had to Split the 4 commands used to retrieve the key and send them seperatly, then concatenate the output manually. Maybe you want to try this Way.

DuaneKaufman, 2021-04-05:
@fgnn - did you ever have luck retrieving keys, or monitoring your system?

I have both a SE7600H and a SE11400H unit for my array. The SE7600 is connected to the SE11400 via RS-485, and the SE11400 is connected through Ethernet to SolarEdge.

I too have tried connecting to RS485-2 (which never responded with anything), and RS485-1 (which is what the SE7600 communicates with) On RS-485-1 I can get data from the SE7600 (optimizers, etc) but not from the SE11400.

I have attached some images from the inside of my SE11400H inverter. Have you had any luck connecting to the USB A connector?

Sincerely,
Duane Kaufman
[image]
[image]

[image]

fgnn, 2021-04-05:
@DuaneKaufman No I never did get the keys or decoded data over the RS485. I ended up using the Modbus TCP interface. While it doesn't give the module level information I was hoping for, it is good enough for my dashboard. I read somewhere that the USB connection is exclusively to be used as a phone charging port when using the SetApp app, which makes sense because I was unsuccessful when trying to detect a virtual serial port on it.

DuaneKaufman, 2021-04-05:
@fgnn  Where does your Modbus TCP interface connect on your inverter? I wonder if I would be able to get data from both inverters using my RS-485-2, and changing something in SetApp?
Your explanation for the USB-A connector on the inverter makes sense.

Sincerely,
Duane Kaufman