jbuehl/solaredge #148: State of (active) network mode

michelwilson, 2020-03-22:
I've been using `semonitor.py` in network mode (`-n <iface>`) for a while. Recently, I updated my repository to the latest change, and everything seems to have stopped working, due to lots of (I think) Python 3-related changes. I've been trying to sort things out, but with little succes so far, neither using Python 2.7 nor using Python 3.0 seems to work. I've also started rummaging around in the code, but that also didn't seem to help. Is it correct that network mode is currently no longer working or not even supported? And what would be the best way to get things to work again?

jbuehl, 2020-03-22:
I also use network mode and I had been using semonitor.py with the `-n `option until about a year ago when I started using another method because of issues that I had been having with the operating system on the server running it.  Now my DNS server is resolving the host name of the SolarEdge monitor servers to my local server so I can have it connected to my local network normally.  I still use the `-t n` option, but I have dropped the `-n` option.  I recall that there were still some issues relating to python3 in that code, but I didn't think anyone else was using it and I didn't want to spend the time fixing them.  I may eventually get around to looking at it, but I am busy with other projects now and it's not a priority for me.  If you want to get it to work and contribute that, it would be appreciated.

michelwilson, 2020-03-22:
Ok, thanks :) will try `-t n` instead of `-n` and set up resolving. Feels a bit cleaner anyway. Are you running it with Python 2.7 currently, or with 3? I also understand not having time to fix this, I was mainly wondering about the state of things :D I might try to put in some work to getting it running again.

jbuehl, 2020-03-22:
I am using python 3.  I have static IP addresses configured in my router for the SolarEdge monitoring servers `prod.solaredge.com`, `prod2.solaredge.com`, and `prod3.solaredge.com` that all point to the local IP address of my server.